mossdef-org/source.mossdef.org #93: [vpn-policy-routing]: omission of fwmask in ip rules

dl12345, 2020-03-31:
I believe that the omission of the fwmask in the generated ip rules is an error. This could lead to missed matches in the presence of other software which is also marking packets. In particular, this lack of a mask in the ip rule interferes with mwan3

The relevant code is in `/etc/init.d/vpn-policy-routing`. A diff follows:

```
--- vpn-policy-routing.bak	2020-03-30 21:58:01.934800934 +0100
+++ vpn-policy-routing	2020-03-30 21:59:51.164684510 +0100
@@ -583,7 +583,7 @@
 			fi
 		done
 		ip -4 route flush cache || ipv4_error=1
-		ip -4 rule add fwmark "$mark" table "$tid" || ipv4_error=1
+		ip -4 rule add fwmark "$mark/$fwMask" table "$tid" || ipv4_error=1
 	fi
 
 	if [ "$ipv6Enabled" -ne 0 ]; then
@@ -596,7 +596,7 @@
 				done
 			fi
 			ip -6 route flush cache || ipv6_error=1
-			ip -6 rule add fwmark "$mark" table "$tid" || ipv6_error=1
+			ip -6 rule add fwmark "$mark/$fwMask" table "$tid" || ipv6_error=1
 		fi
 	fi
```

stangri, 2020-03-31:
Thank you for the patch. Could you please confirm you have tested it with the mwan3 and after the patch it's working correctly?

dl12345, 2020-03-31:
Yes, tested and yes, working correctly. 

I'm using mwan3 as I have one fast but less reliable link (lots of jitter and sometimes lots of dropped packets) and one slower but much higher quality and lower latency link that gets used for DNS and latency sensitive traffic. My mwan3 is configured to failover to the slower link and it also routes the aforementioned traffic to the higher quality link, but with fallback to the faster link should it go down.

vpn-policy-routing is used to selectively route certain traffic that I have via one of two different openvpn tunnels. 

Regarding mwan3 compatibility, the default fwmask for vpn-policy-routing is fine: no clashes with mwan3's mask of 0x3f00. The iptables rules option in vpn-policy-routing needs to set to "Insert" for compatibility with mwan3 and the vpn-policy routing WAN table ID needs to be set to a higher value (I used 4001) as mwan3 uses tables 1001 - 2254.

stangri, 2020-03-31:
> vpn-policy routing WAN table ID needs to be set to a higher value (I used 4001) as mwan3 uses tables 1001 - 2254.

Could you please elaborate on that? Why do the table ID numbers be higher (and not lower) than those of mwan3?

dl12345, 2020-03-31:
Let's use the default mask values for vpn-policy-routing and for mwan3 and a common mark value that both use

```
‭111111110000000000000000‬‬ 0xff0000 vpn-policy-routing mask
0000‭00110000000000000000‬‬ 0x30000  vpn-policy-routing mark
    
00000000‭0011111100000000‬‬ 0x3f00   mwan3 mask  
00000000‭0011111000000000‬ 0x3e00   mwan3 mark
```

combining the two marks on a single packet will give the following

```        
0000‭00110011111000000000‬         combined mark
```

and thus it follows that an ip rule configured to match on just 0x30000 without the mask will fail on the packet marked by both mwan3 (or even sqm) and by vpn-policy routing

dl12345, 2020-03-31:
> > vpn-policy routing WAN table ID needs to be set to a higher value (I used 4001) as mwan3 uses tables 1001 - 2254.
> 
> Could you please elaborate on that? Why do the table ID numbers need to be higher (and not lower) than those of mwan3?

I guess you read my previous comment before I clarified it - see the second comment for the clarification

stangri, 2020-03-31:
Thanks for following up with me on the TID numbers. I'll send the PR later today and I have updated the package in my repo after your initial report.

dl12345, 2020-03-31:
> Thanks for following up with me on the TID numbers. I'll send the PR later today and I have updated the package in my repo after your initial report.

And thanks for taking the time to write this package. I used to do all my policy routing with manual scripts and hotplug scripts and it was a headache to maintain. It's fabulous having a flexible framework that's easier to configure and maintain.

stangri, 2020-03-31:
No-no, thank you for contributing and making OpenWrt better for other users!

Speaking of, if you have any custom user scripts which may be useful for others, feel free to send them!

dl12345, 2020-03-31:
> No-no, thank you for contributing and making OpenWrt better for other users!
> 
> Speaking of, if you have any custom user scripts which may be useful for others, feel free to send them!

Probably the most complete and unique set of scripts I have is for bonding two internet links together (not using two links like mwan3, but actually bonding them together so the throughput is roughly 2x the slowest single link less about 10% for overhead).

It requires a slight kernel modification to the bonding driver to allow it to bond point to point links, which is normally not possible as it doesn't have a MAC address, as well as a server in a data centre to handle the other end of the link along with associated firewall rules. 

I used this 24/7 for about 4 years. It's a pretty specialized use case, so I'm not sure how much call there'd be for a more generalized version, but it's already pretty usable in the sense that it has a uci interface. It was written for 15.05, so will probably need a little updating for 19.07. I don't bond two links together anymore, so don't have the facility to test any changes.

There's [a version of it on my github](https://github.com/dl12345/openwrt-packages/blob/master/BONDING.md).  

Happy to send anything else I come across that might be useful, as I'm in the process of updating my old 15.05 build to 19.07 (99% complete already)

stangri, 2020-03-31:
PR: https://github.com/openwrt/packages/pull/11728, thank you!

closed 2020-03-31 by stangri