mossdef-org/source.mossdef.org #94: [wireshark-helper] Wish: Chooser for Interface

ElreboCM, 2020-04-02:
**Is your feature request related to a problem? Please describe.**
urrently the interface is hardcoded to "br-lan". I need to capture packets in a different VLAN than default LAN. 

**Describe the solution you'd like**
Can you include a configuration option and possible a chooser or textfield for setting the interface?

stangri, 2020-04-02:
Can you please attach your /etc/config/network file so I could test vlan support in Luci?

ElreboCM, 2020-04-02:
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'xxxx:xxxx:xxxx::/48'

config interface 'wan'
	option ifname 'eth0'
	option proto 'static'
	option ipaddr 'XX.XX.XX.XX'
	option netmask '255.255.255.252'
	option gateway 'XX.XX.XX.XX'
	option dns '8.8.8.8 8.8.4.4'

config interface 'lan'
	option type 'bridge'
	option proto 'static'
	option netmask '255.255.255.0'
	option _turris_mode 'managed'
	option ip6assign '60'
	option bridge_empty '1'
	option ifname 'lan1 lan2 lan3'
	option ipaddr '192.168.1.1'

config interface 'wan6'
	option ifname '@wan'
	option proto 'dhcpv6'

config interface 'VLAN1'
	option proto 'static'
	option ifname 'lan4'
	option ipaddr '192.168.2.1'
	option netmask '255.255.255.0'

stangri, 2020-04-02:
So for VLAN1, how would the proper iptables rule look?

ElreboCM, 2020-04-02:
In my case it is sufficient to use lan4 instead of br-lan in your Script as my Vlan is associated with that physical port.

stangri, 2020-04-02:
Pease test luci-app-wireshark-helper (git-20.093.52499-3fd5c07-5) and  wireshark-helper (0.0.1-3) from my repo.

ElreboCM, 2020-04-02:
I tried it unfortuneatly the chooser shows only one option (LAN).

[image]

stangri, 2020-04-02:
> I tried it unfortuneatly the chooser shows only one option (LAN).
> 

Please remove current luci app, refresh opkg lists and install it again. I've updated the app while keeping the same version number.

ElreboCM, 2020-04-03:
Works fine! Thank you!

closed 2020-04-03 by ElreboCM