mossdef-org/source.mossdef.org #95: [vpn-policy-routing] Issue: Interface is not getting created, can't route anything through the VPN

elbasan, 2020-04-05:
**Describe the bug**
Terrible sorry if this isn't the right place to ask but I've tried all day to set this up which I swear worked last year but I switched my router back then and just got back to using Asus routers again.
This is a fresh install, nothing has been added beside `openvpn-openssl`, I'm using nordvpn

**Your configs**

1. ```/etc/config/dhcp```

```sh
config dnsmasq
	option domainneeded '1'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option nonegcache '0'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv6 'server'
	option ra 'server'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'
```

2. ```/etc/config/network```
```sh
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd24:4ca6:ad46::/48'

config interface 'wan'
	option ifname 'eth0.1'
	option proto 'dhcp'

config device 'wan_eth0_1_dev'
	option name 'eth0.1'
	option macaddr '70:8b:cd:e9:e1:29'

config interface 'wan6'
	option ifname 'eth0.1'
	option proto 'dhcpv6'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0.2'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config device 'lan_eth0_2_dev'
	option name 'eth0.2'
	option macaddr '70:8b:cd:e9:e1:28'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0 6t'

config interface 'vpnclient'
	option proto 'none'
	option ifname 'ovpnc0'

config interface 'vpnserver'
	option proto 'none'
	option ifname 'ovpns0'
	option auto '1'
```

3. ```/etc/config/firewall```
```sh
config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option src_ip 'fc00::/6'
	option dest_ip 'fc00::/6'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config include
	option path '/etc/firewall.user'

config zone
	option name 'vpnclient'
	option network 'vpnclient'
	option input 'REJECT'
	option forward 'ACCEPT'
	option output 'REJECT'
	option masq '1'
	option mtu_fix '1'

config zone
	option name 'vpnserver'
	option network 'vpnserver'
	option input 'ACCEPT'
	option forward 'REJECT'
	option output 'ACCEPT'
	option masq '1'

config forwarding
	option src 'vpnserver'
	option dest 'wan'

config forwarding
	option src 'vpnserver'
	option dest 'lan'

config forwarding
	option src 'vpnserver'
	option dest 'vpnclient'

config rule
	option name 'Allow-OpenVPN-Inbound'
	option target 'ACCEPT'
	option src '*'
	option proto 'tcp'
	option dest_port '1194'
```

4. ```/etc/config/vpn-policy-routing```
```sh
config vpn-policy-routing 'config'
	list ignored_interface 'vpnserver'
	option append_src_rules '! -d 192.168.200.0/24'
	option enabled '1'
	list webui_supported_protocol 'tcp'
	list webui_supported_protocol 'udp'
	list webui_supported_protocol 'tcp udp'
	list webui_supported_protocol 'icmp'
	list webui_supported_protocol 'all'
	option verbosity '0'
	option strict_enforcement '0'
	option src_ipset '0'
	option ipv6_enabled '0'
	option iptables_rule_option 'append'
	option iprule_enabled '0'
	option webui_enable_column '0'
	option webui_protocol_column '0'
	option webui_chain_column '0'
	option webui_sorting '1'

config include
	option path '/etc/vpn-policy-routing.netflix.user'

config include
	option path '/etc/vpn-policy-routing.aws.user'
	option enabled '0'

config policy
	option interface 'wan'
	option name 'test'
	option dest_addr 'whatismyip.com'
```

**OpenVPN Config**
```
config openvpn 'vpnclient'
	option client '1'
	option dev_type 'tun'
	option dev 'ovpnc0'
	option proto 'udp'
	option config '/etc/openvpn/de748.nordvpn.com.udp.ovpn'
	list pull_filter 'ignore "redirect-gateway"'
	option route_nopull '1'
	
config openvpn 'vpnserver'
	option port '1194'
	option proto 'tcp'
	option server '192.168.200.0 255.255.255.0'
```

**Expected behavior**
netflix and whatismyip should be routed through the vpn.

**VPN Policy Routing run-time information**

1. Output of ```/etc/init.d/vpn-policy-routing reload``` with ```verbosity``` set to 2:
```sh
vpn-policy-routing 0.2.1-7 started with gateways:
wan/eth0.1/192.168.0.1 [✓]
```

2. Output of ```/etc/init.d/vpn-policy-routing support```:
```sh
vpn-policy-routing 0.2.1-7 running on OpenWrt 19.07.2. WAN (IPv4): wan/dev/192.168.0.1.
============================================================
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
============================================================
Routes/IP Rules
default         192.168.0.1     0.0.0.0         UG    0      0        0 eth0.1
IPv4 Table 201: default via 192.168.0.1 dev eth0.1
IPv4 Table 201 Rules:
32765:  from all fwmark 0x10000/0xff0000 lookup 201
============================================================
IP Tables PREROUTING
-N VPR_PREROUTING
-A VPR_PREROUTING -d 104.27.198.90/32 -m comment --comment test_whatismyip_com -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -d 104.27.199.90/32 -m comment --comment test_whatismyip_com -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
IP Tables FORWARD
-N VPR_FORWARD
============================================================
IP Tables INPUT
-N VPR_INPUT
============================================================
IP Tables OUTPUT
-N VPR_OUTPUT
============================================================
Current ipsets
============================================================
Your support details have been logged to '/var/vpn-policy-routing-support'. [✓]
```

elbasan, 2020-04-05:
also somehow f*ked up and can't connect via LAN anymore, only Wifi works

stangri, 2020-04-06:
Check footnotes for VPN tunnels support: https://github.com/stangri/openwrt_packages/blob/master/vpn-policy-routing/files/README.md.

closed 2020-04-06 by stangri