mossdef-org/source.mossdef.org #98: [vpn-policy-routing] Issue: wg server + wg client setup not working

kaisengit, 2020-04-25:
**Describe the bug**
My setup looks as following:

- ISP-provided router which has direct internet access
- Behind that router is my OpenWrt router
- all my local devices connect to the OpenWrt router either via lan or wifi
- it runs a wireguard client routing all my traffic though it (this has been working fine for years; the openvpn client also works fine)
- a raspberry client serves a nextcloud instance and dns server which also works fine
- now I want to add a wireguard server to connect to my network from outside
- when the the wireguard client is down (i.e. direct access to the internet for all clients) the connection to the wireguard server works fine, so it is correctly set up
- as soon as the wireguard client is up, routing all traffic through it, I can't connect to the wireguard server even though the firewall and vpn-policy-routing are configured correctly
- my nextcloud instance, also with a rule set up via vpn-policy-routing, still works fine
- according to tcpdump there is incoming traffic on the correct port but no answer from my router
- I can switch from the wireguard client to a OpenVPN client; this does not help with the problem however

I have already tried unsuccessfully to get help at the openwrt forum: https://forum.openwrt.org/t/wireguard-server-with-double-nat-and-vpn-client/58787

There is at least one other recent bug report with the same (or similar) issue:
https://forum.openwrt.org/t/trouble-with-openvpn-and-vpn-policy-routing/59687

**Your configs**

1. ```/etc/config/dhcp```

```sh
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option noresolv '1'
	option localservice '1'
	list server '198.245.51.147'
	list server '10.0.254.1'
	list server '172.16.0.1'
	option confdir '/tmp/dnsmasq.d'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv6 'server'
	option ra 'server'
	option ra_management '1'
	list dhcp_option '6,192.168.1.177'
	list dns 'fd26:1a8:5625::409'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config dhcp 'vpn0'
	option start '100'
	option leasetime '12h'
	option limit '150'
	option interface 'vpn0'
	list dhcp_option '6,178.162.222.1'
[... 459 lines cut ...]
IP Tables PREROUTING
-N VPR_PREROUTING
-A VPR_PREROUTING -s 192.168.1.177/32 -p tcp -m multiport --sports 443 -m comment --comment https -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -s 192.168.1.177/32 -p tcp -m multiport --sports 80 -m comment --comment http -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_PREROUTING -m set --match-set wg_client dst -c 0 0 -j MARK --set-xmark 0x30000/0xff0000
-A VPR_PREROUTING -m set --match-set vpn0 dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
IP Tables FORWARD
-N VPR_FORWARD
============================================================
IP Tables INPUT
-N VPR_INPUT
============================================================
IP Tables OUTPUT
-N VPR_OUTPUT
-A VPR_OUTPUT -p udp -m multiport --sports 5892 -m comment --comment wg-server -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
Current ipsets
create wan hash:net family inet hashsize 1024 maxelem 65536 comment
create vpn0 hash:net family inet hashsize 1024 maxelem 65536 comment
create wg_client hash:net family inet hashsize 1024 maxelem 65536 comment
============================================================
Your support details have been logged to '/var/vpn-policy-routing-support'. [✓]
   ```

stangri, 2020-04-25:
I can confirm it's not currently working as expected. Thanks for testig the OpenVPN client instead of the Wireguard client as it was first thing on my list to try. When you tested switching to an OpenVPN client rather than Wireguard client, have you tried switching to TCP rather than UDP as a protocol to the client? When I tested the simulatenous server/client last time it was with the OpenVPN server/client and the README recommends having them use different protocols. Can you try different protocols for WG server/OpenVPN client (essentially meaning switching OpenVPN client to TCP as WG is UDP only)?

It has been on my list to investigate this issue, however I have only one router I can test it with and with my SO working from home, I'm very limited on testing time.

kaisengit, 2020-04-25:
At least I am not crazy; it is good hearing you confirm this issue as it has been driving me nuts. I have indeed tried using the OpenVPN client with TCP. Unfortunately the result was exactly the same. I checked with tcpdump and packets were incoming but no response was sent (so exactly the same as with OpenVPN UDP and wireguard).

stangri, 2020-04-27:
Can you please post the output of `cat /proc/sys/net/ipv4/conf/eth1.2/rp_filter`?

kaisengit, 2020-04-27:
```
root@LEDE:~# cat /proc/sys/net/ipv4/conf/eth1.2/rp_filter
0
```

stangri, 2020-04-28:
I went thru the code changes since mid-2018 and I haven't discovered anything which would affect the broken behavior. I suspect it may be some change in recent (as in 18.06/19.07) branches to do with WAN routing.

Any chance you could try testing VPR on a LEDE 17.x? I've done some testing thru the weekend, but couldn't figure out what's wrong and ran out of time. I can't experiment with my router on the weekdays.

kaisengit, 2020-04-28:
Thank you for your help! I am not sure when (or if) I have the time to flash LEDE 17 but I'll keep that in mind.
One thing I also had on my mind is this simple rule here to get my Internet IP even when the vpn client on the router is active:
```
config policy
	option name 'ddns-ip-check'
	option interface 'wan'
	option dest_addr 'checkip.spdyn.de'
```
One thing I have noticed with this rule: it works fine for all clients in my network but I have never managed to get it to work for the router itself. I even tried to force it by setting `option src_addr '127.0.0.1'` Maybe this is also the problem with vpn client rule? That it only works for other clients in the network and not the router.
So one test I would be able to do this weekend is to check whether another device on my network can act as wireguard server and I can connect to it.

stangri, 2020-04-29:
Yes, this is similar to another OUTPUT rules not working. Weirdly, for me it only doesn't work via WAN, it does work via a second VPN tunnel.

If you have more than one VPN tunnel (ie: two different wireguard tunnels), please test if the OUTPUT rules (domains, etc.) are working for you via alternative VPN tunnel.

kaisengit, 2020-11-20:
I know it's been a few months but in case it's still relevant: the problem seems to be indeed related to rules only applying for the router itself. Applying the rule to other devices in the network works fine.
But please note that I am still on version 0.2.1-13 due to the openwrt I use not upgrading to DSA.

shudack, 2021-01-27:
I have the same problem, I configured the WG Server and OpenVPN Client (NordVPN), and when OpenVPN is active, the WG Client (Android) will not have access to the external network. If I disable the OpenVPN Client, I have access again.

i tried the configuration:
config policy
        option interface 'wan'
        option name 'Wireguard'
        option src_addr '192.168.1.1'
        option src_port '51820'

I have other policies set up and they work well.

Was anyone able to fix this?

Thanks.

stangri, 2021-01-27:
> I have the same problem, I configured the WG Server and OpenVPN Client (NordVPN), and when OpenVPN is active, the WG Client (Android) will not have access to the external network. If I disable the [...]

You forgot `option chain 'OUTPUT'`.

shudack, 2021-01-27:
> You forgot `option chain 'OUTPUT'`.

I tried with chain:
`config policy
        option interface 'wan'
        option name 'Wireguard'
        option src_addr '192.168.1.1'
        option src_port '51820'
        option chain 'OUTPUT'`

But it didn't work either.

stangri, 2021-01-27:
> But it didn't work either.

Check README for required information for troubleshooting and post it in the forum.

shudack, 2021-01-28:
> > But it didn't work either.
> 
> Check README for required information for troubleshooting and post it in the forum.

Solved with added on my OpenVPN (NordVPN):

--pull-filter ignore redirect-gateway
And I used PBR to route specific source subnets to the VPN.

Thanks!

closed 2022-09-29 by stangri