mossdef-org/source.mossdef.org #102: [vpn-policy-routing] Issue: VPN menu disappears because of vpn-policy-routing

NicholasKK, 2020-05-09:
**Describe the bug**
Since yesterday, after updating different luci parts, the vpn menu AND vpn policy-based routing has been disappeared from ui. I tried different troubleshooting and found out that "VPN policy routing 0.2.1-13" AND "luci-app-vpn-policy-routing git-20.128.71839-5366acb-73" are the cause. If I uninstall the packages, the VPN menu appears again, and when I install them, the VPN menu and all it's submenus disappear.
I tried to reset the router, and tried again without any specifik setup, just by installing updating packages, and this happened again.

**Your configs**

1. ```/etc/config/dhcp```

   ```sh
config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv6 'server'
        option ra 'server'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'
   ```

2. ```/etc/config/network```

   ```sh
config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'xxxx:12ba:74e1::/48'

config interface 'lan'
        option type 'bridge'
        option ifname 'eth0.1'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config device 'lan_eth0_1_dev'
        option name 'eth0.1'
        option macaddr 'xx:xx:xx:xx:f5:1c'

[... 198 lines cut ...]
        option enabled 0
   ```

**To Reproduce**
Steps to reproduce the behavior:
 I'm not sure what info you need, however, as I wrote before, I did not change any setup in the router after reset, just to see if I can fix the problem. So to be able to reproduce the issue, just update all packages and reboot, check the ui and you won't have the VPN menu and cant change any setting related to vpn or vpn policy based routing.
This version is written under openwrt ui after all updates has been done: Powered by LuCI openwrt-19.07 branch (git-20.128.71839-5366acb)](https://github.com/openwrt/luci) / OpenWrt 19.07.2 r10947-65030d81f3
4. See error

**Expected behavior**
There should be a VPN menu in luci, where you can access vpn-policy-routing config.

**VPN Policy Routing run-time information**

1. Output of ```/etc/init.d/vpn-policy-routing reload``` with ```verbosity``` set to 2:

   ```sh

   ```

2. Output of ```/etc/init.d/vpn-policy-routing support```:

   ```sh

   ```

stangri, 2020-05-09:
Please list what specific luci parts did you upgrade.

NicholasKK, 2020-05-09:
Hi again,
Updating "luci-theme-bootstrap" and "luci-base" packages breaks the vpn menu, only if luci-app-vpn-policy-routing is installed.
Have a look at following 3 scenarios. the router has been reset after each scenario and the only custom setting after each reset was enabling wireless connection and installing vpn-policy-routing 0.2.1-13 and luci-app-vpn-policy-routing git-20.128.71839-5366acb-73.

Each of the following packages has been upgraded from git-20.057.55219-13dd17f-1 » git-20.128.71839-5366acb-1

1. scenario: luci-theme-bootstrap (the whole luci top menu disappeared) -> luci (still no menu) -> luci-base (still no menu) -> luci-mod-admin-full (menu is back without vpn menu, unless I remove luci-app-vpn-policy-routing, then it's possible to see the vpn menu again, with submenues such as openvpn)

2. scenario: luci (no problem with menu) -> luci-mod-admin-full (no problem with menu) -> luci-theme-bootstrap (the whole menu disappears) -> luci-base (menu is back without vpn menu, unless I remove luci-app-vpn-policy-routing, then it's possible to see the vpn menu again, with submenues such as openvpn) 

3. scenario: luci (no problem with menu) -> luci-base (problem with luci login page, error in bootstrap) -> luci-mod-admin-full (still problem displaying luci page) -> luci-theme-bootstrap (fixes the error page, you can login to luci now, but vpn menu disappears unless luci-app-vpn-policy-routing is removed, then you can see the vpn menu in luci where you can access for example openvpn submenu.)

stangri, 2020-05-10:
Do not upgrade any of the core components (like luci and luci-base) thru opkg.

NicholasKK, 2020-05-10:
Upgrading those components with opkg did not throw error, but having luci-app-vpn-policy-routing did. during my initial problem, I had many other components installed too, luci apps for sqm, openvpn and many others. as soon as I removed just 1 component, vpn policy, it fixes the problem. And if I don't install luci-app-vpn-policy-routing, upgrading those components won't cause any problem!
So I guessing that this will become a problem in near future, when somehow, people begin using the higher version of luci and luci-base. It is may be a better idea to fix the error than "not upgrading components".
But thanks anyway, I have used many hours troubleshooting to help you pinpoint the problem...

closed 2020-05-11 by stangri

keeper1978, 2020-05-14:
This is already a problem, I can't install anything without updating luci-base
Executing package manager
Installing luci-app-openvpn (git-20.131.47987-bcdb9fd-1) to root...
Downloading http://downloads.openw…7987-bcdb9fd-1_all.ipk
Installing luci-compat (git-20.131.47987-bcdb9fd-1) to root...
Downloading http://downloads.openw…7987-bcdb9fd-1_all.ipk
Errors
Collected errors:
* check_data_file_clashes: Package luci-compat wants to install file /usr/lib/lua/luci/tools/webadmin.lua
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/add.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/apply.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/arrow.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/down.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/download.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/edit.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/fieldadd.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/file.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/find.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/folder.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/help.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/key.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/link.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/reload.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/remove.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/reset.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/save.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/up.gif
But that file is already provided by package * luci-base
* check_data_file_clashes: Package luci-compat wants to install file /www/luci-static/resources/cbi/user.gif
But that file is already provided by package * luci-base
* opkg_install_cmd: Cannot install package luci-app-openvpn.
The opkg install command failed with code 255.

stangri, 2020-05-14:
> This is already a problem, I can't install anything without updating luci-base
> Executing package manager
> Installing luci-app-openvpn (git-20.131.47987-bcdb9fd-1) to root...
> Downloading [...]

First of all, which version of OpenWrt are you on? Is it a vanilla OpenWrt or some custom image/build? 
Second, why did you decide that your luci-app-openvpn problem is relevant to this particular issue?

keeper1978, 2020-05-14:
OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.134.55291-ba0fb08

keeper1978, 2020-05-14:
I mean, with a clean installation without an update, I can’t deliver any packages from the repository

a1ad, 2020-05-15:
Same here, i needed to update the packages

a1ad, 2020-05-16:
> I mean, with a clean installation without an update, I can’t deliver any packages from the repository

Found a solution. Dont upgrade Luci-base and bootstrap, instead run this: "opkg install --force-overwrite luci-compat"

stangri, 2020-05-23:
Hey @NicholasKK, I should apologize, there was an issue (since then resolved) with the luci app. After your report I did check with the more senior devs and was told no changes needed for my app. Had I known the .3 release was so close I probably would have been more assertive/proactive on reproducing the issue.

NicholasKK, 2020-05-26:
> Hey @NicholasKK, I should apologize, there was an issue (since then resolved) with the luci app. After your report I did check with the more senior devs and was told no changes needed for my app. [...]

Hi @stangri . I was (and still am) sure that you are doing whatever you can to keep this great addon up to date. I was the first to report it and it wasn't clear for you if it was just me or other people experiencing the same issue too.
Anyways, I thank you (once again) for a great job you are doing in the community.