mossdef-org/source.mossdef.org #103: [vpn-policy-routing] Issue: VPN menu disappear after installing luci app.

a1ad, 2020-05-15:
Luci VPN menu disappear after installing luci-app-vpn-policy-routing

Default 19.07.2 install fully up to date.

Steps to reproduce the behavior:
1. Install 19.07.2
2. Update all packages
3. opkg update; opkg install openvpn-openssl luci-app-openvpn > VPN menu is ok
4. opkg install vpn-policy-routing luci-app-vpn-policy-routing > VPN menu is gone

I tried with the version in the openwrt repo's and with your latest version:
luci-app-vpn-policy-routing - git-20.134.55291-ba0fb08-73
vpn-policy-routing - 0.2.1-15

If i remove luci-app-vpn-policy-routing and reboot router the menu is back.

stangri, 2020-05-15:
> 2. Update all packages

Don't do that.

closed 2020-05-15 by stangri

a1ad, 2020-05-16:
> > 1. Update all packages
> 
> Don't do that.

I am using a build script for Proxmox LXC that first update all the packages. Only "vpn-policy-routing" is breaking stuff...

a1ad, 2020-05-16:
> > 1. Update all packages
> 
> Don't do that.

They are recommending to upgrade luci-base on several topics:
https://forum.openwrt.org/t/unable-to-install-luci-app-sqm-conflict-between-luci-base-and-luci-compat/63863

But found a solution. Dont upgrade Luci-base and bootstrap, instead run this: "opkg install --force-overwrite luci-compat"

stangri, 2020-05-23:
Apparently there was an issue with the luci app, but it's been resolved recently.