mossdef-org/source.mossdef.org #104: [antminer-monitor] Issue: Warning: Unable to obtain device information!

ghoffman9, 2020-06-09:
**Describe the bug**
on mvebu caiman (wrt1900ac v1):
Model	Linksys WRT1900AC v1
Architecture	ARMv7 Processor rev 2 (v7l)
Firmware Version	OpenWrt SNAPSHOT r13520-68b94f0fb4 / LuCI Master git-20.154.54590-05f07e5
Kernel Version	5.4.43

after upgrade to r13519 or r13520, the advanced-reboot feature gives the error:
Warning: Unable to obtain device information!
**To Reproduce**
Steps to reproduce the behavior:
1. reboot
2. reflash build
3. same error
4. tried upgrading opkg - current version gives same error
Package luci-app-advanced-reboot (git-20.113.70804-9d02ef1) installed in root is up to date.

4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Antminer Monitor run-time information**

1. Output of ```/etc/init.d/antminer-monitor start```:

   ```sh

   ```

stangri, 2020-06-09:
I was thrown off by the antminer-monitor title. There was a discussion of changing device names again, that's what probably happened, can you post the output of `cat /tmp/sysinfo/board_name`?

stangri, 2020-06-09:
Duplicate of [#4135](https://github.com/openwrt/luci/issues/4135).

closed 2020-06-09 by stangri

ghoffman9, 2020-06-10:
thnak you.
output of sysinfo/board_name:
linksys,wrt1900ac-v1