mossdef-org/source.mossdef.org #107: [vpn-policy-routing] Issue: IPv6 rules not being populated

NoMonkeyNoMission, 2020-06-13:
**Describe the bug**
VPR_FORWARD, INPUT and OUTPUT rules not being populate in the IPv6 firewall.

Recently, my ISP enabled IPv6 on my connection, and whilst I can access the IPv6 Internet via my ISP, no IPv6 traffic can be tunnelled over the (Wireguard) VPN. It looks to me as though some IPv6 firewall rules aren't being populated.

My VPN provider (Mullvad) has given me a single IP address (/128 prefix), so I'm having to employ NAT6.

**Your configs**

1. ```/etc/config/dhcp```

```
config dnsmasq 'main'
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option localservice '1'
	option noresolv '1'
	option dnssec '1'
	option serversfile '/tmp/adb_list.overall'
	list server '127.0.0.1#5453'
	list server '0::1#5453'
	option confdir '/tmp/dnsmasq.d'

config dnsmasq 'family'
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/family/'
	option domain 'family'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases.family'
	option serversfile '/tmp/adb_list.overall'
	option localservice '1'
	option noresolv '1'
	list interface 'family'
	list notinterface 'lo'
	option dnssec '1'
	list server '185.228.168.168'
	list server '185.228.169.168'
	list server '2a0d:2a00:1::'
	list server '2a0d:2a00:2::'

config dhcp 'lan'
	option instance 'main'
	option interface 'lan'
	option start '100'
	option limit '150'
	option dhcpv6 'server'
	option ra 'server'
	option ra_management '1'
	option force '1'
	option leasetime '168h'
	option ra_default '1'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
[... 522 lines cut ...]
============================================================
IPv6 Tables FORWARD
-N VPR_FORWARD
============================================================
IP Tables INPUT
-N VPR_INPUT
-A VPR_INPUT -m set --match-set mullvad dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_INPUT -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
IPv6 Tables INPUT
-N VPR_INPUT
============================================================
IP Tables OUTPUT
-N VPR_OUTPUT
-A VPR_OUTPUT -m set --match-set mullvad dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_OUTPUT -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
IPv6 Tables OUTPUT
-N VPR_OUTPUT
============================================================
Current ipsets
create wan hash:net family inet hashsize 1024 maxelem 65536 comment
create mullvad hash:net family inet hashsize 1024 maxelem 65536 comment
create mullvad6 hash:net family inet hashsize 1024 maxelem 65536 comment
   ```

stangri, 2020-06-13:
Thank you for elaborate report. Please test with the most recent version from this repo.

NoMonkeyNoMission, 2020-06-13:
Thank you. Have upgraded to 0.2.1-17 and same result: IPv6 PBR aren't being honoured. However, I do see that only the chains which are actually in use are created now; which suggests that my observation was unrelated to the issue at-hand.

```
vpn-policy-routing 0.2.1-17 running on OpenWrt 19.07.3. WAN (IPv4): wan/eth0/xx.yy.zz.6. WAN (IPv6): wan6/@wan/::/0.
============================================================
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
============================================================
Routes/IP Rules
default         losubs.subs.bng 0.0.0.0         UG    0      0        0 pppoe-wan

IPv4 Table 201: default via xx.yy.zz.6 dev pppoe-wan 
10.0.0.0/24 dev br-guest proto kernel scope link src 10.0.0.1 
192.168.20.0/24 dev br-streaming proto kernel scope link src 192.168.20.1 
192.168.30.0/24 dev br-family proto kernel scope link src 192.168.30.1 
192.168.99.0/24 dev wgserver proto kernel scope link src 192.168.99.1 
IPv4 Table 201 Rules:
32759:	from all fwmark 0x10000/0xff0000 lookup 201 

IPv4 Table 202: default via xx.yy.zz.6 dev mullvad 
10.0.0.0/24 dev br-guest proto kernel scope link src 10.0.0.1 
192.168.20.0/24 dev br-streaming proto kernel scope link src 192.168.20.1 
192.168.30.0/24 dev br-family proto kernel scope link src 192.168.30.1 
192.168.99.0/24 dev wgserver proto kernel scope link src 192.168.99.1 
IPv4 Table 202 Rules:
32758:	from all fwmark 0x20000/0xff0000 lookup 202 
IPv6 Table 201: default from xxxx:yyyy:zzzz:/48 via fe80::xxxx:yyyy:zzzz:e000 dev pppoe-wan proto static metric 512 pref medium
IPv6 Table 201: default from xxxx:yyyy:zzzz:::/64 via fe80::xxxx:yyyy:zzzz:e000 dev pppoe-wan proto static metric 512 pref medium
IPv6 Table 201: xxxx:yyyy:zzzz:::/64 dev pppoe-wan proto static metric 256 pref medium
IPv6 Table 201: fe80::/10 dev pppoe-wan metric 1 pref medium
IPv6 Table 201: fe80::/10 dev pppoe-wan proto kernel metric 256 pref medium
IPv6 Table 202: fc00:xxxx:yyyy:zzzz::39a6 dev mullvad proto kernel metric 256 pref medium
============================================================
Mangle IP Table: PREROUTING
-N VPR_PREROUTING
-A VPR_PREROUTING -s 192.168.30.0/24 -p udp -m comment --comment Family -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -s 192.168.30.0/24 -p tcp -m comment --comment Family -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -s 192.168.10.0/24 -m comment --comment Private -c 15065 15750647 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -m set --match-set mullvad dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
Mangle IPv6 Table: PREROUTING
-N VPR_PREROUTING
-A VPR_PREROUTING -s fdf3:e716:6c30:aa::/64 -m comment --comment Private6 -c 537 47903 -j MARK --set-xmark 0x20000/0xff0000
============================================================
NAT IP Table: PREROUTING
iptables: No chain/target/match by that name.
============================================================
NAT IPv6 Table: PREROUTING
ip6tables: No chain/target/match by that name.
============================================================
Current ipsets
create wan hash:net family inet hashsize 1024 maxelem 65536 comment
create mullvad hash:net family inet hashsize 1024 maxelem 65536 comment
create mullvad6 hash:net family inet hashsize 1024 maxelem 65536 comment
============================================================
```

stangri, 2020-06-13:
Oh, did you mean the ip6tables rules for ipsets?

NoMonkeyNoMission, 2020-06-13:
Yes, I don't think so (but I'll confess to being a little out of my comfort zone here). I'm just 'guessing' as to the cause. Here's the rule which isn't being honoured, anyway:

```
config policy
	option name 'Private6'
	option src_addr 'fdf3:e716:6c30:aa::1/64'
	option proto 'tcp udp'
	option interface 'mullvad'
```

stangri, 2020-06-13:
Proper policy-based IPv6 routing requires NAT6, which I'm not sure is well supported in OpenWrt yet. I'll try to include the relevant paragraph in README for the next update.

NoMonkeyNoMission, 2020-06-13:
That would be really good; particularly if you've got a setup that works.

For what it's worth, I'm using [the script from here](https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6) and have set the appropriate options on the firewall zone:

```
config zone
	option name 'wan_vpn'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	option network 'mullvad'
	option masq6 '1'
	option masq6_privacy '0'
```

Still; no dice. But, perhaps I've gone about it the wrong way.

NoMonkeyNoMission, 2020-06-14:
I've managed to get this working without using VPN-PBR; by:

1. Creating the route:
```
config route6
	option interface 'mullvad'
	option target '::/0'
```

2. Ignoring my ISP's prefix delegation on those interfaces where I wish IPv6 traffic to be directed over the VPN with ```list ip6class 'local'```

3. NAT6

So, I've actually turned-off IPv6 support in the vpn-policy-routing config now. I'd like to be able to figure-out how to get it working with VPN PBR, but for now I'm happy that I've achieved my main outcome.

stangri, 2020-06-15:
If you want to create a gist with your config, I can include a link to it in a README.

NoMonkeyNoMission, 2020-06-17:
https://gist.github.com/NoMonkeyNoMission/3bd8413029b1f728c1f00bc1ac0e98b4

stangri, 2020-07-29:
Thank you so much, sorry it took me a while, I've updated the README at my repo, I'll push it to the main repo in the next update.

stangri, 2022-09-29:
Sorry, there have been some improvements in the IPv6 code of `pbr` specifically, do you still need some workarounds for the proper IPv6 policies routing or can you confirm if things work as they are?

stangri, 2023-01-14:
Please create a new issue if the problem still exists in `pbr`.

closed 2023-01-14 by stangri