mossdef-org/source.mossdef.org #108: [vpn-policy-routing] Issue: conflict between IPv6 and Append local IP Tables rules

NoMonkeyNoMission, 2020-06-13:
**Describe the bug**
"IPv6 Enabled" with "Append local IP Tables rules" containing an IPv4 address (e.g. ```! -d 192.168.0.0/16``` throws an error:

```
iptables -t mangle -I VPR_PREROUTING -j MARK --set-xmark 0x020000/0xff0000  -s fdf3:e716:6c30:aa::1/64 ! -d 192.168.0.0/16 -m comment --comment Private6
```

**Your configs**

1. ```/etc/config/vpn-policy-routing```

```
config vpn-policy-routing 'config'
	option verbosity '2'
	option strict_enforcement '1'
	option boot_timeout '30'
	list ignored_interface 'wgserver'
	option dest_ipset 'ipset'
	list webui_supported_protocol 'tcp'
	list webui_supported_protocol 'udp'
	list webui_supported_protocol 'tcp udp'
	list webui_supported_protocol 'icmp'
	list webui_supported_protocol 'all'
	option enabled '1'
	option src_ipset '0'
	option ipv6_enabled '1'
	option iptables_rule_option 'append'
	option iprule_enabled '0'
	option webui_enable_column '0'
	option webui_chain_column '0'
	option webui_sorting '1'
	option webui_protocol_column '1'
	list supported_interface 'wan'
	list supported_interface 'mullvad'
	option append_src_rules '! -d 192.168.0.0/16'

config policy
	option chain 'PREROUTING'
	option name 'Private'
	option src_addr '192.168.10.1/24'
	option proto 'tcp udp'
	option interface 'mullvad'

config policy
	option chain 'PREROUTING'
	option name 'Family'
	option src_addr '192.168.30.1/24'
	option proto 'tcp udp'
	option interface 'mullvad'

config policy
	option name 'Private6'
	option src_addr 'fdf3:e716:6c30:aa::1/64'
	option proto 'tcp udp'
	option interface 'mullvad'
   ```

**To Reproduce**
Steps to reproduce the behavior:
1. Enabled IPv6 support
2. Add a local IP Tables rules containing an IPv4 address
3. reload
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

It should be possible to append an IPv4 local IP tables rule with IPv6 support enabled. The IPv6 firewall tables should still be populated; ignoring the local IP(v4) Tables rules. OR, it should be possible to specify IPv4 local IP tables rules and IPv6 IP tables rules separately.

**VPN Policy Routing run-time information**

1. Output of ```/etc/init.d/vpn-policy-routing reload``` with ```verbosity``` set to 2:

```
Creating table 'wan/pppoe-wan/xxx.yyy.zzz.aaa/xxx:yyyy:zzz:xxx::1/64
fe80::1/10' [✓]
Creating table 'mullvad/xx.yy.zz.aa/fc00:xxxx:yyyy:zzzz::39a6/128' [✓]
Routing 'Private6' via mullvad [✗]
Routing 'Private' via mullvad [✓]
Routing 'Family' via mullvad [✓]
vpn-policy-routing 0.2.1-13 started with gateways:
wan/pppoe-wan/aaa.xxxx.yyyy.zzzz/2a02:xxxx:yyyy:zzzz::1/64
fe80::1/10 [✓]
mullvad/10.99.57.166/fc00:xxxx:yyyy:zzzz::aaaa/128
ERROR: iptables -t mangle -I VPR_PREROUTING -j MARK --set-xmark 0x020000/0xff0000  -s fdf3:e716:6c30:aa::1/64 ! -d 192.168.0.0/16 -m comment --comment Private6
   ```

2. Output of ```/etc/init.d/vpn-policy-routing support```:

```
pn-policy-routing 0.2.1-13 running on OpenWrt 19.07.3. WAN (IPv4): wan/dev/xxx.yyy.zzz.aaa. WAN (IPv6): wan6/dev6/::/0.
============================================================
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
============================================================
Routes/IP Rules
default         losubs.subs.bng 0.0.0.0         UG    0      0        0 pppoe-wan
IPv4 Table 201: default via xxx.yyy.zzz.aaa dev pppoe-wan
10.0.0.0/24 dev br-guest proto kernel scope link src 10.0.0.1
192.168.20.0/24 dev br-streaming proto kernel scope link src 192.168.20.1
192.168.30.0/24 dev br-family proto kernel scope link src 192.168.30.1
192.168.99.0/24 dev wgserver proto kernel scope link src 192.168.99.1
IPv4 Table 201 Rules:
32695:	from all fwmark 0x10000/0xff0000 lookup 201
IPv4 Table 202: default via xx.yy.zz.166 dev mullvad
10.0.0.0/24 dev br-guest proto kernel scope link src 10.0.0.1
192.168.20.0/24 dev br-streaming proto kernel scope link src 192.168.20.1
192.168.30.0/24 dev br-family proto kernel scope link src 192.168.30.1
192.168.99.0/24 dev wgserver proto kernel scope link src 192.168.99.1
IPv4 Table 202 Rules:
32694:	from all fwmark 0x20000/0xff0000 lookup 202
IPv6 Table 201: default from 2a02:8010:672e::/48 via re:da:cted: dev pppoe-wan proto static metric 512 pref medium
IPv6 Table 201: default from re:da:cted::/64 via re:da:cted: dev pppoe-wan proto static metric 512 pref medium
IPv6 Table 201: re:da:cted:/64 dev pppoe-wan proto static metric 256 pref medium
IPv6 Table 201: fe80::/10 dev pppoe-wan metric 1 pref medium
IPv6 Table 201: fe80::/10 dev pppoe-wan proto kernel metric 256 pref medium
IPv6 Table 202: re:da:cted: dev mullvad proto kernel metric 256 pref medium
============================================================
IP Tables PREROUTING
-N VPR_PREROUTING
-A VPR_PREROUTING -s 192.168.30.0/24 ! -d 192.168.0.0/16 -p udp -m comment --comment Family -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -s 192.168.30.0/24 ! -d 192.168.0.0/16 -p tcp -m comment --comment Family -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -s 192.168.10.0/24 ! -d 192.168.0.0/16 -m comment --comment Private -c 369 496844 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -m set --match-set mullvad dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_PREROUTING -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
IP6 Tables PREROUTING
-N VPR_PREROUTING
============================================================
IP Tables FORWARD
-N VPR_FORWARD
-A VPR_FORWARD -m set --match-set mullvad dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_FORWARD -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
IPv6 Tables FORWARD
-N VPR_FORWARD
============================================================
IP Tables INPUT
-N VPR_INPUT
-A VPR_INPUT -m set --match-set mullvad dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_INPUT -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
IPv6 Tables INPUT
-N VPR_INPUT
============================================================
IP Tables OUTPUT
-N VPR_OUTPUT
-A VPR_OUTPUT -m set --match-set mullvad dst -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_OUTPUT -m set --match-set wan dst -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
============================================================
IPv6 Tables OUTPUT
-N VPR_OUTPUT
============================================================
Current ipsets
create wan hash:net family inet hashsize 1024 maxelem 65536 comment
create mullvad hash:net family inet hashsize 1024 maxelem 65536 comment
create mullvad6 hash:net family inet hashsize 1024 maxelem 65536 comment
============================================================
Your support details have been logged to '/var/vpn-policy-routing-support'
   ```

stangri, 2020-06-13:
Thank you for elaborate report. Please test with the most recent version from this repo.

NoMonkeyNoMission, 2020-06-13:
Thanks. Same result, I'm afraid:

```
Creating table 'wan/pppoe-wan/xx.yy.zzz/fxxx:yyyy:zzzzz:ff::1/64
fe80::1/10' [✓]
Creating table 'mullvad/xx.yy.zzz/fxxx:yyyy:zzzzz:ff/128' [✓]
grep: /etc/tor/torrc: No such file or directory
grep: /etc/tor/torrc: No such file or directory
Routing 'Private6' via mullvad [✗]
Routing 'Private' via mullvad [✓]
Routing 'Family' via mullvad [✓]
vpn-policy-routing 0.2.1-17 started with gateways:
wan/pppoe-wan/xx.yy.zzz/fxxx:yyyy:zzzzz:ff/64
fe80::1/10 [✓]
mullvad/xx.yy.zzz/fxxx:yyyy:zzzzz:ff/128
ERROR: iptables -t mangle -I VPR_PREROUTING -j MARK --set-xmark 0x020000/0xff0000  -s fdf3:e716:6c30:aa::1/64 ! -d 192.168.0.0/16 -m comment --comment Private6

vpn-policy-routing 0.2.1-17 monitoring interfaces: wan wan6 mullvad 
```

stangri, 2020-06-13:
Ah, right, I fixed mismatched family when explicitly specified in the policy, forgot about append options.

Should be fixed in `vpn-policy-routing 0.2.1-19` available from my repo now.

NoMonkeyNoMission, 2020-06-13:
Same error with 0.2.1-19, I'm afraid.
```
Creating table 'wan/pppoe-wan/.../...::1/64
fe80::1/10' [✓]
Creating table 'mullvad/.../.../128' [✓]
grep: /etc/tor/torrc: No such file or directory
grep: /etc/tor/torrc: No such file or directory
Routing 'Private6' via mullvad [✗]
Routing 'Private' via mullvad [✓]
Routing 'Family' via mullvad [✓]
vpn-policy-routing 0.2.1-19 started with gateways:
wan/pppoe-wan/.../...::1/64
fe80::1/10 [✓]
mullvad/.../...::39a6/128
ERROR: iptables -t mangle -I VPR_PREROUTING -j MARK --set-xmark 0x020000/0xff0000  -s ddf3:e716:6c30:aa::1/64  ! -d 192.168.0.0/16  -m comment --comment Private6

vpn-policy-routing 0.2.1-19 monitoring interfaces: wan wan6 mullvad
```

stangri, 2020-06-14:
You're right, sorry, rushed the fix. Please check vpn-policy-routing 0.2.1-21. It should come up as a warning, not an error.

NoMonkeyNoMission, 2020-06-14:
First, thanks for working on this fix - much appreciated.

Better. I see the service warning now, rather than error. However, it skips creating rules entirely rather than creating them without the append options. E.g.:

When the warning is displayed:
```
============================================================
Mangle IPv6 Table: PREROUTING
-N VPR_PREROUTING
============================================================
```

When I remove the append option such that the warning is not displayed:

```
============================================================
Mangle IPv6 Table: PREROUTING
-N VPR_PREROUTING
-A VPR_PREROUTING -s ddf3:e716:6c30:aa::/64 -m comment --comment Private6 -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
============================================================
```

I'd expect the rule to be created, just without the append option, in this case.

stangri, 2020-06-15:
> I'd expect the rule to be created, just without the append option, in this case.

Let me sleep on this, I'm not sure it should be the behaviour.

NoMonkeyNoMission, 2020-06-15:
Perhaps an alternative is to allow IPv6 and IPv4 appends to be specified separately? My rationale being that - for my setup at least - I need to run both IPv4 and v6; not either-or. As it stands, if I specify any append options then I have to choose between IPv4 or IPv6; I can't run both.

But, I appreciate you've got the package 'as a whole' to consider and I'm only making arguments for one side of the case, here.

stangri, 2022-09-29:
Sorry, I should have addressed it sooner, the whole `Append local IP Tables rules` feature has been replaced by the IGNORE target, so hopefully this has been resolved.

closed 2022-09-29 by stangri