mossdef-org/source.mossdef.org #109: [vpn-policy-routing] Wish:what technology vpn-policy routing take when do ipv6 policy routing?

youxiaojie, 2020-07-05:
1.I see your vpn-policy-routing go's default by vpn and only ipset's ip goes directly, am I right?
can I do reverse: default is my normal wan without vpn, and certain ip/domains go through vpn?
2.what technology you take to ipv6 policy routing? I only see to use ipv6 npt 1:1 or nat n:1 to convert prefix before routing. but it ugly way I think. for example my ipv6 of pc is 240e:xxxx and router has wan6 240e:xxxx, and wireguard 2001:470:xxxx. unlike ipv4 to easy with nat, how to policy with ipv6?If I direct send src ip 240e's packet  to 2001 vpn It does not communicate.

stangri, 2020-07-06:
> 1.I see your vpn-policy-routing go's default by vpn and only ipset's ip goes directly, am I right?

No, default can be set to whatever you want your default routing to be set.

> can I do reverse: default is my normal wan without vpn, and certain ip/domains go through vpn?

Yes, README has some examples related to this scenario.

> 2.what technology you take to ipv6 policy routing? I only see to use ipv6 npt 1:1 or nat n:1 to convert prefix before routing. but it ugly way I think. for example my ipv6 of pc is 240e:xxxx and [...]

I don't use IPv6, you do need some sort of NAT solution, I'll be adding a link to a user-submitted information for IPv6 routing with VPR to the readme sometime next week.

youxiaojie, 2020-07-07:
does vpn policy routing 1. using ipset to collect a set of domain or ips，then 2. use IPtable to set mark，3.using ip rule to setup policy based routing？4.nat the output packet for correct source ip. 5.do i need to set return path check?

stangri, 2020-07-07:
1. Mark packets using either ipsets or iptables rules.
2. Force packets to the tables with appropriate gateway.

Except fwmark nothing is changed on the packet itself.

closed 2022-09-29 by stangri