mossdef-org/source.mossdef.org #110: [vpn-policy-routing] Issue:

jief666, 2020-07-10:
I got these errors when trying to start the service :
```
Failed to set up 'wan/eth0.2/192.168.0.1'
Failed to set up 'wwan//0.0.0.0'
Failed to set up 'VPN/tun0/10.8.0.2 \033[0;32m[\xe2\x9c\x93]\033[0m'
failed to set up any gateway
```

My router IP address is NOT 192.168.0.1.
So my question is : is '192.168.0.1' hard coded ?

jief666, 2020-07-10:
Sorry, I misunderstood the message. 'wan/eth0.2/192.168.0.1' is the address of the gateway. which is correct on that interface.

In the end, I added . /lib/functions.sh at the beginning of the file '/etc/rc.d/S94vpn-policy-routing'

klonwar, 2021-09-13:
> . /lib/functions.sh

Could you please show your `S94vpn-policy-routing` file? I've tried everything, but still have same error. 
My file content:
```bash
#!/bin/sh /etc/rc.common
# Copyright 2017-2020 Stan Grishin (stangri@melmac.net)
# shellcheck disable=SC2039,SC1091,SC2018,SC2019,SC3043,SC3057,SC3060

PKG_VERSION='0.3.5-3'

# sysctl net.ipv4.conf.default.rp_filter=1
# sysctl net.ipv4.conf.all.rp_filter=1

# shellcheck disable=SC2034
START=94
# shellcheck disable=SC2034
USE_PROCD=1

if type extra_command 1>/dev/null 2>&1; then
...
```

stangri, 2021-09-14:
@klonwar could you please check if you have `ip` or `ip-full` package installed? if you replace `list` with `ls` in the `ip` command does it help? If you have just `ip` package, can you please try `ip-full`? 

Thanks!

klonwar, 2021-09-14:
> could you please check if you have ip or ip-full package installed

Yes, I have `ip-full` package installed. 
[image]

With `ls` in `ip` command it still trows `"ip rule show" does not take any arguments.` error. 

Сan this error appear because I have an old version of openwrt (15.05.1)?

stangri, 2021-09-15:
Oh, I don't think it's been tested on CC. Have you tried turning off the IPv6 support?

klonwar, 2021-09-16:
Yes, it's already disabled. I guess I just need to install the latest version of openwrt. I'll try this later
[image]