mossdef-org/source.mossdef.org #112: [antminer-monitor] Wish: Policy routing to different gateway (via API)

Talkabout, 2020-07-25:
**Is your feature request related to a problem? Please describe.**
In my case the vpn gateway is not the openwrt device itself but a different machine (because my openwrt device does not provide enough openssl performance).

**Describe the solution you'd like**
Instead of routing via tun/tap I would like to route to a different ip address

**Describe alternatives you've considered**
Currently I am using my vpn tunnel based on subnets, but I need the (very nice!) feature to route based on domains also. Additionally the integration into the openwrt UI is a big plus :)

Would be great if you could include this! Shouldn't be that much of a big deal, as the logic simply requires a different IP address.

Thanks!

Bye

stangri, 2020-07-25:
You can either install VPR on the same device which is running the VPN client or, since it 

> Shouldn't be that much of a big deal

Just use the custom user file with the relevant commands.

Talkabout, 2020-07-25:
Thanks for the quick answer!

In both cases I will not be able to configure the routing policy nicely via the OpenWRT UI. Is there a particular reason why routing via a gateway (additionally to the vpn interfaces) is not allowed? Is there a high complexity behind it?

I thought about installing VPR on the VPN gateway but was a little bit concerned about dependencies. Is there a documentation what packages need to be instaled along with VPR?

Thanks!

Bye

Talkabout, 2020-07-25:
I tried to run VPR on my VPN Gateway but there are way too many errors (mostly because of dependencies to OpenWRT it seems), so that is not an option. My VPN Gateway is running a Debian Linux (Buster).

stangri, 2020-07-25:
> In both cases I will not be able to configure the routing policy nicely via the OpenWRT UI.

Why? In a former case install OpenWrt onto the device managing VPN traffic and VPR WebUI with it in the latter, continue using VPR/WebUI as is and add relevant commands to a custom user file.

> Is there a particular reason why routing via a gateway (additionally to the vpn interfaces) is not allowed?

VPR operates on network interfaces defined in OpenWrt. If you can create a dummy interface and include it in the list of supported interfaces it should work.

> I thought about installing VPR on the VPN gateway but was a little bit concerned about dependencies. Is there a documentation what packages need to be instaled along with VPR?

If you mean just copying the init file, it heavily depends on uci and OpenWrt's `/etc/config`. You can probably just save the output of `iptables-save` and `ipset save` and re-use that on the VPN gateway.

stangri, 2020-07-25:
> I tried to run VPR on my VPN Gateway but there are way too many errors (mostly because of dependencies to OpenWRT it seems), so that is not an option. My VPN Gateway is running a Debian Linux (Buster).

Can you not install OpenWrt in a VM there and configure VPN connection and VPR there? There will be some loss of performance but with moden virtualization technologies it would probably be negligible.

Talkabout, 2020-07-25:
> VPR operates on network interfaces defined in OpenWrt. If you can create a dummy interface and include it in the list of supported interfaces it should work.

This is an interesting approach. I need to check how this can be done, but it seems to be the best option.

Thanks!