mossdef-org/source.mossdef.org #114: [slider-support] Issue:

mantheman, 2020-08-04:
**Describe the bug**
Built Openwrt 19.07.3 for GL-AR150. Added "slider-support". 
Added support for AR150:
`gpio-8   (                    |BTN_8               ) in  hi`
so check-slider.ar150 includes `checkslider() { grep 'gpio-8.*hi' /sys/kernel/debug/gpio; }` (without the "auto" part that apparently the other supported routers have). Also added Travelmate and Relayd.

--
Started fresh, added my wifi to travelmate -> so far so good, everything works in normal router mode (the slider is nearest the reset button, and slider-support has not been triggered).
`cat /etc/config/slider-support` are the default values included in the package.

Switch slider to the right, and nothing happens. Did a little digging which led me to the config file that now says:
`option enabled ''\''1'\'''` . Changed this manually to `option enabled '1'`, then everything works as expected.

I think this is the bug:
```
set_enabled() { uci -q set "${packageName}.config.enabled='$1'"; uci commit "${packageName}"; }
```
change to:
```
set_enabled() { uci -q set "${packageName}.config.enabled=$1"; uci commit "${packageName}"; }

```
Also, why is the default configuration `enabled '0'`, but automatically get's changed to `enabled '1'` (well, as of right now it's ''\''1'\''') when I move the slider the first time? If it's not going to respect the enabled flag then why include it? It's no big deal for me since I want to have it enabled for my travel router anyways but you might want to take a look at it. Also I think it's a good idea to that have it enabled by default, since if you have a need for this package then you probably want it to be enabled by default:)

mantheman, 2020-08-06:
[Fixed](https://github.com/stangri/openwrt_packages/commit/4c4856ce4300341ab7fbd33040671ffc36b7352d)

stangri, 2020-08-08:
Sorry it took me a while to fully address. 

I've added support for AR150, could you please confirm the slider on it is BTN_1, like on AR300M and not BTN0 like on AR-750?

Package sets enabled/disabled on start/stop to make sure it is not reloaded/restarted when manually stopped.

mantheman, 2020-08-08:
Hi. No, the slider on gl-ar150 is on BTN_8. 

Btw you *can* include the «auto» part in the checkslider function.
I’m using the 19.07 branch  and it happens to support GL-AR150 on both AR71xx and ATH79 targets (the only GL.inet router to do so, I believe) which I didn’t realize at first. The AR71Xx target didn’t include the «auto» part in /sys/kernel/debug/gpio, but the ath79 target does. 

Then again, the «auto» part of the function makes it incompatible with devices still on ar71xx when using 18.06 or 19.07..? (including 2 of the 3 originally supported routers). Since 19.07 is the stable version you might want to omit the "auto" part to make it compatible with 19.07 or even 18.06.

ar71xx targets in 19.07:
6408 
6416 
Domino Pi
GL-AR150 
GL-AR300
GL.iNet GL-AR300M
GL.iNet GL-AR750
GL.iNet GL-AR750S
GL.iNet GL-MiFi
GL.iNet GL-USB150

ath79 targets in 19.07:
GL-AR150
GL.iNet GL-AR300M-Lite
GL.iNet GL-X750

stangri, 2020-08-08:
Thanks, fixed the Makefile for AR150 to include BTN_8.

Can you please elaborate on the whole `"auto" part`, I'm not sure I understand what you mean.

mantheman, 2020-08-08:
I'm sorry, disregard that, I got things mixed up. I could have sweared that the checkslider function was something akin to this:
checkslider() { grep -q 'button left.***auto**.*hi' /sys/kernel/debug/gpio; }

closed 2020-08-30 by stangri