mossdef-org/source.mossdef.org #115: [slider-support] Wish:

mantheman, 2020-08-12:
Hi.

I would like to use this script with a router that doesn't have a slider, and use a predefined button  instead (for example longpress reset or other button). People can just use the Openwrt wiki to set up their own toggle that calls the script, so no need for additional button logic etc.

I guess it's just a matter of adding another "enabled" variable and check for this variable instead of the position of the gpio? 
You've previously said that "Package sets enabled/disabled on start/stop to make sure it is not reloaded/restarted when manually stopped."
Should I call the new variable "enabled" since you already use that variable, or can I just use the same variable since the original usage of it wouldn't be needed if a button toggle is used?

Thanks! This is super useful!

stangri, 2020-08-12:
This was designed for a clear mode of operation with the slider with the limited selection of routers. I don't think I can support all other various modes of operations on different routers with different button press time-outs.

If you were to develop something, feel free to send the PR, but unless elegantly implemented to support any other router's button, I don't think it will be accepted.

mantheman, 2020-08-12:
I didn't mean that you needed to explicitly support any router at all. Users need to follow the Openwrt guide to make their chosen button to activate the script. Since the script would just rely on a common persistent uci variable , it won't have to match the various physical gpios/button. Though, I can see that might be a  little "too hacky" to include this is any official capacity. 

Anyways thanks for answering 👍

closed 2020-08-30 by stangri