mossdef-org/source.mossdef.org #116: [vpn-policy-routing] Issue: Reloading vpn-policy-routing due to ifupdate of wan6

jonozzz, 2020-08-27:
**Describe the bug**
It's constantly reloading every few seconds even with default configs apparently due to updates to the wan6 interface. It's basically impossible to use. How do I make it stop?

Thu Aug 27 12:17:45 2020 user.notice vpn-policy-routing: Reloading vpn-policy-routing due to ifupdate of wan6 (eth0.2)
Thu Aug 27 12:18:38 2020 user.notice vpn-policy-routing: Reloading vpn-policy-routing due to ifupdate of wan6 (eth0.2)
Thu Aug 27 12:21:49 2020 user.notice vpn-policy-routing: Reloading vpn-policy-routing due to ifupdate of wan6 (eth0.2)

**Your configs**

1. ```/etc/config/dhcp```

   ```sh

   ```

2. ```/etc/config/network```

   ```sh

   ```

3. ```/etc/config/firewall```

   ```sh

   ```

4. ```/etc/config/vpn-policy-routing```

   ```sh

   ```

**To Reproduce**
Steps to reproduce the behavior:
1. Install vpn-policy-routing
2. In my case my ISP gives me IPv6 so I have a working wan6 interface.
3. logread -f
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**VPN Policy Routing run-time information**

1. Output of ```/etc/init.d/vpn-policy-routing reload``` with ```verbosity``` set to 2:

   ```sh

   ```

2. Output of ```/etc/init.d/vpn-policy-routing support```:

```
vpn-policy-routing 0.2.1-13 running on OpenWrt SNAPSHOT. WAN (IPv4): wan/dev/73.35.198.1.
============================================================
Dnsmasq version 2.82  Copyright (c) 2000-2020 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-DNSSEC no-ID loop-detect inotify dumpfile
============================================================
Routes/IP Rules
default         XXX 0.0.0.0         UG    0      0        0 eth0.2
iptables: No chain/target/match by that name.
============================================================
iptables: No chain/target/match by that name.
IP Tables PREROUTING
============================================================
IP Tables FORWARD
============================================================
iptables: No chain/target/match by that name.
IP Tables INPUT
iptables: No chain/target/match by that name.
============================================================
IP Tables OUTPUT
============================================================
Current ipsets
create wan hash:net family inet hashsize 1024 maxelem 65536 comment
create ipsec hash:net family inet hashsize 1024 maxelem 65536 comment
create openvpn hash:net family inet hashsize 1024 maxelem 65536 comment
add openvpn 192.168.128.0/17 comment "f5: 192.168.128.0/17"
add openvpn 10.128.0.0/9 comment "f5: 10.128.0.0/9"
add openvpn 172.16.0.0/12 comment "f5: 172.16.0.0/12"
create torguard hash:net family inet hashsize 1024 maxelem 65536 comment
============================================================
Your support details have been logged to '/var/vpn-policy-routing-support'. [✓]
   ```

stangri, 2020-08-27:
You're missing required items in your report.

jonozzz, 2020-08-27:
For whoever has the same issue with ipv6 & hotplug edit this file and add that line.

`/etc/hotplug.d/iface/70-vpn-policy-routing`

```
#!/bin/sh

if [ "$ACTION" != "ifup" ] && [ "$ACTION" != "ifupdate" ]; then exit 0; fi
[ "$ACTION" = ifupdate -a -z "$IFUPDATE_ADDRESSES" -a -z "$IFUPDATE_DATA" ] && exit 0   # <--- ADD THIS

logger -t vpn-policy-routing "Reloading vpn-policy-routing due to $ACTION of $INTERFACE ($DEVICE)"
/etc/init.d/vpn-policy-routing reload
``

stangri, 2021-02-11:
The iface hotplug file is there for legacy (Chaos Calmner/15.05 TurrisOS) support, it should not be used anymore.

closed 2021-02-11 by stangri

ffainelli, 2023-11-25:
Sorry to comment on this closed issue, this was still occurring to me with OpenWrt 22.03.5 and pbr 1.1.1-7 and this file was still present, albeit looking different. My wan6 connectivity is provided by DHCPv6, eventually I skipped the logging entirely by keying off "wan6" which is not great but gets the job done.

stangri, 2023-11-25:
I've been informed there are some phantom calls for ipv6 updates and I've had to develop workarounds for other packages. If you can try the latest `pbr` from my private repo, which no longer installs that file and let me know how it goes I'd be much obliged.