mossdef-org/source.mossdef.org #117: [vpn-policy-routing] Issue: Wrong service gateway selection after restarting the router

misiek735, 2020-09-05:
Hi!

Router: Xiaomi Router 3g,
Firmware OpenWrt 19.07-SNAPSHOT r11144-2ca5a386ee / LuCI openwrt-19.07 branch git-20.197.49655-210b489
Kernel 4.14.187.

My config is based on the instruction from readme "If the Wireguard tunnel on your router is used as default routing(...)".

After restarting the router, those devices which has been configured to work through vpn does not have the access to the internet, but those devices which are configured to go through wan has normal internet access.
When I click manually in luci the button (service control) to restart the vpn policy routing, then the service gateway is recognized properly (as vpn, not as wan). Changing the Boot Time-out from the default 30 seconds to 60 seconds does not solve the problem.

Thank you in advance for any suggestions.

stangri, 2020-09-08:
Post configs/output mentioned in the README please.

misiek735, 2020-09-08:
Of course!

Just wanted to mention that all my devices which are connected to the router (by wifi or by kable) are configured with a static ip.

etc/config/dhcp
[code]
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.auto'
	option localservice '1'
	list rebind_domain 'free.aero2.net.pl'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config host
	option mac 'blablabla'
	option ip 'some number :)'
	option dns '1'
	option name 'my phone'
[/code]

etc/config/network

[code]
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd01:bafd:f261::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0.1'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'

config device 'lan_eth0_1_dev'
	option name 'eth0.1'
	option macaddr 'blablabla :)'

config interface 'wan'
	option ifname 'eth0.2'
	option proto 'dhcp'
[... 186 lines cut ...]
	list webui_supported_protocol 'icmp'
	list webui_supported_protocol 'all'
	option wan_dscp '1'
	option EsWindscribe_dscp '2'
	option HELWindscribe_dscp '3'
	option StockolmWindscr_dscp '4'
	option enabled '1'
	option boot_timeout '30'
	option Barcelona_dscp '5'

config include
	option path '/etc/vpn-policy-routing.netflix.user'
	option enabled '0'

config include
	option path '/etc/vpn-policy-routing.aws.user'
	option enabled '0'

config policy
	option src_addr 'static ip of the device'
	option name 'my phone name'
	option interface 'Barcelona'

Under this config policy there are other devices which I added. 
[/code]

misiek735, 2020-09-08:
Unfortunately I don't know what do I have to do to provide you the information about "VPN Policy Routing run-time information" with verbocity 2. Do I have to copy-paste all the text from the file /etc/init.d/vpn-policy-routing ?

Thank you in advance for your help!

stangri, 2021-02-11:
Sorry I dropped the ball on this. I believe your default gateway is wireguard, so you should not be setting any policies to target wireguard interface.

closed 2022-09-29 by stangri