mossdef-org/source.mossdef.org #118: [vpn-policy-routing] Wish: block routing

h4de5, 2020-10-19:
**Is your feature request related to a problem? Please describe.**
I want to block certain devices from the internet. 
I am almost sure, that I could solve this, using the firewall page, but I don't like that interface over there - and I almost bricked a device the last time I fiddled around with iptables...

**Describe the solution you'd like**
there should be an additinal value for "block access" in the dropdown. 

**Describe alternatives you've considered**
I tried to simple create a new interface that is not bridged or forwarded to WAN - which showed up in the vpn-policy-routing dropdown eventually. but it failed with the following error - I think because the interface didn't have an IP:

```
ip_set xxx
Please unset 'src_addr', 'src_port' and 'dest_port' for policy "xxx"
```

**Additional context**
i would also appriciate it, ff you can tell me where my initial idea went wrong and how I can fix it!

best regards
PS: great work!

stangri, 2020-10-20:
Have you tried to create fake/non-connected vpn interface and enabling strict policies?

h4de5, 2020-10-20:
I didn't name it vpn but block. Would that make a difference? It was an unmanaged interface without any additional settings. But as it did not get an ip it seems vpn-policy-routing cannot use it for routing. Ill try to give it a static ip next

closed 2021-02-11 by stangri