mossdef-org/source.mossdef.org #123: [vpn-policy-routing] Wish: Using bgpview.io API to get CIDR by ASN

kkeker, 2020-11-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

There is a problem with the fact that to use ipinfo to get cidr via asn you need to pay money or parse the site, as before. Now the script specifies the site hackertarget.com, but there are very small limits and even at home you may encounter restrictions.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I suggest using this free API https://bgpview.docs.apiary.io/#reference/0/asn-prefixes/view-asn-prefixes

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I have looked at many solutions on the internet. It seems to me to be optimal with a simple API and at the same time free.

**Additional context**
Add any other context or screenshots about the feature request here.

tophirsch, 2021-02-02:
+1 for using bgpview.io

I ran into these limitations with hackertarget.com and did a quick fix with jq parsing the json object from bgpview.io.

File: vpn-policy-routing.netflix.user
```
#!/bin/sh
# This file is heavily based on code from https://github.com/Xentrk/netflix-vpn-bypass/blob/master/IPSET_Netflix.sh

TARGET_IPSET='wan'
TARGET_ASN='2906'
TARGET_FNAME="/var/tmp_AS${TARGET_ASN}"

TARGET_URL="https://api.bgpview.io/asn/${TARGET_ASN}/prefixes"

curl -s "$TARGET_URL" 2>/dev/null | jq -r '.data.ipv4_prefixes[].prefix' > "$TARGET_FNAME"

awk -v ipset="$TARGET_IPSET" '{print "add " ipset " " $1}' "$TARGET_FNAME" | ipset restore -!
rm -f "$TARGET_FNAME"
```

I hope this could help someone.

stangri, 2021-02-02:
> I hope this could help someone.

Tremendously. I've included your code in the new netflix.user file in this [commit](https://github.com/stangri/source.openwrt.melmac.net/blob/a56e15c0330a026fd6b5942d2ffa3762b993e600/vpn-policy-routing/files/vpn-policy-routing.netflix.user).

I would like to make it default, however I don't want to include `jq` as dependency and it will silently fail otherwise. If you could rewrite the code to use grep/sed/awk instead, I will happily make it default.

tophirsch, 2021-02-03:
Glad to hear that.

Well, my skills with sed are not that great, but I managed to get the same output as with jq:

`curl -s "$TARGET_URL" 2>/dev/null | sed -ne 's/.*"ipv4_prefixes":\(.*\),"ipv6_prefixes":.*/\1\n/g; s/\}\},/&\n/g; s/"parent":{"prefix":/"parent":{"parent_prefix":/gp' | sed -ne 's/.*"prefix":"\(.\{0,20\}\)",".*/\1/g; s/\\//gp' > "$TARGET_FNAME"`

Let me explain:

1. get string between **ipv4_prefixes** and **ipv6_prefixes**, append **\n**
2. convert every list item to a single line -> replace **}},** with itself and append **\n**
3. replace **prefix** in _parent_ object with **parent_prefix**, avoids a lot of duplicates
4. get values of **prefix**, which are the networks we want to know
5. remove **\\** in every network value

As you can see this is heavily based on the order of the individual items and is likely to fail.

I see the problem with jq as additional package, but maybe it could be added as dependency to the whole vpn-policy-routing package and improve the robustness of parsing the json files. I am also parsing the aws file with jq, which works fine.

Good idea to refactor the whole file and test if jq is installed. I think it should be possible to move the awk and rm line out of the different cases. Something like:
```
if [ -s "$TARGET_FNAME" ]; then
	awk -v ipset="$TARGET_IPSET" '{print "add " ipset " " $1}' "$TARGET_FNAME" | ipset restore -!
	rm -f "$TARGET_FNAME"
fi
```
I attached the script that I am using at the moment (use without .txt extension):
[vpn-policy-routing.netflix.user.txt](https://github.com/stangri/source.openwrt.melmac.net/files/5918588/vpn-policy-routing.netflix.user.txt)

stangri, 2021-02-03:
Thanks @tophirsch , ended up with this: https://github.com/stangri/source.openwrt.melmac.net/blob/4595347bcf746deb43e62220b02f3b347d487c4e/vpn-policy-routing/files/vpn-policy-routing.netflix.user.

stangri, 2021-02-11:
Thanks, implemented in latest code as of now.

closed 2021-02-11 by stangri