mossdef-org/source.mossdef.org #124: [simple-adblock] Wish: tag releases

jmtd, 2021-01-29:
It would be very helpfui if you would maintain Git tags in this repository that correspond to the version of the plugin that is synced over in the openwrt package repository.

For example, right now the version in the openwrt packages repository is 1.8.4-4. If there was a tag in this repository for 1.8.4 or 1.8.4-4 then it would be much easier to find out what is newer here than in the openwrt repository.

Thanks!

stangri, 2021-01-30:
Thanks, I've looked into it, but my understanding is that with multiple packages within the same repo, tagging each release for each package would get messy. Please advise if I misunderstood this.

jmtd, 2021-01-30:
You'd have the same issue with the tag namespace as you have with the issues and PR namespaces on GitHub: you'd need to disambiguate them somehow. Conventionally you're using "[simple-adblock]" in the issue title here to disambiguate which package this issue is about; you would have to use a tag prefix like `simple-adblock-X.Y.Z` to achieve the same thing. Or split them out to their own repos.

stangri, 2022-09-29:
Sorry about taking so long. I've looked at https://docs.github.com/en/rest/releases/releases#create-a-release and I can probably create tags from the build script I use, however sometimes I rebuild packages when there's a new OpenWrt point release, so I will either have to delete/create the tag with the same package version or track if the package was indeed updated. If you could suggest some logic to resolve this, I'd be grateful.

stangri, 2023-01-14:
Please reopen this/let me know if you're willing to contribute the shell script code to achieve this.

closed 2023-01-14 by stangri