mossdef-org/source.mossdef.org #125: [vpn-policy-routing] Wish: Retry upon failure to apply rule

yitzhaq, 2021-02-11:
Whenever my ISP does a DHCP renew on my WAN connection, this triggers a firewall reload, which in turn triggers a VPR reload. For reasons I have not yet had the patience nor time to debug (some race condition somewhere perhaps), VPR will - particularly in this scenario - frequently fail to apply some of my defined rules (no noticeable pattern as to which), leaving my routing in an unpredictable and undesired state. Usually the fix is to manually restart VPR, at which point it will usually succeed on every rule. For obvious reasons I don't want to remove any triggers, and not knowing what exactly causes the failures makes it harder to know where to apply possible workarounds.

As a simple workaround I figured I could perhaps look at the exit code and trigger an additional VPR restart when something fails, but it seems to exit 0 regardless, so that fell through, and starting to parse its output seems a horrible path to go down. Then it occurred to me that this could just as well be a function of the script itself. It's dumb, I admit, but it could help in some situations such as this, or other failure scenarios which can cause traffic to leak or such problems.

It probably shouldn't retry indefinitely, but if there was a configurable option to specify number of times to retry from start upon hitting a failure, it should be simple to avoid a fail loop.

Any chance?

FWIW these are my firewall triggers:

router - ~ # ls /etc/hotplug.d/firewall 
30-banip               90-nat6.fw*            99-vpn-policy-routing
(nat6.fw is the NAT6 script that's been floating around for a long while)

And my iface ones:

router - ~ # ls /etc/hotplug.d/iface   
00-netstate  11-sqm*      15-teql*     20-firewall  20-ntpd*     30-nlbwmon*  98-netmount
(98-netmount is a CIFS/NFS mounting script)

Many thanks in advance from a monthly sponsor.

stangri, 2021-02-11:
Hey @yitzhaq I can definitely fix the return code, I hope I'll be able to push an update today.

To (hopefully) fix the problem of VPR restarting too soon, I could implement a 1 or 2 seconds sleep timer before the restart. I may have to do a special build for you to try it out.

stangri, 2021-02-11:
The return code is properly set in 0.3.2-8 (available from source or as a package from my repos).

With service restarts -- I have previously replaced restart (which involves stop and then start) with the reload (just start). In 0.3.2-8 I have also restored the restart functionality, it will somewhat delay the start by invoking stop first.

In your hotplug scripts, as well as in the `service_triggers()` function in lines 1003, 1005 and 1006 replace `reload` with `restart`: https://github.com/stangri/source.openwrt.melmac.net/blob/master/vpn-policy-routing/files/vpn-policy-routing.init#L1003-L1006

stangri, 2021-02-11:
Also, which OpenWrt version is it? The iface hotplug should NOT be necessary usually.

yitzhaq, 2021-02-12:
Thanks for the quick response and fixes! OpenWRT version is 19.07.4. I don't have an iface hotplug for VPR though (see the list I pasted), if that's what you meant? Only for firewall, and that's what in turn triggers VPR. Or are you saying I also shouldn't need a firewall hotplug in iface? Cause that would surprise me.

So just to be sure I get this straight:
I should replace reload with restart as per your instructions in both hotplug and the script itself. Is this for (temporary) testing purposes, or something I would need to do permanently? I'm hoping for the former, obviously, to avoid maintenance overhead with new releases.
It might take a little while to test this properly, to see that it (hopefully) keeps working across a number of WAN DHCP renews. I can so far report that 0.3.2-8, without your suggested changes (I've yet to add them), does not solve the problem by itself, though I guess that also wasn't expected.

I hope the delay introduced by doing restart rather than reload will be sufficient to get around the race condition, assuming that is in fact what causes this. I've used VPR for a few years now, and this didn't use to be a problem, but since I couldn't pinpoint exactly when it started my suspicion was that it might have been introduced when I started using banIP a while back. Though that didn't entirely gel with me, since it loads its (admittedly semi-huge set of) IPs as ipsets. But I wasn't aware of the reload/restart change in VPR, so this seems a likely explanation - or perhaps some combination of the two.

That said, I do still think my other suggestion (a configurable setting to try n restarts upon failure) holds some merit, dumb as it may be, if only for added robustness (possibly against future, unanticipated failure modes). But with the return code fix (thanks again for that) I should be able to get around that myself in the hotplug trigger script (should it prove necessary) if you don't feel it's worthwhile to add.

stangri, 2021-02-12:
Sorry about the hotplug confusion, I see you don't have the iface one, only firewall one.

Yes, replacing `reload` with `restart` is only temporary testing thing. 

Attempting to start again if there are errors is a solid idea, I need to think about how to best implement it tho to avoid recursive loop.

yitzhaq, 2021-02-13:
I'm afraid I'm still seeing the errors today, even with these measures in place.

yitzhaq, 2021-02-13:
BTW, the idea behind having a configurable option to restart n times upon failure is to avoid a recursive loop (and obviously some errors can be permanent), though I'm sure there's more to it than the novel idea itself. I'd probably have set it to 5 or something like that, which I imagine should be enough to get around most of what else might be going in on parallel.

stangri, 2021-02-13:
@yitzhaq is there any way you could capture the system log of things, esecially events before/after failed VPR restart?

yitzhaq, 2021-02-15:
It's quite verbose and a lot to dig through, which is why I haven't had the patience to sift through it much so far. There's too much of it to obfuscate for posting here, though I could probably email it to you directly.

From digging through it a bit more now though, I do see a possible failure scenario. The firewall reload seems to trigger not only on WAN renew, but on every interface, and I have quite a few WireGuard tunnels. So this I suspect might provide a clue:

# grep "Reloading firewall due to ifup of" /var/log/messages | grep "Feb 15 23"
Feb 15 23:21:06 routern firewall: Reloading firewall due to ifup of wan (eth1)
Feb 15 23:21:07 routern firewall: Reloading firewall due to ifup of wan_6 (6rd-wan_6)
Feb 15 23:21:08 routern firewall: Reloading firewall due to ifup of wireguard0 (wireguard0)
Feb 15 23:21:09 routern firewall: Reloading firewall due to ifup of wireguard4 (wireguard4)
Feb 15 23:21:10 routern firewall: Reloading firewall due to ifup of wireguard6 (wireguard6)
Feb 15 23:21:11 routern firewall: Reloading firewall due to ifup of wireguard2 (wireguard2)
Feb 15 23:21:12 routern firewall: Reloading firewall due to ifup of wireguard5 (wireguard5)
Feb 15 23:21:13 routern firewall: Reloading firewall due to ifup of wireguard3 (wireguard3)

(All of the above tunnels have a remote peer defined, so they are tied WAN as such, in so much as the remote peer connection happens over WAN. wireguard1, for instance, does not have a remote peer, and does not trigger.)

To me, it seems as though the WAN renew also triggers a reload of each of the tunnels, each of which in turn triggers its own firewall reload, which in turn triggers multiple VPR reloads. And while the VPR reloads (or restarts as they are right now) appear to happen sequentially, at least, there seems to be indication that they are taking place simultaneously with other triggers (like banIP) still loading. Frankly I'm surprised this doesn't break worse than it does - ugh.

Going with this theory, I guess it boils down to whether this should be considered an OpenWrt bug, or somehow an issue with my local setup, which I can or should work around somehow in the firewall hotplug trigger (or elsewhere?). And whether it's a failure scenario which VPR should try to recognize and work around somehow. I dunno how common this is.

stangri, 2021-02-15:
I believe PROCD shouldn't allow PROCD-based init scripts to run simultaneously, but also I have my own locks in the init script to prevent simultaneous execution.

One more thing I wanted to suggest you test is inserting this line in line 993 of [service_triggers](https://github.com/stangri/source.openwrt.melmac.net/blob/master/vpn-policy-routing/files/vpn-policy-routing.init#L990-L1010):

```
PROCD_RELOAD_DELAY=3000
```

The value is in microseconds, if 3 seconds is not long enough, try increasing it to 5-10-15.

yitzhaq, 2021-02-16:
Yeah it doesn't look as though any VPR re(load|start)s are running simultaneously, though at least judging by the ordering of log output it might appear as though VPR and banIP might be tripping over each other to some extent, in spite of the 30- and 99- prefixes. Either way, it would clearly be better if the triggers got compressed down to a single reload chain, rather than a sudden queued burst of eight of them, but that's not a VPR problem of course.

I'll try adding the PROCD_RELOAY_DELAY, and set it high to begin with just to see if it has a beneficial effect. If it does I'll try gradually lowering it, and if not I'll try temporarily disabling banIP to see if that can at least help pinpoint things.

yitzhaq, 2021-02-19:
Looking promising after 3 days with delay set to 15 seconds, no errors so far. If there are no errors after the weekend I'll start gradually lowering it.

stangri, 2021-02-20:
Thanks for testing, version 0.3.2-12 has this as configurable option in the config file: https://github.com/stangri/source.openwrt.melmac.net/blob/master/vpn-policy-routing/files/vpn-policy-routing.conf#L12.

In config it's in seconds, not milliseconds!

yitzhaq, 2021-02-24:
I think you nailed it. No failures anymore at 15 nor 10 seconds, after running for a week. I did get an error last night after having lowered it to 5 seconds, but I'll experiment with values in the 5-10s range to find the sweet spot.

Many thanks for fixing and excellent help troubleshooting!

As for this issue; the various fixes added as a result are strictly speaking something other than what the title and initial description requests (retry on failure), which you have said you might still be interested in adding. Do you wanna just leave this issue open for that purpose, or would you rather I close this and open another?

stangri, 2021-02-24:
Yes, I think we can keep it open, wouldn't hurt to keep a reminder it's on a wish list.

yitzhaq, 2021-04-07:
Hm, now the firewall hotplug trigger seems to have stopped working. Happened within the past few days. WAN (and thus every wireguard interface) still reloads every few hours, and while other triggers (like nlbwmon and banIP) fire, VPR does not. My only VPR trigger is /etc/hotplug.d/firewall/70-vpn-policy-routing

Related to 796398d perhaps? Want me to try downgrading?

stangri, 2021-04-08:
I'll look into it on the weekend.

stangri, 2021-04-11:
The interface reloads are not the same as firewall reloads. For the interface reloads the init file has the PROCD declarations and in the case they don't work, there's an example of the file you can place in `/etc/hotplug.d/iface/`. Do you have an iface hotplug script in place?

yitzhaq, 2021-04-11:
I don't, since it hasn't been necessary previously, and you also said before that it usually isn't. I'd be happy to try it, however I'd expect that something like this should trigger a VPR reload by virtue of the fw being reloaded, no? Cause that's what's now stopped working.

`Apr 11 06:00:00 router firewall: Reloading firewall due to ifupdate of wan (eth1)`

hotplug.d/firewall has banIP and VPR, and the banIP trigger fires but the VPR trigger does not:

`Apr 11 06:00:00 router banIP-0.3.11[29917]: start banIP processing (reload)`

BanIP, like VPR, does not have an iface hotplug.d trigger, only a firewall trigger.

stangri, 2021-04-11:
And there's nothing even in the logs? Also just to confirm -- which version are you running at the moment?

yitzhaq, 2021-04-11:
That's right, VPR logs nothing when this happens, and routing table shows that rules defined in VPR are not applied, which obviously is unfortunate. It works fine after I do a manual restart.
```
router - ~ # opkg list-installed vpn-policy-routing
vpn-policy-routing - 0.3.4-1
```

stangri, 2021-04-12:
can you please do:
```sh
cat /etc/hotplug.d/firewall/70-vpn-policy-routing 
cat /etc/hotplug.d/iface/70-vpn-policy-routing
```

yitzhaq, 2021-04-12:
```
router - ~ # cat /etc/hotplug.d/firewall/70-vpn-policy-routing 
#!/bin/sh

[ "$ACTION" = "reload" ] || exit 0

logger -t "vpn-policy-routing" "Reloading vpn-policy-routing due to $ACTION of firewall"
/etc/init.d/vpn-policy-routing reload
router - ~ # cat /etc/hotplug.d/iface/70-vpn-policy-routing
cat: can't open '/etc/hotplug.d/iface/70-vpn-policy-routing': No such file or directory
```
```
router - ~ # grep "Reloading vpn-policy-routing" /var/log/messages
[1] router - ~ #
```
Last router reboot was April 9th, no entries at least since then.

stangri, 2021-04-12:
If you comment out (insert `#` in front) the line `[ "$ACTION" = "reload" ] || exit 0` -- does that help?

yitzhaq, 2021-04-12:
Will try, I'll know in a day or three.

yitzhaq, 2021-04-13:
That seems to have been right on the money. The action is now somehow not "reload", but "add" and "remove".

```
Apr 13 17:57:30 routern vpn-policy-routing: Reloading vpn-policy-routing due to remove of firewall
Apr 13 17:57:31 routern vpn-policy-routing: Reloading vpn-policy-routing due to add of firewall

```

stangri, 2021-04-13:
Thanks for testing this!

Just to be clear (because I believe the issue was opened for a different problem originally), in your case the VPR wasn't being reloaded on the interface up/down and the way to get it reloaded was to use the firewall hotplug script? Is this what's going on?

yitzhaq, 2021-04-13:
So the original issue back in February (sorry for recycling, I mistakenly thought they might be related) was that VPR was reloading, but some rules were randomly failing to apply. That was worked around with PROCD_DELAY.

The (new) issue now is that the VPR reload was failing to trigger at all on firewall reload, which in turn is triggered by WAN ifupdate (which also triggers a slew of WireGuard interface ifupdates, each of which trigger an additional fw reload). And the reason for that seems to be that the firewall action is now no longer reload, but remove and subsequent add. Why that's changed on my system I can't really say; I've made a bunch of fw changes lately, which fit the timeline, but nothing I can immediately think of which might cause the action(s) to have changed. At least not deliberately.

The original problem needed changing of the firewall hotplug script during testing (change from reload to restart), but this has not been needed since setting the PROCD_DELAY, so I've been using the stock hotplug script, which worked until recently. This new issue obviously will need to either not look at action (like now), or trigger on action "add" in addition to "reload".

stangri, 2021-04-13:
The firewall hotplug script is only for when the firewall is reloaded/restarted manually. When interfaces go down and come up, VPR should be reloaded thru PROCD. If I do ifdown/ifup on a supported (WAN/wireguard) interface, the VPR reload is triggered without invoking firewall hotplug script on my system. What's the OpenWrt version you're running and what type of interfaces are you using that VPR can't detect the reload of?

yitzhaq, 2021-04-14:
Oh! Then it sounds like I may have jumped to some erroneous conclusions - thanks for clarifying.
I'm on 19.07.4, but I don't think that's related. I was on the same version long before this started.
I also don't think my interface types are too strange, at least they shouldn't have changed for quite some time.

guest (eth2.nnn)
lan (br-lan)
openconnect (openconnect[0-n])
openwireless (eth2.nnn)
tun0 (openvpn)
tv (br-tv)
tv_tagged_down (eth2.nnn)
tv_untagged_up (eth3)
wan (eth1)
wan_6 (6rd-wan_6)
wireguard[0-6]

Most of these are unmanaged, and the interesting ones are wan and the wireguard ones. The nnns are just VLAN tags.

I'm trying to remember what changes I made around the same time which might be remotely related. The first thing that springs to mind is moving the NAT6 script everyone's using from /etc/hotplug/firewall/ to instead being a firewall include:

```
config include 'nat6'
        option path '/etc/firewall.nat6'
        option reload '1'
```
I struggle to see how it could be related, but as already proven, my understanding of the exact mechanics involved here seems less developed than what I originally thought..

stangri, 2021-04-15:
What could have happened is that you've had an iface hotplug.d script (from the README) earlier but then you've installed the VPR version which deleted that script during install.

If you could reproduce the error of the VPR not being reloaded after the WAN/WG interfaces are updated on either 19.07.7 or 21.02 (when it's released) I will try to look into it. Alternatively, please create the iface hotplug script on your system (it's in the README) -- newer VPR versions will not delete it upon install/reinstall.

yitzhaq, 2021-04-15:
I'm pretty sure I never had an iface script. And if I ever did, I doubt it would have gone missing within just the past few weeks. It's been a fair while since my last upgrade, to 19.07.4 (I could never get sysupgrade working on x86_64, so I always reinstall and restore, which is a total hassle and thus tends to get put off), and I don't have any iface files or dirs in my sysupgrade.conf, so if I ever had one that would've disappeared with every upgrade/reinstall/restore, of which there have been plenty.

I've generally been keeping up with VPR releases from your repo. My hunch - for whatever that may be worth - is that one of the changes I made in my last round of fw changes triggered this seemingly elusive issue. The question, if so, would be what.
Is there any chance that adding an include script with option reload 1 could somehow trip over your reload mechanism? I'm not too keen on testing that right now as I can't make sweeping changes quite as easily as when I changed this, but if you think there might be anything to the theory, I could try to test it next chance I get. FWIW, before making that last round of changes, my only include was /etc/firewall.user, which does not have option reload 1.

What I've done for now is this in /etc/config/firewall:

```
config include 'vpr'
        option path '/etc/firewall.vpr-reload'
        option reload '1'
```

Which is simply this:

```
router - ~ # cat /etc/firewall.vpr-reload 
#!/bin/sh
# Ensure VPR is always reloaded whenever firewall is reloaded
# <https://github.com/stangri/source.openwrt.melmac.net/issues/125>

/etc/init.d/vpn-policy-routing reload
```

It's obviously just a dirty hack, but at least it ties the triggering of VPR reload directly to the firewall reloading. Probably no better or worse than the iface hack, but it seems to work (famous last words..), and executes nicely in sequential order together with other included firewall scripts. The separate scriptlet is just cause I got lazy figuring out a sensible way to pass the reload argument to `/etc/init.d/vpn-policy-routing` from within uci.

Let me know if you think this is for some reason a spectacularly bad idea, and if not I'll probably leave it until the next upgrade, and get back to it then as you suggest.

yitzhaq, 2021-08-16:
So unfortunately, since a few versions ago, I've started experiencing frequent problems with this again. I haven't made any changes in my end - IOW the previously discussed workarounds are still in place - but currently this is failing again every 1-2 days, which is quite frustrating. Just to be clear; by "failing" I mean that VPR will fail on one or more routes during one or more of its reloads, which will remain failed until next triggered reload. Current procd_reload_delay is 16 (same as it was for months and mostly without issue), and the ugly workaround script from my previous comment is still in place. I haven't done any recent testing, but this started happening after updating to the latest version, which appears to include several reload-related fixes done through the summer, so I strongly suspect that has something to do with it.
I have not yet had a chance to upgrade OpenWRT, still waiting for a final release of 21.02.

There is one thing I've been thinking of. All my interfaces (save for one OpenVPN tunnel) have fixed IPs. This includes WAN, all the WG tunnels (obviously), and a few OpenConnect ones. WAN is special however, in that the IP is effectively static, but needs to be served by DHCP. If I don't make a renew by lease expiry, it stops passing traffic, but DHCP will always assign me the same IP. I can't just define it statically (I've tried, and failed).
So what I was wondering is whether I could simply remove one or more triggers (VPR or firewall or something), or maybe all of them, or perhaps make it/them conditional based on something, like interface or some state to look for or whatever. Is this anything to consider or look into? Could it help? As it looks to me, nothing about my setup actually *changes* whenever DHCP on WAN assigns me a "new" IP, yet all the triggers still run, and as of now, cause problems.

If this path is the wrong one to be going down; what else can be done to make this all less frail for a setup like mine, and which is likely to not break again with more future script changes? Even when it doesn't fail, traffic is still temporarily dropped several times every day whenever a reload happens (or rather, numerous reloads happen in a row because of all the dependent tunnels), and routes get (needlessly?) recreated - would be wonderful if this too was possible to avoid. Very frustrating when your SSH connections suddenly all get nuked at once every few hours.

Thanks again for all your time spent on this.

stangri, 2021-08-17:
Can you try removing `vpn-policy-routing` and installing [this new package](https://dev.melmac.net/repo/pbr_0.9.1-3_all.ipk) from my repo?

If you have an elaborate config, you can just do: `sed 's/vpn-policy-routing/pbr/g' /etc/config/vpn-policy-routing > /etc/config/pbr`.

I've changed the package name from `policy-routing` to just `pbr`, but the rest of information [on reddit](https://www.reddit.com/r/openwrt/comments/otuj7f/im_working_on_the_next_iteration_of/) is accurate.

yitzhaq, 2021-08-20:
Interesting developments! I will be playing with PBR this weekend, but a few questions that spring to mind:

1. Is testing on 19.07 safe?

2. Should I get rid of the current workarounds (procd delay, wrapper script with sleep etc) to test this as-is out of the box, or keep them?

3. While fighting with VPR recently after failures, I've seen it doing reloads with interface name as argument - is this a first iteration of what PBR aims to solve?

4. A reload of WAN will still trigger a reload of all dependent interface (numerous tunnels) - do you think the architectural changes in PBR will help with that? Or is that what testing aims to determine?

5. You mention that "all policies follow the priority/order as they are listed". Just to be clear; does this mean the caveats from the README about "Policies Priorities" no longer applies, or am I misunderstanding? In other words, can I now enable src_ipset and dest_ipset (which would presumably lower processing time) without worrying that my carefully selected ordering will get overridden?

I like the new name, BTW! And I hope my testing can be of help, since I use a number of different tunnels (9 as of writing), across several different protocols (WireGuard, OpenConnect, OpenVPN), with what I suppose might qualify as an "elaborate config" including both src rules, dest rules and DNSmasq ipsets, different chains etc.

stangri, 2021-08-21:
> 1. Is testing on 19.07 safe?

You can still go back and forth between VPR and PBR, but PBR should work on 19.07. I think I dropped 18.06 support from all new versions, but not 19.07.

> 2\. Should I get rid of the current workarounds (procd delay, wrapper script with sleep etc) to test this as-is out of the box, or keep them?

Please carry the existing config over (that's where PROCD delay is, isn't it), if you have a wrapper script with sleep (despite PROCD delay), you may still need it, however I'd appreciate if you could test it cleanly.

> 3\. While fighting with VPR recently after failures, I've seen it doing reloads with interface name as argument - is this a first iteration of what PBR aims to solve?

Yes, reloads not caused by an interface status change and reloads caused by an interface status change should be dramatically faster.

> 4\. A reload of WAN will still trigger a reload of all dependent interface (numerous tunnels) - do you think the architectural changes in PBR will help with that? Or is that what testing aims to determine?

Oh, I'm not sure if I've tested ifdown/ifup of wan, I was testing only bring down/up the WG interfaces. I'll need to test what happens when I bring down/up WAN and it triggers the avalanche of other interfaces status changes. You will most likely see PBR reload for each interface, but they should be faster and maybe you can mitigate that with increasing PROCD delay.

> 5\. You mention that "all policies follow the priority/order as they are listed". Just to be clear; does this mean the caveats from the README about "Policies Priorities" no longer applies, or am I [...]

That's right, if needed, a special ipset is created for each policy and an iptables rule is created for that specific ipset respecting its priority.

That's one of the biggest changes, if you could test that, it would be awesome! Also, I would welcome some feedback from you on how to present the options for preferring ipsets over general iptables rules.

yitzhaq, 2021-08-21:
> > 1. Is testing on 19.07 safe?
> 
> You can still go back and forth between VPR and PBR, but PBR should work on 19.07. I think I dropped 18.06 support from all new versions, but not 19.07.

No oddities that seem version-related discovered so far, at least.

> Please carry the existing config over (that's where PROCD delay is, isn't it), if you have a wrapper script with sleep (despite PROCD delay), you may still need it, however I'd appreciate if you [...]

I'm testing cleanly now, but I discovered while tinkering that my last VPR upgrade overwrote the call to my sleep wrapper script workaround (should've thought of that! but it was just done as a dirty and frail hack), which may be the reason for the many recent failures I've been experiencing.
For now I'll try to just raise PROCD delay really high (35 instead of 16) in hopes that it might have roughly the same effect, if not I may experiment with reintroducing it.

> Yes, reloads not caused by an interface status change and reloads caused by an interface status change should be dramatically faster.

Sounds great on paper, but lately I've seen plenty of signs suggesting reloads tripping over each other, which could be a likely reason for failure. Hoping this will work better, and make it possible to lower the PROCD delay.

> Oh, I'm not sure if I've tested ifdown/ifup of wan, I was testing only bring down/up the WG interfaces. I'll need to test what happens when I bring down/up WAN and it triggers the avalanche of other [...]

Hope you'll be able to test this a bit, because I believe this is the source of most of my headaches over the past half a year. I at least will have no choice but to test this, and will try to do so as cleanly as possible (while experimenting more PROCD delay, as mentioned).

My wan reloads several times a day, is that not normal DHCP behavior? Would *love* to get rid of this by any means possible (if at all possible).

> That's right, if needed, a special ipset is created for each policy and an iptables rule is created for that specific ipset respecting its priority.

I've gotten a rough impression of this now, and it seems quite sweet! Frankly, this is how I wished VPR had worked all along, so I could use ipsets for some of my more ridiculous rules (at most I've had four-digit numbers of iptables rules, which needless to say impacts performance, though I've since made a concentrated effort to optimize the worst of that away). 

> That's one of the biggest changes, if you could test that, it would be awesome!

You bet! My first observation is that, from your Reddit list, it appears as though I'm currently getting 1.2, despite having `option resolver_ipset` set to `dnsmasq.ipset`. All domain-based rules seem to be resolved on (re)load rather than on access. Is there something more/other I need to set to get 1.1 instead?

It also seems to fail on any domain with both an A and AAAA record. Based on the error output I'm guessing it's trying to stick the resolved IPv6 in together with the IPv4, which obviously won't work too well.

> Also, I would welcome some feedback from you on how to present the options for preferring ipsets over general iptables rules.

If you can elaborate what you mean and want to accomplish, I can try to opine a bit.

yitzhaq, 2021-08-21:
Another observation: it seems a pbr_iface_dst_net ipset is not created if you haven't added any CIDR rules for a particular iface. That makes sense, but it makes it harder to reuse in user scripts. Another complicating factor is the cfg suffix naming.
I guess the bundled user scripts would need a bit of massaging to work well with PBR. Or maybe none of this is new, and I just haven't played around enough with the existing ipset support before.

yitzhaq, 2021-08-21:
> You bet! My first observation is that, from your Reddit list, it appears as though I'm currently getting 1.2, despite having `option resolver_ipset` set to `dnsmasq.ipset`. All domain-based rules [...]

Hm, actually.. dnsmasq ipset *does* appear to be working, from my brief testing. But pbr still seems to be resolving domains on every (re)load, and reporting failures about domain-based rules, where it's trying to insert iptables entries with mixed IPv4 and IPv6. Not sure why that might be happening.

yitzhaq, 2021-08-23:
Experience so far seems to suggest pbr has the same avalanche issues (reloads tripping over each other and causing failures) that I've experienced with vpr, I'm afraid..

yitzhaq, 2021-08-24:
router - ~ # ps | grep -i pbr
 1665 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1667 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1669 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1671 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1672 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1673 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1674 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1675 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1676 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1678 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1679 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1682 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1683 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 1684 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 8977 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
 9803 root      1076 S    sh -c config() { echo "You cannot use UCI in firewall includes!" >&2; exit 1; }; . /usr/share/pbr/pbr.firewall.include
 9804 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
11022 root      1788 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
11023 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
11034 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
11035 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
11036 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
11038 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
11434 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr running
11979 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr running
19655 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr running
24220 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr running
25882 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr running
25884 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr running
28202 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr running
28502 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
29815 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr running
31693 root      1524 S    /bin/sh /etc/rc.common /etc/init.d/pbr reload
32614 root      1240 S    grep -i pbr

yitzhaq, 2021-08-24:
I notice pbr installs both a hotplug trigger in iface, a hotplug trigger in firewall, and a config include reload in /etc/config/firewall. Only the first seems to use interface as an argument, while the other two are general reloads.

I remember from before you saying the iface hotplug trigger should normally not be needed. Has that now perhaps changed? Does one need all three?

stangri, 2021-08-24:
I can't recall what prompted me to create the firewall include, but looks like that one is not necessary (not to mention generating errors).

On OpenWrt 19.07 and up, the reload should be triggered by PROCD triggers, so you can try removing both hotplugs and firewall include and see if that works for you.

yitzhaq, 2021-08-29:
>> That's great if you wanna use wan in your user script(s), but less so if you wanna use a different interface. So at least for my needs, it would be helpful if one was created for each interface, and not just wan. Though with memory concerns, perhaps this should be optional and/or behind a setting?
> 
> Those "default" ipsets are created for each interface if the user scripts are enabled.

Indeed, that seems to be working fine now. Excellent, and quite an elegant solution! Guessing this is a new addition, cause I don't believe I saw those before - even with user script enabled.

>> The problem with using only hash:ip is that if your CIDR list is long, storing it as hash:ip results in a huge list, and with something like the Netflix script, also exceeding maxelem of 64K.
> 
> Yes, I've realized that when replying to your message earlier today. I was trying to use the same type of ipset everywhere which backfired.
> 
> Please try pbr 0.9.1-11 (and remove unnecessary hotplug/reload files again).

Running that now, will report in a few days.

The reload remains commented out in my firewall config, and I see it attempted to remove my (already manually removed) firewall hotplug:

```
* remove_obsolesced_files: unlinking /etc/hotplug.d/firewall/70-pbr failed: No such file or directory.
* remove_obsolesced_files: unlinking /etc/uci-defaults/99-pbr failed: No such file or directory.
```

 It left the iface hotplug, which I figured was intentional, but would be handled by this bit:
```
[ -s /etc/openwrt_release ] && . /etc/openwrt_release
[ "${DISTRIB_RELEASE//19.07}" = "$DISTRIB_RELEASE" ] && exit 0
```
I have nevertheless removed the file entirely, as you requested, so now I should be running as cleanly as can be. If this continues to work well, I will experiment with gradually lowering PROCD delay, possibly removing it entirely, though I suspect at least the latter will likely mean asking for trouble.

Since this wasn't really covered before; what actually *is* the status of IPv6 support in PBR/VPR? The existence of `ipv6_enabled`, and the docs in general, seems to suggest that it's (intended to be) fully supported. However, [this](https://docs.openwrt.melmac.net/vpn-policy-routing/#a-word-about-ipv6-routing), combined with my own experience (and admittedly still somewhat limited knowledge of IPv6), suggests there is rather more to the story.

stangri, 2021-08-29:
> Since this wasn't really covered before; what actually _is_ the status of IPv6 support in PBR/VPR?

Well, I haven't tested it myself, but PBR should be creating IPv6 tables/rules just like the legacy ones, however due to specifics of IPv6 routing, they cannot be enforced without NAT6 which requires additional configuring. I believe mwan3 has the same issue and it's been discussed on the forum, but I don't believe anyone with the IPv6 stack on all their interfaces has ever tested VPR/PBR nor produced a how-to on running it.

My recommendation for now would be to disable IPv6 on the router if you get legacy addresses on all interfaces and need policy-based routing of any kind (either using PBR/VPR or not).

yitzhaq, 2021-09-08:
> Well, I haven't tested it myself, but PBR should be creating IPv6 tables/rules just like the legacy ones, however due to specifics of IPv6 routing, they cannot be enforced without NAT6 which [...]

I don't have IPv6 on all my interfaces, but I do have it on wan (as 6rd) and wireguard0, and could have it on more interfaces if I wanted, though haven't really seen the need since I never got routing quite working. I am using NAT6 (with the script that's been floating around for years), but have never had any success managing to create workable routing for it (at a PBR level). The closest I've gotten was thanks to the aforementioned (now sadly missing) gist linked to in the VPR docs, which brought me one step closer by at least demonstrating a working way to specify a per-interface gateway to use, whereas before it was more of an all-or-nothing type of situation.

> My recommendation for now would be to disable IPv6 on the router if you get legacy addresses on all interfaces and need policy-based routing of any kind (either using PBR/VPR or not).

Sounds to me as though the IPv6 support in VPR/PBR is more hypothetical than proven, then, if you don't yet know of anyone who has actually gotten it working as intended. IMHO this may be worth a mention in the docs; maybe there are others who, like me, have been pulling their hair over the lack of success. It does add tables/rules, as you say, but it's not like they're gonna work out of the box like IPv4 does.

I remember sending you the link to another gist a while back to add to the docs; I still think that would be helpful for anyone attempting to use IPv6 with PBR. The method that's worked best for me has been, rather than disabling IPv6 outright (which I always consider last resort), to strip AAAA in dnsmasq from the domains that I apply policies for (with dnsmasq.ipset), so that I only (forcibly) fall back to IPv4 for the domains where I need specific policy-based routing. The gist I linked provides a helpful example of that, as well as a good explanation of the core problem. These two gists together gave me the setup I use today, which at least is functional, even if not ideal.

>> I have nevertheless removed the file entirely, as you requested, so now I should be running as cleanly as can be.
> 
> Yes, please let me know. So far response to PBR from reddit/forum wasn't what I expected, I don't even know if there are people other than you who are actively testing it, definitely nobody has provided as much feedback.

Happy if my feedback can be of help, rather than just a geek ranting verbosely into the void :o)

> Like I said PBR is very unlikely to be accepted into 19.07, so if I can establish that the PROCD triggers work well for many users on 21.02 and up, I would remove iface hotplug and just leave information about it in the README than include it in the package.

So far the pure PROCD trigger approach has worked very well for me on 19.07.4, I'm happy to say. I'm still gradually lowering the delay, second by second, but I have yet to encounter any problems since cleaning up my setup as you suggested. Very pleased about that - many thanks.

My remaining issue now (apart from the previously discussed A/AAAA issue which I've worked around by simply setting `ipv6_enabled` to 0) is that, while the CIDRs from my user script are now getting added correctly into the appropriate dst_net ipset, the associated route doesn't appear to be taking. I'm guessing that might be because another earlier rule is overriding it, but I haven't yet started diving in to try to figure out why.

So what's Secure Reload? :)
(my subtle way of hinting that a changelog, or at least better commit messages, would be super ;)

yitzhaq, 2021-09-23:
PBR 0.9.2-4 unfortunately fails for me. Specifically, any PREROUTING rule with interface wan stops working, traceroute shows it not getting past the router. OUTPUT rules, or PREROUTING rules with different interfaces than wan (or even switching one of the failing rules to another interface than wan), appear to work. Problem affects both IP- and domain-based rules.

I reverted to 0.9.2-2 and it started working again. Still on 19.07.4.

yitzhaq, 2021-09-23:
Oh, and I should also mention; with 0.9.2-4, I saw error messages for certain domain-based rules that (also) resolve to AAAA, even with `ipv6_enabled` set to 0. So seems there may still be issues with this.

stangri, 2021-09-24:
> PBR 0.9.2-4 unfortunately fails for me. Specifically, any PREROUTING rule with interface wan stops working, traceroute shows it not getting past the router. OUTPUT rules, or PREROUTING rules with [...]

Thanks for letting me know, I'll compare revisions and hopefully will see something that broke PREROUTING. For me the 0.9.2-4 on 21.02.0 works tho.

Secure Reload is inserting the killswitch rule before reload happens and removing it after all the tables and iptables/ipset rules are created -- to prevent leaks during the service reload. It can, of course, backfire if some policies result in some domain names which need to be resolved, that's why it's an option.

stangri, 2021-09-24:
> Oh, and I should also mention; with 0.9.2-4, I saw error messages for certain domain-based rules that (also) resolve to AAAA, even with `ipv6_enabled` set to 0. So seems there may still be issues with this.

Damn, I've implemented the change to be more consistent about obtaining interface's device name in 0.9.2-4 and I guess that functionality wasn't backported to 19.07.

Let me ponder on how to overcome this.

yitzhaq, 2022-01-10:
> Thanks for letting me know, I'll compare revisions and hopefully will see something that broke PREROUTING. For me the 0.9.2-4 on 21.02.0 works tho.

Did you ever figure anything out regarding this? I've finally gotten around to updating to 21.02.0, and pbr (including 0.9.3-5) has the same problem for me there. Downgrading to 0.9.2-2 is once more needed to get around the issue.

yitzhaq, 2022-01-10:
> > > PBR 0.9.2-4 unfortunately fails for me. Specifically, any PREROUTING rule with interface wan stops working, traceroute shows it not getting past the router. OUTPUT rules, or PREROUTING rules with different interfaces than wan (or even switching one of the failing rules to another interface than wan), appear to work. Problem affects both IP- and domain-based rules.
> > > I reverted to 0.9.2-2 and it started working again. Still on 19.07.4.
> > 
> > 
> > Thanks for letting me know, I'll compare revisions and hopefully will see something that broke PREROUTING. For me the 0.9.2-4 on 21.02.0 works tho.
> 
> Did you ever figure anything out regarding this? I've finally gotten around to updating to 21.02.1, and pbr (including 0.9.3-5) has the same problem for me there. Downgrading to 0.9.2-2 is once more needed to get around the issue.

Accidental discovery; when LuCI altered my network configuration to use [the new syntax](https://openwrt.org/releases/21.02/notes-21.02.0#new_network_configuration_syntax_and_boardjson_change) (mostly ifname->device), even 0.9.2-2 started failing in the same manner as later versions. In fact, every version of pbr now fails (I haven't gone back and tested vpr). Only by reverting to the old network configuration syntax, and using 0.9.2-2 (and possibly earlier) was I able to get it working again. It also seems the problem might be affecting tun0 in addition to wan, but it does not appear to be affecting any of the wireguard* interfaces nor the openconnect0 interface.

I hope this can nudge some sort of clue as to what might be going on. I'd really like to get this sorted, and move to a newer and more sustainable syntax, rather than continue with this current makeshift setup any longer than absolutely necessary.

stangri, 2022-01-11:
Sorry for delayed response, I'll have a look at the code. I'm on 21.02.1 with the new (afaik) network config schema and things seem to work.

yitzhaq, 2022-01-12:
No need for apologies. I'm just grateful you still take the time to look into my many and occasionally strange challenges.

One more nugget of information which I hope might help narrow this down; even when lan->wan traffic works (under the particular conditions described earlier), wg->wan traffic still fails in the same manner. That is to say, an endpoint connected through wg gets traffic through to wg interfaces, and to lan and openconnect0, but not to wan or tun0.

I've checked the firewall again, and this is all in the same zone, so nothing should be blocking there. I also believe this used to be working with vpr at some point, without the overall firewall design having changed in any significant manner since then.

And finally, to clarify; the problem only occurs to rules with `option interface 'wan'`, `option dest_addr`, and chain PREROUTING, but **without** `option src_addr`. Rules in PREROUTING that also **include** `option src_addr` seem to work, and rules in OUTPUT also seem to work even without `option src_addr`. Thus this may just be something wonky with my setup somehow, somewhere, but I could swear this used to work at some point..

yitzhaq, 2022-01-12:
After a bit more testing: adding `option src_addr` to the failing PREROUTING rules that lack it makes no difference, unfortunately. So that's more likely a red herring.

stangri, 2022-01-17:
I've compared differences between 0.9.2-2 and the current version and besides cosmetics the only real change is `insert_lock` and `remove_lock`. If you want to remove lines [1050](https://github.com/stangri/source.openwrt.melmac.net/blob/afc5144bf6c771829e003ab1fd4ce44c05f321d7/pbr/files/pbr.init#L1050) and [1121](https://github.com/stangri/source.openwrt.melmac.net/blob/afc5144bf6c771829e003ab1fd4ce44c05f321d7/pbr/files/pbr.init#L1121) from the most current init file, it should be functionally the same as 0.9.2-2.
Can you test current version with those two lines removed/commented out and let me know if that works for you?

stangri, 2022-09-29:
Not sure if you've tried `pbr` on 22.03 and/or are still having this issue, please reopen with the new relevant information if you do.

closed 2022-09-29 by stangri

yitzhaq, 2023-04-06:
> Not sure if you've tried `pbr` on 22.03 and/or are still having this issue, please reopen with the new relevant information if you do.

I finally found some time to look at this. After updating to 22.03, and latest pbr-iptables (for dnsmasq.ipset), the problem still persisted. Eventually I ended up recreating my entire config, bit by bit, from scratch, one rule at a time. After all the rules were back in place, I could no longer reproduce the problem. Given that things are now finally working as intended, and it seems no one else were hit by whatever fringe issue I've been seeing, I don't really feel particularly inclined to spend more time trying to pin down what might have caused this. As an additional relief, this exercise also resolved some other strange routing issues I'd been living with for a while, so I'm now quite the happy camper.

I had a number of options set in the config section which I could find no trace of in the current docs, and which things now seem to be working just fine without. I could probably whip up a diff of that if you're particularly interested (and file a new issue about it), but if not I say we just let this one rest.

Thanks as always for awesome software!