mossdef-org/source.mossdef.org #126: [vpn-policy-routing] Issue: append_src_rules setting is not supported in vpn-policy-routing 0.3.2-6.

Jorman, 2021-02-11:
Hi, after updating I get this error.
My configuration is simple, I've OpenVPN server running on `VPN_SERVER/tun0/10.8.0.1` and Wireguard client running on `TEST_WG_PIA/10.x.x.x` and only one device in my lan is routed to use the WG interface

So, inside the `vpn-policy-routing` configuration file I've this line `option append_src_rules '! -d 10.8.0.0/24'`
But now, with the new app, seems to be incompatible.

My purpose is to reach my lan device, that is routed through the WG interface, when I'm not at home and I make a connection to my VPN_SERVER

I searched a bit, but I don't have found any solution

How I can fix this?

J

stangri, 2021-02-11:
Hello,

`append_src_rules` and `append_dest_rules` have been both replaced with `IGNORE` target. Search the https://docs.openwrt.melmac.net/vpn-policy-routing/ for `option interface 'ignore'` examples.

Policies targeting IGNORE target should be at the top of the policies list.

I will amend the README with the transition guide today or tomorrow.

Jorman, 2021-02-11:
Hi, thank you, I'll try to make some test.

stangri, 2021-02-11:
https://docs.openwrt.melmac.net/vpn-policy-routing/#ignore-requests-replace-append_src_rules

Jorman, 2021-02-12:
Hi, thanks
I made a little test, but maybe I did it wrong, this's what I set up:
```
config policy
	option interface 'wan'
	option name 'Ftp-Rule1'
	option src_addr '10.0.0.100'
	option src_port '20-21'

config policy
	option interface 'wan'
	option src_addr '10.0.0.100'
	option name 'Ftp-Rule2'
	option src_port '6900-7000'

config policy
	option chain 'PREROUTING'
	option name 'Nas'
	option proto 'tcp udp'
	option src_addr '10.0.0.100'
	option interface 'TEST_WG_PIA'

config policy
	option name 'Ignore Local Requests by Destination'
	option dest_addr '10.8.0.0/24'
	option interface 'ignore'

config vpn-policy-routing 'config'
	option verbosity '2'
	option ipv6_enabled '0'
	option strict_enforcement '1'
	option boot_timeout '30'
	list ignored_interface 'vpn_server'
	option src_ipset '0'
	option iptables_rule_option 'append'
	option iprule_enabled '0'
	option webui_enable_column '0'
	option webui_protocol_column '0'
	option webui_chain_column '0'
	option webui_sorting '1'
	option webui_show_ignore_target '1'
	list webui_supported_protocol 'tcp'
	list webui_supported_protocol 'udp'
	list webui_supported_protocol 'tcp udp'
	list webui_supported_protocol 'icmp'
	list webui_supported_protocol 'all'
	list supported_interface 'pia_vpn'
	list supported_interface 'test_wg_pia'
	option enabled '1'
	option dest_ipset '0'
	option resolver_ipset 'dnsmasq.ipset'

config include
	option path '/etc/vpn-policy-routing.netflix.user'
	option enabled '0'

config include
	option path '/etc/vpn-policy-routing.aws.user'
	option enabled '0'
```
I rebooted but won't work.
What do you think?

stangri, 2021-02-12:
Move the `ignore`-target policy to the top.

Jorman, 2021-02-12:
Yep, thank you, that worked

closed 2021-02-12 by Jorman

stangri, 2021-02-12:
Thanks for prompt testing and confirmation, you were also instrumental in the part of the README which deals with migration from old config to the new one.

Jorman, 2021-02-12:
Thank you too for your precious works!