mossdef-org/source.mossdef.org #129: [vpn-policy-routing] Issue: curl should be a dependency

venomousZealot, 2021-03-07:
**Describe the bug**
The example user includes failed with
```
    Error running custom user file '/etc/vpn-policy-routing.netflix.user'
    Error running custom user file '/etc/vpn-policy-routing.aws.user'
```
Installing curl from the software repository solved the problem.

stangri, 2021-03-07:
Thanks for the report! While it would certainly solve the problem, adding dependency on `curl` would dramatically increase the total size of installed package with dependencies for everyone, even if they are not using the custom user files.

I suggest [this](https://github.com/stangri/source.openwrt.melmac.net/commit/35407df3af0a42ae5765a0e89f78e48e88fe4375) as the solution.

stangri, 2021-03-11:
Closing due to lack of activity.

closed 2021-03-11 by stangri