mossdef-org/source.mossdef.org #130: [simple-adblock] Issue: extra_command: not found

Jorman, 2021-04-02:
Hi, during the update I got this error, here the log:
```
BusyBox v1.28.4 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 18.06.4, r7808-ef686b7292
 -----------------------------------------------------
root@OpenWrt:~# opkg-upgrade

Simple OPKG Updater v0.3.5

Done | Updating package lists
Done | Getting upgradable packages list

Packages available for upgrade: 2

+-----+--------------------+---------+---------+
|   # | Package            | Current | Update  |
+-----+--------------------+---------+---------+
|   1 | simple-adblock     | 1.8.7-1 | 1.8.7-3 |
|   2 | vpn-policy-routing | 0.3.3-1 | 0.3.4-1 |
+-----+--------------------+---------+---------+

Proceed with upgrade? (Y/y to proceed) y

.... | Upgrading packages

Upgrading simple-adblock on root from 1.8.7-1 to 1.8.7-3...
Downloading https://raw.githubusercontent.com/stangri/openwrt-repo/master/simple-adblock_1.8.7-3_all.ipk
Stopping service and removing rc.d symlink for simple-adblock
/etc/rc.common: /etc/init.d/simple-adblock: line 12: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 13: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 14: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 15: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 16: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 17: extra_command: not found
WARNING: Some recommended packages are missing, install them by running:
opkg update; opkg --force-overwrite install grep sed coreutils-sort;
Stopping simple-adblock 1.8.7-1... [✓]
/etc/rc.common: /etc/init.d/simple-adblock: line 12: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 13: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 14: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 15: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 16: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 17: extra_command: not found
Syntax: /etc/init.d/simple-adblock [command]

Available commands:
        start   Start the service
        stop    Stop the service
        restart Restart the service
        reload  Reload configuration files (or restart if service does not implement reload)
        enable  Enable service autostart
        disable Disable service autostart

/etc/rc.common: /etc/init.d/simple-adblock: line 12: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 13: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 14: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 15: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 16: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 17: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 12: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 13: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 14: extra_command: not found
/etc/rc.common: /etc/init.d/simple-adblock: line 15: extra_command: not found
[... 69 lines cut ...]
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/arm_cortex-a15_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Package grep (2.26-1) installed in root is up to date.
Unknown package 'sed'.
Package coreutils-sort (8.23-4) installed in root is up to date.
Collected errors:
 * opkg_install_cmd: Cannot install package sed.
root@OpenWrt:~#
```
Do you think is temporary or is related to OpenWrt version?

J

stangri, 2021-04-02:
With the imminent release of 21.02, most recent versions of simple-adblock and vpn-policy-routing in my repo do not support OpenWrt 18.06. Either stay on the version available from the official OpenWrt repo if available, grab an older version from my repo (which is github repository so you can go back in commits and find an older version) or upgrade to 19.07/21.02.

closed 2021-04-02 by stangri

Jorman, 2021-04-03:
Ok, thank you.
I don't know if an upgrade can be a good idea for my R7800, I'll search in forum for that, I know that this involves some extra work, not a big problem, I think, but is better to pay attention.
Do you know a good way to upgrade and recover the configuration without risk to brick or lost all?