mossdef-org/source.mossdef.org #131: dnsmasq.ipset breaks luci-app-simple-adblock

wengole, 2021-04-10:
Hi,

When I got into advanced tab in Luci for Simple Adblock, enable dnsmasq IP Set and hit save and apply, I am presented with the following error in the browser and am no longer able to manage Simple AdBlock

```/usr/lib/lua/luci/i18n.lua:48: bad argument #1 to 'format' (string expected, got nil)
stack traceback:
	[C]: in function 'format'
	/usr/lib/lua/luci/i18n.lua:48: in function 'translatef'
	/usr/lib/lua/luci/model/cbi/simple-adblock.lua:139: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1340: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1023: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>```

stangri, 2021-04-10:
What a great catch, that's a big one! I pushed a fixed IPK to my repo so you can just reinstall the luci app or if you don't have my repo added, you can download the updated IPK from here: https://dev.melmac.net/repo/luci-app-simple-adblock_1.8.7-3_all.ipk. I believe this newer version will work with the `simple-adblock` 1.8.4 you have installed.

If you're on OpenWrt 19.07 or later, I would also recommend you update the principal `simple-adblock` package to 1.8.7-3.

stangri, 2021-04-10:
Oh, please let me know when you can test this, so I can merge related PRs.

wengole, 2021-04-11:
That fixed it, thanks

stangri, 2021-04-11:
Thanks for reporting this!

closed 2021-04-11 by stangri