mossdef-org/source.mossdef.org #132: [vpn-policy-routing] Issue:cannot be restarted automatically when the WAN gateway is changed

happy201807, 2021-04-29:
**Describe the bug**
WAN is the IP obtained by pppoe. Once the WAN dials again, the vpn policy routing cannot automatically restart when the gateway changes.

**Your configs**

1. ```/etc/config/dhcp```

   ```sh

   ```

2. ```/etc/config/network```

  ```
 config interface 'wan'
	option proto 'pppoe'

 config interface 'wg0'
	option proto 'wireguard'
	option listen_port '51820'
	list addresses '10.49.0.4/32'
	option private_key '...'
	option delegate '0'

config wireguard_wg0
	option public_key '...'
	option route_allowed_ips '1'
	list allowed_ips '0.0.0.0/0'
	option endpoint_host '...'
	option endpoint_port '51820'
	option persistent_keepalive '25'
	option preshared_key '...' 
```

3. ```/etc/config/firewall```

```
config rule
	option target 'ACCEPT'
	option src 'wan'
	option proto 'udp'
	option name 'Wireguard_VPN'
	option family 'ipv4'
	option dest_port '51820'

config zone
	option name 'wg-vpn'
	option input 'ACCEPT'
	option forward 'ACCEPT'
	option output 'ACCEPT'
	option masq '1'
	option device 'wg0'
	option network 'wg0'

config forwarding 'wg_wan'
	option src 'wg-vpn'
	option dest 'wan'

config forwarding 'wg_lan'
	option src 'wg-vpn'
	option dest 'lan'

config forwarding
	option src 'lan'
	option dest 'wg-vpn'
   ```

4. ```/etc/config/vpn-policy-routing```

```
[... 1390 lines cut ...]
		'name:string' \
		'enabled:bool:0' \
		'interface:network' \
		'proto:or(string)' \
		'chain:or("", "PREROUTING", "FORWARD", "INPUT", "OUTPUT")' \
		'src_addr:list(neg(or(host,network,macaddr)))' \
		'src_port:list(neg(or(portrange, string)))' \
		'dest_addr:list(neg(host))' \
		'dest_port:list(neg(or(portrange, string)))'
}

# shellcheck disable=SC2120
validate_include() {
	uci_validate_section "${packageName}" include "${1}" \
		'path:string' \
		'enabled:bool:0'
}

```

2. Output of ```/etc/init.d/vpn-policy-routing support```:

   ```sh

   ```

stangri, 2021-04-29:
Which version of OpenWrt?

Also, please do not paste the content of the /etc/init.d/vpn-policy-routing, paste the output of the script running with parameters mentioned in the issue report template.

happy201807, 2021-04-30:
openwrt-21.02 branch 
Upgrade to 0.34-4 seems to have solved the problem today.

closed 2021-05-01 by happy201807