mossdef-org/source.mossdef.org #133: [vpn-policy-routing] Wish: I2P Support

FZXizzbKv7jg, 2021-05-06:
Support I2P provided by package i2pd.

stangri, 2021-11-15:
I have zero experience with that protocol, and I don't have the time to experiment with it right now. If you can provide the following information I can have a look at supporting it:

1. How to identify i2p interface.
2. How to obtain i2p interface gateway.

FZXizzbKv7jg, 2022-07-07:
service name: i2pd

config file /etc/i2pd/i2pd.conf:

## Where to write pidfile (default: i2pd.pid, not used in Windows)                                                                      
# pidfile = /run/i2pd.pid                                                                                                                                          

## Network interface to bind to                                                                                                                    
## Updates address4/6 options if they are not set                                                                                                  
ifname = vpniface

## External IPv4 or IPv6 address to listen for connections                                   
## By default i2pd sets IP automatically                                                                                                
## Sets published NTCP2v4/SSUv4 address to 'host' value if nat = true                                                                   
## Sets published NTCP2v6/SSUv6 address to 'host' value if ipv4 = false                      
# host = 1.2.3.4   

## Port to listen for connections                                                                                                                  
## By default i2pd picks random port. You MUST pick a random number too,                                                                           
## don't just uncomment this                                                                                                                       
# port = 4567  

[httpproxy]                                                                                              
## Uncomment and set to 'false' to disable HTTP Proxy                            
# enabled = true                                                                                                                        
## Address and port service will listen on                                                   
address = 192.168.1.1                                                            
port = 4444 

tunnels config:
/etc/i2pd/tunnels.conf

FZXizzbKv7jg, 2022-07-07:
service name: i2pd

config /etc/i2pd/i2pd.conf:

```
## Where to write pidfile (default: i2pd.pid, not used in Windows)
# pidfile = /run/i2pd.pid
                                                                                                                                         
## Network interface to bind to                        
## Updates address4/6 options if they are not set                                                                                                             
ifname = vpniface

## External IPv4 or IPv6 address to listen for connections  
## By default i2pd sets IP automatically                                                                                                                 
# host = 1.2.3.4   

## Port to listen for connections                                      
## By default i2pd picks random port. You MUST pick a random number too,    
## don't just uncomment this                                         
# port = 4567  

[httpproxy]                                                                    
## Uncomment and set to 'false' to disable HTTP Proxy                       
# enabled = true                                                         
## Address and port service will listen on                                     
address = 192.168.1.1                                                       
port = 4444    
```

tunnels config:
/etc/i2pd/tunnels.conf

stangri, 2022-07-07:
So on your system there's an interface called 'vpniface'?

stangri, 2022-09-29:
> The implementation for i2p should be similar to tor's. Suffix is .i2p .

So it's a proxy where you can send HTTP/HTTPS requests?

FZXizzbKv7jg, 2022-09-30:
Yes right.