mossdef-org/source.mossdef.org #137: [vpn-policy-routing] Issue: `icmp`-based policy gives spurious warning (probably because there is no report port set)

posita, 2021-09-19:
**Describe the bug**
A clear and concise description of what the bug is.

**Your configs**

1. ```/etc/config/vpn-policy-routing```

```
# …
config policy
	option name 'microsoft.com Skype/Teams audio/video'
	option dest_addr '13.107.64.0/18 52.112.0.0/14 52.120.0.0/14'
	option dest_port '3478-3481'
	option proto 'udp'
	option interface 'wan'

config policy
	option name 'microsoft.com ICMP'
	option dest_addr '13.107.64.0/18 52.112.0.0/14 52.120.0.0/14'
	option proto 'icmp'
	option interface 'wan'
# …
```

IP ranges were copied from [here](https://docs.microsoft.com/en-us/microsoft-365/enterprise/urls-and-ip-address-ranges?view=o365-worldwide) ([motivation](https://docs.microsoft.com/en-us/microsoft-365/enterprise/microsoft-365-vpn-implement-split-tunnel?view=o365-worldwide#implement-vpn-split-tunneling)).

**To Reproduce**

``` sh
$ /etc/init.d/vpn-policy-routing restart
…
WARNING: Please unset 'proto' or set 'proto' to 'all' for policy: 'microsoft.com ICMP', destination: '13.107.64.0/18'
WARNING: Please unset 'proto' or set 'proto' to 'all' for policy: 'microsoft.com ICMP', destination: '52.112.0.0/14'
WARNING: Please unset 'proto' or set 'proto' to 'all' for policy: 'microsoft.com ICMP', destination: '52.120.0.0/14'
```

I'm pretty sure the warning is coming from [here](https://github.com/stangri/source.openwrt.melmac.net/blob/9b49166a4f9f31236e116148aab8933997c46445/vpn-policy-routing/files/vpn-policy-routing.init#L557-L566), since the rule has a `proto` but doesn't have a local address, local port, or remote port.

**Expected behavior**

ICMP packets don't have port numbers, so I _**think**_ the above rule should be warning-free.

stangri, 2021-09-20:
Hey Matt, thanks for another quality report. I agree it shouldn't be reported as a warning. Let me sleep on it to figure out how to best address it.

stangri, 2021-09-22:
If you can/want to switch to `pbr` for now, it should not be reporting this policy with the warning: https://dev.melmac.net/repo/pbr_0.9.2-4_all.ipk and https://dev.melmac.net/repo/luci-app-pbr_0.9.2-4_all.ipk.

posita, 2021-09-24:
Thanks! What's `pbr`? (I'm having trouble finding information about it, aside from [this](https://openwrt.org/docs/guide-user/network/routing/pbr_netifd), which looks pretty spartan.) Is `pbr` meant to replace/supersede `vpn-policy-routing` and/or `vpnbypass`?

stangri, 2021-11-15:
pbr is the new version of vpn-policy-routing, but due to a dramatic change on how dnsmasq ipsets are handled, I've forked it into a separate package.

Besides the file/section name as of now, it's config-compatible with vpn-policy-routing.

closed 2021-11-15 by stangri

posita, 2021-12-10:
Are there docs available for handling migrations from `vpn-policy-routing`? I noticed https://docs.openwrt.melmac.net/pbr/ is absent (referenced [here](https://github.com/stangri/source.openwrt.melmac.net/tree/master/pbr/files)).

posita, 2021-12-10:
Also, does that mean `vpn-policy-routing` is deprecated?

stangri, 2021-12-11:
> Are there docs available for migrating from `vpn-policy-routing`? I noticed https://docs.openwrt.melmac.net/pbr/ is absent (referenced [here](https://github.com/stangri/source.openwrt.melmac.net/tree/master/pbr/files)).

Not yet, I'm hoping to post the README some time mid/late December.

> Also, does that mean `vpn-policy-routing` is deprecated? [This](https://openwrt.org/docs/guide-user/network/routing/pbr#vpn_policy_routing) suggests it's not (i.e., it still has a purpose). If that's the case, should this issue be reopened (and fixed)?

Not deprecated until the new version is in the official repo.