mossdef-org/source.mossdef.org #139: [vpn-policy-routing] Wish: Only reload, when needed on monitored devices (e.g. really a new IP)

rkorn86, 2021-09-29:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when DHCP renew of wan interface results in same ip, but VPR reloads. My ISP' lease time is 1h (but i get the same CG-NATted IP for months), so basically every hour VPR reloads and my vpn routed connections drop.

**Describe the solution you'd like**
Please check if the new IP is same like the old one and dont reload VPR if they are.

**Describe alternatives you've considered**
There is no alternative :-(

**Additional context**
`- NONE -`

stangri, 2021-11-15:
Thank you for the suggestion, I'll keep it in mind when I have time to work on it. Since service already stores json data upon successful run, I might as well store each interface IP address and only reload routing tables if it has changed.

stangri, 2022-09-29:
I've gone back and forth on this and here's where the problem is for clean/fast reloads of the interfaces.

I've seen logs of some users with an OpenVPN tunnels which do not come up automatically on start, so WAN and let's say WG interfaces come up and when the OpenVPN is brought up, the interface gets sandwiched between the other two. This requires a full restart of policy-based routing, as it needs to re-create all tables and rules with the proper markings.

So while a fast single interface reload was supported for a while, I've disabled it in the latest version of `pbr` for the reason above. I'd like to bring it back, but I lack the facility/time to test all different variations of interface updates to ensure the consistent user experience.

closed 2022-09-29 by stangri