mossdef-org/source.mossdef.org #140: [vpn-policy-routing] Issue: Failed to load kernel modules

cr0wdelex, 2021-10-01:
**Describe the bug**
A clear and concise description of what the bug is.

**Your configs**
1.  Info:
`
OpenWrt` SNAPSHOT r17631-9c3b1d5563 / LuCI Master git-21.271.74796-7c943a1
vpn-policy-routing 0.3.5-3
`
2. ```/etc/config/network```

   ```config` interface 'WARP'
	option proto 'wireguard'
	option delegate '0'
	option private_key 'key'
	list addresses '172.16.0.2/32'
	option listen_port 'port'

config wireguard_WARP
	option description 'Cloudfare'
	option public_key 'key'
	option endpoint_host 'engage.cloudflareclient.com'
	option endpoint_port '2408'
	option persistent_keepalive '25'
	list allowed_ips '0.0.0.0/0'
   ```

3. ```/etc/config/vpn-policy-routing```

   ```config vpn-policy-routing 'config'
	option strict_enforcement '1'
	option src_ipset '0'
	option dest_ipset '0'
	option iptables_rule_option 'append'
	option procd_reload_delay '1'
	option webui_sorting '1'
	list webui_supported_protocol 'tcp'
	list webui_supported_protocol 'udp'
	list webui_supported_protocol 'tcp udp'
	list webui_supported_protocol 'icmp'
	list webui_supported_protocol 'all'
	option enabled '1'
	list ignored_interface 'WG'
	list ignored_interface 'tun0'
	option boot_timeout '60'
	option webui_enable_column '1'
	option webui_protocol_column '1'
	option webui_chain_column '1'
	option webui_show_ignore_target '1'
	option resolver_ipset 'dnsmasq.ipset'
	option ipv6_enabled '0'
	option verbosity '2'

config include
	option path '/etc/vpn-policy-routing.netflix.user'
	option enabled '0'

config include
	option path '/etc/vpn-policy-routing.aws.user'
	option enabled '0'

config include
	option path '/etc/warp/VPR-Facebook.user'

config include
	option path '/etc/warp/VPR-Speedtest.user'

config include
	option path '/etc/warp/VPR-Twitch.user'

config include
	option path '/etc/warp/VPR-Taobao.user'
   ```

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest version 

**Expected behavior**
A clear and concise description of what you expected to happen.

**VPN Policy Routing run-time information**

1. Output of ```/etc/init.d/vpn-policy-routing reload```:
```
Running /etc/warp/VPR-Facebook.user ipset v7.6: Error in line 6: Hash is full, cannot add more elements
[✗]
Running /etc/warp/VPR-Speedtest.user ipset v7.6: Error in line 1: Hash is full, cannot add more elements
[✗]
Running /etc/warp/VPR-Twitch.user ipset v7.6: Error in line 1: Hash is full, can not add more elements
[✗]
Running /etc/warp/VPR-Taobao.user ipset v7.6: Error in line 1: Hash is full, can not add more elements
[✗]
vpn-policy-routing 0.3.5-3 monitoring interfaces: wan WARP [✓]
vpn-policy-routing 0.3.5-3 started with gateways:
wan/pppoe-wan/IP [✓]
WARP/IP
ERROR: Failed to load kernel modules
ERROR: Error running custom user file '/etc/warp/VPR-Facebook.user'!
ERROR: Error running custom user file '/etc/warp/VPR-Speedtest.user'!
ERROR: Error running custom user file '/etc/warp/VPR-Twitch.user'!
ERROR: Error running custom user file '/etc/warp/VPR-Taobao.user'!
```

stangri, 2021-10-01:
> ERROR: Failed to load kernel modules

Have you [installed required packages](https://docs.openwrt.melmac.net/vpn-policy-routing/#requirements)?

cr0wdelex, 2021-10-01:
Yes, They are all installed.
```
root@MiRouter:~# opkg install ipset resolveip ip-full kmod-ipt-ipset iptables
Package ipset (7.6-1) installed in root is up to date.
Package resolveip (2) installed in root is up to date.
Package ip-full (5.14.0-1) installed in root is up to date.
Package kmod-ipt-ipset (5.4.145-1) installed in root is up to date.
Package iptables (1.8.7-1) installed in root is up to date.
root@MiRouter:~# opkg install dnsmasq-full
Package dnsmasq-full (2.86-2) installed in root is up to date.
```

stangri, 2021-10-01:
Can you please run:
```sh
modprobe xt_set
modprobe ip_set
modprobe ip_set_hash_ip
```

cr0wdelex, 2021-10-02:
Here the result:
```
root@MiRouter:~# modprobe xt_set
modprobe: FATAL: Module xt_set not found in directory /lib/modules/5.4.145
root@MiRouter:~# modprobe ip_set
modprobe: FATAL: Module ip_set not found in directory /lib/modules/5.4.145
root@MiRouter:~# modprobe ip_set_hash_ip
modprobe: FATAL: Module ip_set_hash_ip not found in directory /lib/modules/5.4.145
```

I have checked ```/lib/modules/5.4.145``` and these files are existed:
```
ip_set.ko
ip_set_hash_ip.ko
xt_set.ko
```

stangri, 2021-10-02:
Thanks for checking, if the files are there, but modprobe complains it's definitely a problem outside of scope of my package.

cr0wdelex, 2021-10-02:
1 more thing, I disabled all the route and it still has Faled to load kernal modules.
`
root@MiRouter:~# /etc/init.d/vpn-policy-routing reload
Creating table 'wan/pppoe-wan/IP' Flush terminated
[✓]
Creating table 'WARP/172.16.0.2' Flush terminated
[✓]
vpn-policy-routing 0.3.5-3 monitoring interfaces: wan WARP [✓]
vpn-policy-routing 0.3.5-3 started with gateways:
wan/pppoe-wan/IP [✓]
WARP/172.16.0.2
ERROR: Failed to load kernel modules

`

stangri, 2021-10-03:
> And about the modprobe, chmod permission for `/lib/modules/5.4.145`, the checkings are successfully

So the permissions on `/lib/modules/5.4.145` were wrong in the snapshots image? You may want to report it on the openwrt/openwrt repo then.

closed 2021-10-03 by stangri