mossdef-org/source.mossdef.org #143: [vpn-policy-routing] Issue: lede 17.01.4 error no work

predators46, 2021-11-02:
`Service Errors
Failed to set up 'wan/usb0/192.168.42.129'
Failed to set up 'l2tp/l2tp-l2tp/192.168.18.1 \033[0;32m[\xe2\x9c\x93]\033[0m'
failed to set up any gateway!`

stangri, 2021-11-02:
Try installing the lower version from 19.07 repository.

predators46, 2021-11-02:
> Try installing the lower version from 19.07 repository.

Thank you. i also tried l2tp and vpn policy routing on openwrt 21.02.1 here i want mobile legends game app not to use l2tp and i added remote port udp and tcp but i still get too high ping

`
config vpn-policy-routing 'config'
	option verbosity '2'
	option strict_enforcement '1'
	option src_ipset '0'
	option dest_ipset '0'
	option resolver_ipset 'dnsmasq.ipset'
	option ipv6_enabled '0'
	option boot_timeout '30'
	option iptables_rule_option 'append'
	option procd_reload_delay '1'
	option webui_sorting '1'
	list webui_supported_protocol 'tcp'
	list webui_supported_protocol 'udp'
	list webui_supported_protocol 'tcp udp'
	list webui_supported_protocol 'icmp'
	list webui_supported_protocol 'all'
	option webui_enable_column '1'
	option webui_protocol_column '1'
	option webui_chain_column '1'
	option webui_show_ignore_target '1'
	option enabled '1'

config include
	option path '/etc/vpn-policy-routing.netflix.user'
	option enabled '0'

config include
	option path '/etc/vpn-policy-routing.aws.user'
	option enabled '0'

config policy
	option name 'twitter'
	option dest_addr 'twitter.com'
	option interface 'wan'

config policy
	option interface 'wan'
	option name 'mobile legends'
	option dest_port '2702 3702 5517 5520-5529 8001 9000-9010 9992 10003 30000-30300'
	option proto 'udp'

config policy
	option interface 'wan'
	option name 'speedtest'
	option dest_addr 'speedtest.net'

config policy
	option interface 'wan'
	option name 'Ml'
	option dest_port '4001-4009,5000-5221,5224-5241,5243-5508,5551-5559,5601-5700'
	option proto 'udp'

config policy
	option interface 'wan'
	option name 'Ml2'
	option dest_port '5000-5221,5224-5227,5229-5241,5243-5508,5551-5559,5601-5700,9001,9443'
	option proto 'tcp'

config policy
	option interface 'wan'
	option name 'Ml3'
	option dest_port '5520-5529,10003,30000-30300'
	option proto 'tcp'`

stangri, 2021-11-15:
Impossible to draw any conclusions based on the single config file. I recommend you seek help on the forum.

closed 2021-11-15 by stangri