mossdef-org/source.mossdef.org #145: [vpn-policy-routing] Issue: wireguard IPv6 default route is not setup

TheLinuxGuy, 2021-11-28:
**Describe the bug**
IPv6 routing in vpn-policy-routing with wireguard is not working properly. I have noticed that it seems that 'vpn-policy-routing' takes ownership of how openwrt/wireguard tunnel operates on IPv6. Specifically, if ipv6 is disabled in VPR then my tunnel that was previously working just fine and getting IPv6 routes automatically no longer gets anything.

Interestingly, enabling IPv6 in VPR does allow a route to show up but this route is wrong and is not 'default' - my ipv6 is broken unless I do this trick `ip -6 route add default dev vpsgw table vpsgw` 

This is very similar to the FAQ posted workaround but my setup is complex. I have VLANs - each vlan has a route lookup table - in this case VPR (202 for ovpn, 203 for wireguard). I am migrating to vpn-policy-routing from a manually configured route tables for specific vlans.

Perfect example:
```
root@meow:~# ip -6 route show table 203
fd25:8911:3113:9525::10 dev vpsgw proto static metric 1024 pref medium
root@meow:~# ping6 -I vpsgw google.com
PING google.com (2607:f8b0:4008:804::200e): 56 data bytes
ping6: sendto: Permission denied
root@meow:~# ip -6 route add default dev vpsgw table vpsgw
root@meow:~# ping6 -I vpsgw google.com
PING google.com (2607:f8b0:4008:80a::200e): 56 data bytes
64 bytes from 2607:f8b0:4008:80a::200e: seq=0 ttl=113 time=259.802 ms

```

**Your configs**

1. ```/etc/config/dhcp```

   ```sh
config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option expandhosts '1'
        option nonegcache '0'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'
        option ednspacket_max '1232'
        option domain 'gfm'
        option local '/gfm/'
        option noresolv '1'
        list server '127.0.0.1#5453'
        list server '0::1#5453'

config dhcp 'lan'
        option interface 'lan'
        option leasetime '12h'
        option dhcpv4 'server'
        option dhcpv6 'server'
        option ra 'server'
        option ra_slaac '1'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'
        option ra_default '1'
        option start '20'
        option limit '50'
        option ra_management '1'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
[... 575 lines cut ...]
-A VPR_PREROUTING -s 192.168.32.21/32 -m comment --comment nzb -c 0 0 -g VPR_MARK0x030000
============================================================
Mangle IPv6 Table: PREROUTING
-N VPR_PREROUTING
============================================================
Mangle IP Table MARK Chain: VPR_MARK0x010000
-N VPR_MARK0x010000
-A VPR_MARK0x010000 -c 0 0 -j MARK --set-xmark 0x10000/0xff0000
-A VPR_MARK0x010000 -c 0 0 -j RETURN
============================================================
Mangle IP Table MARK Chain: VPR_MARK0x020000
-N VPR_MARK0x020000
-A VPR_MARK0x020000 -c 0 0 -j MARK --set-xmark 0x20000/0xff0000
-A VPR_MARK0x020000 -c 0 0 -j RETURN
============================================================
Mangle IP Table MARK Chain: VPR_MARK0x030000
-N VPR_MARK0x030000
-A VPR_MARK0x030000 -c 0 0 -j MARK --set-xmark 0x30000/0xff0000
-A VPR_MARK0x030000 -c 0 0 -j RETURN
============================================================
Current ipsets
============================================================
Your support details have been logged to '/var/vpn-policy-routing-support'. [✓]

   ```

TheLinuxGuy, 2021-11-28:
Sorry about that formatting. I have an ansible playbook that I can share privately with you; as long as you have an openvpn tunnel and a wireguard host - you should be able to reproduce my environment within minutes.

Feel free to DM me and I can zip it after removing my secrets.

TheLinuxGuy, 2021-11-28:
I tried to workaround this bug by adding a hotplug script to add the route and as seen in the logs; vpn-policy-routing is overriding my route tables and wiping it out.

```
Sun Nov 28 17:25:05 2021 user.notice nat6: Done setting up nat6 for zone="vpsgw" on devices:
Sun Nov 28 17:25:05 2021 user.notice nat6: Restarting ODHCP v6 to fix clients not getting V6 addresses
Sun Nov 28 17:25:05 2021 **user.notice hotplug: gfm Device: vpsgw / Action: ifup**
Sun Nov 28 17:25:06 2021 user.notice vpn-policy-routing [19517]: Creating table 'wan/eth1/192.168.12.1/2607:fb90:4021:8890:b5b4:4941:0:733/128 2607:fb90:4021:8890:4c7a:edff:fe09:adf5/64 fe80::4c7a:edff:fe09:adf5/64'
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: Creating table 'sharktun/tun0/10.8.8.8/fe80::44d7:c399:2c77:5707/64' [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: Creating table 'vpsgw/10.1.1.10/fd25:8911:3113:9525::10/128' [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: Routing 'blank' via ignore [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: Routing 'qbit' via sharktun [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: Routing 'sonar' via vpsgw [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: Routing 'nzb-search' via vpsgw [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: Routing 'radar' via vpsgw [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: Routing 'localnets' via ignore [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: Routing 'nzb' via vpsgw [✓]
Sun Nov 28 17:25:07 2021 user.notice vpn-policy-routing [19517]: service monitoring interfaces: wan sharktun vpsgw [✓]

```
my script

```
root@meow:/etc/hotplug.d/iface# cat 99-my-action
[ "${ACTION}" = "ifup" -a "${DEVICE}" = "vpsgw" ] && {
    logger -t hotplug "gfm Device: ${DEVICE} / Action: ${ACTION}"
    ip -6 route add default dev vpsgw table vpsgw
}
[ "${ACTION}" = "ifdown" -a "${DEVICE}" = "vpsgw" ] && {
    logger -t hotplug "gfm Device: ${DEVICE} / Action: ${ACTION}"
    ip -6 route add default dev vpsgw table vpsgw
}
```

I guess I could try to workaround even that bug by adding a wait timer - but really we should fix the bug in this package.

TheLinuxGuy, 2021-11-28:
I came to learn that even with a hotplug script, if I modify the configuration via the web interface the tables get reset again upon save - and its a signal that hotplug isn't catching. 

I'm not super familiar with how this works and how to make sure everytime 'vpn-policy-routing [19517]: Creating table' happens the hotplug gets called?

the hotplug:
```
# github.com/TheLinuxGuy - workaround https://github.com/stangri/source.openwrt.melmac.net/issues/145

[ "${ACTION}" = "ifup" -a "${DEVICE}" = "vpsgw" ] && {
    logger -t hotplug "Device: ${DEVICE} / Action: ${ACTION}"
    logger -t hotplug "thelinuxguy: working around IPv6 routing bug #145. Waiting 5 seconds."
    sleep 5
    ip -6 route add default dev vpsgw table vpsgw
    logger -t hotplug "thelinuxguy: route added."; 
} 
[ "${ACTION}" = "ifupdate" -a "${DEVICE}" = "vpsgw" ] && {
    logger -t hotplug "Device: ${DEVICE} / Action: ${ACTION}"
    logger -t hotplug "thelinuxguy: working around IPv6 routing bug #145. Waiting 5 seconds."
    sleep 5
    ip -6 route add default dev vpsgw table vpsgw
    logger -t hotplug "thelinuxguy: route added."; 

[ "${ACTION}" = "ifdown" -a "${DEVICE}" = "vpsgw" ] && {
    logger -t hotplug "Device: ${DEVICE} / Action: ${ACTION}"
    logger -t hotplug "thelinuxguy: cleaning up IPv6 default route. vpn-policy-routing bug #145. "
    ip -6 route add default dev vpsgw table vpsgw
} 
```

TheLinuxGuy, 2021-11-29:
Fixed this myself.

PR to fix for everyone: https://github.com/stangri/source.openwrt.melmac.net/pull/146

stangri, 2021-11-29:
Your configs are a contradicting mix of the manual policies configs and use of VPR. I'd recommend you clean out unnecessary network/firewall entries and retest.

BryanJacobs, 2022-03-24:
I have a cleaner config - no manual `ip rule` policies, only VPR. When I use vpn-policy-routing I don't get IPv6 connectivity - the router returns with ICMP unreachable for all outbound v6 traffic. When I add a "custom user file include" whose sole content is the line `ip -6 route add default dev <myvpnif> table <myvpnif>`, I am able to route packets over the tunnel.

I'd recommend adding either that line by default (my preference...) or a facility to let users add to what goes into the automatically created route tables without resorting to scripts.

stangri, 2023-01-14:
Please create a new issue if this problem still exists in `pbr`.

closed 2023-01-14 by stangri