mossdef-org/source.mossdef.org #147: [vpn-policy-routing] Wish: Softether support

tew42, 2021-12-01:
**Is your feature request related to a problem? Please describe.**
Due to firewall limitations, my VPN server at work is set up as a Softether VPN Server (offering unique firewall-traversing capabilities, not to mention much higher performance than OpenVPN). Working connections require use of either the Softether VPN client (or SSTP).

**Describe the solution you'd like**
I'd love to be able to use vpn-policy-routing for my OpenWRT Softether VPN connection
https://openwrt.org/packages/pkgdata/softethervpn5-libs
https://openwrt.org/packages/pkgdata/softethervpn5-client

**Describe alternatives you've considered**
Alternatively, support for SSTP in vpn-policy-routing could be helpful, but I've had trouble getting SSTP to work on OpenWRT.

**Additional context**
Going off what you asked in the I2P wish - 
Softether sets up a new network device, with a name always beginning with "vpn_"
The corresponding network interface has to be created manually, so the name for that can be freely chosen
In the other thread you also asked about how to obtain the gateway - I'm not entirely sure which gateway you're referring to, and how I could get that info

Thanks!

stangri, 2021-12-01:
Please confirm if you have a Softether tunnel running on your OpenWrt installation and which version of OpenWrt is it.

tew42, 2021-12-02:
OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.301.66596-f28aaa3

Yes, I have a softether tunnel up and running, and I can confirm it working for static routes to predefined IPs (although I'm having some trouble getting default routing configured)

stangri, 2021-12-02:
Can you post or email me the output of ifconfig?

tew42, 2021-12-02:
Sure thing! Some light redaction which I don't think will impact the information content.

ifconfig
for softether, the vpn_ part is fixed, the "soft0" is the customizable bit; the corresponding gateway in my case is 192.168.192.1 btw
```
br-lan    Link encap:Ethernet  HWaddr xxxx
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: xxxx/64 Scope:Link
          inet6 addr: xxxx/60 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:424905 errors:0 dropped:10 overruns:0 frame:0
          TX packets:1101414 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:55379951 (52.8 MiB)  TX bytes:1293393871 (1.2 GiB)

eth0      Link encap:Ethernet  HWaddr xxxx
          inet6 addr: xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10911052 errors:0 dropped:0 overruns:41 frame:0
          TX packets:910975 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1943573635 (1.8 GiB)  TX bytes:558199733 (532.3 MiB)
          Interrupt:4

eth0.1    Link encap:Ethernet  HWaddr xxxx
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:112351 errors:0 dropped:0 overruns:0 frame:0
          TX packets:375488 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6120232 (5.8 MiB)  TX bytes:477659678 (455.5 MiB)

eth0.2    Link encap:Ethernet  HWaddr xxxx
          inet addr:PUBLIC IP  Bcast:PUBLIC IP  Mask:255.255.255.0
          inet6 addr: xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10570894 errors:0 dropped:0 overruns:0 frame:0
          TX packets:523774 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1725906815 (1.6 GiB)  TX bytes:74344070 (70.8 MiB)

ifb4eth0.2 Link encap:Ethernet  HWaddr xxxx
          inet6 addr: xxxx/64 Scope:Link
          UP BROADCAST RUNNING NOARP  MTU:1500  Metric:1
          RX packets:10567181 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10567181 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:32
          RX bytes:1873218274 (1.7 GiB)  TX bytes:1873218274 (1.7 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:53332 errors:0 dropped:0 overruns:0 frame:0
          TX packets:53332 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6510232 (6.2 MiB)  TX bytes:6510232 (6.2 MiB)

vpn_soft0 Link encap:Ethernet  HWaddr xxxx
          inet addr:192.168.192.10  Bcast:192.168.192.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:40834 errors:0 dropped:0 overruns:0 frame:0
          TX packets:20515 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:10400289 (9.9 MiB)  TX bytes:1535203 (1.4 MiB)

wlan0     Link encap:Ethernet  HWaddr xxxx
          inet6 addr: xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:708544 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1064062 errors:0 dropped:0 overruns:0 carrier:0
[... 75 lines cut ...]
        "notify_proto":{}
        "remove":{}
        "set_data":{}
'network.interface.wan6' @1b81f91f
        "up":{}
        "down":{}
        "renew":{}
        "status":{}
        "prepare":{}
        "dump":{}
        "add_device":{"name":"String","link-ext":"Boolean","vlan":"Array"}
        "remove_device":{"name":"String","link-ext":"Boolean","vlan":"Array"}
        "notify_proto":{}
        "remove":{}
        "set_data":{}
'network.rrdns' @7ecaeae7
        "lookup":{"addrs":"Array","timeout":"Integer","server":"String","port":"(unknown)","limit":"Integer"}
'network.wireless' @3fc32dca
        "up":{}
        "down":{}
        "reconf":{}
        "status":{}
        "notify":{}
        "get_validate":{}
```

stangri, 2021-12-03:
Thank you, this is very promissing. What do you get for `ubus call network.interface.VPN status` and also this block of code:

```sh
. /lib/functions.sh
iface='VPN'
network_get_device dev "$iface"
network_get_gateway gw "$iface"
echo "$iface - $dev - $gw"
ip -4 a list dev "$dev"
```

tew42, 2021-12-03:
VPN status (the two routes are manually configured static ones)
```
root@OpenWrt:~# ubus call network.interface.VPN status
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 36615,
        "l3_device": "vpn_soft0",
        "proto": "dhcp",
        "device": "vpn_soft0",
        "metric": 0,
        "dns_metric": 0,
        "delegation": false,
        "ipv4-address": [
                {
                        "address": "192.168.192.10",
                        "mask": 24
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [

        ],
        "route": [
                {
                        "target": "xxxx",
                        "mask": xx,
                        "nexthop": "192.168.192.1",
                        "source": "0.0.0.0/0"
                },
                {
                        "target": "xxxx",
                        "mask": xx,
                        "nexthop": "192.168.192.1",
                        "source": "0.0.0.0/0"
                }
        ],
        "dns-server": [
                "1.1.1.1",
                "8.8.8.8"
        ],
        "dns-search": [

        ],
        "neighbors": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [
                        {
                                "target": "0.0.0.0",
                                "mask": 0,
                                "nexthop": "192.168.192.1",
                                "source": "192.168.192.10/32"
                        }
                ],
                "dns-server": [

                ],
                "dns-search": [

                ],
                "neighbors": [

                ]
        },
        "data": {
                "leasetime": 43200
        }
}

```

tew42, 2021-12-03:
The second block of code needed lib\functions\network.sh I believe? Seems that the gateway retrieval fails
```
VPN - vpn_soft0 -
24: vpn_soft0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    inet 192.168.192.10/24 brd 192.168.192.255 scope global vpn_soft0
       valid_lft forever preferred_lft forever
```

stangri, 2021-12-03:
Can you post/email your `/etc/config/network`?

tew42, 2021-12-03:
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'xxxx'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config device
	option name 'eth0.2'
	option macaddr 'xxxx'

config interface 'wan'
	option device 'eth0.2'
	option proto 'dhcp'
	option metric '16'

config interface 'wan6'
	option device 'eth0.2'
	option proto 'dhcpv6'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '2 3 4 5 0t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '1 0t'

config interface 'VPN'
	option proto 'dhcp'
	option device 'vpn_soft0'
	option delegate '0'
	option defaultroute '0'

config device
	option name 'vpn_soft0'
	option ipv6 '0'

config route
	option interface 'VPN'
	option target 'xxxx'
	option netmask 'xxxx'
	option gateway '192.168.192.1'

config route
	option interface 'VPN'
	option target 'xxxx'
	option netmask 'xxxx'
	option gateway '192.168.192.1'

```

stangri, 2021-12-03:
Thank you, I wasn't sure if Softether has a specific protocol requirement for the interface definition.

For the current installation of VPR, you can try adding `VPN` to the list of supported interfaces, it may actually work as is.

tew42, 2021-12-04:
Good thought, hadn't realized it was possible to force-add an interface.

Seems that the policy routes are applied correctly (e.g. all PC-source traffic over VPN); however, the gateway configuration doesn't appear to work:
wan/eth0.2/xxxx ✓
VPN/vpn_soft0/192.168.192.10

I.e., the gateway is set for the VPN *client* IP, rather than the server gateway (which would be .1)

stangri, 2021-12-06:
> Seems that the policy routes are applied correctly (e.g. all PC-source traffic over VPN); however, the gateway configuration doesn't appear to work: wan/eth0.2/xxxx ✓ VPN/vpn_soft0/192.168.192.10
> 
> I.e., the gateway is set for the VPN _client_ IP, rather than the server gateway (which would be .1)

Yeah, that's the only information available from the router. Did the actual policy routing work tho?

tew42, 2021-12-06:
Yes - I set created policy to divert certain local IPs to the VPN, which then ("successfully") went nowhere (due to the gateway problem). Haven't tried more advanced scenarios yet, but it does seem that with a "forced" interface the underlying plugin logic essentially works.

I wonder if the correct gateway info could be pulled from this snippet here? Because this one seems to be the automatically-generated default route of the VPN interface (inactive due to my config since I'm having some issues with default routing, but the gateway info is correct)
```
                "route": [
                        {
                                "target": "0.0.0.0",
                                "mask": 0,
                                "nexthop": "192.168.192.1",
                                "source": "192.168.192.10/32"
```

stangri, 2021-12-07:
My understanding is that it's where `network_get_gateway` gets information from, but on your system it returned nothing.

As a work-around, you can create a custom user script and replace the gateway for the table created for this VPN. Long term, I'd like to implement softether support using network OpenWrt functions. 

If you can switch to `pbr` package, it should support softether without any tricks.

tew42, 2021-12-07:
According to the documentation:
```
# determine IPv4 gateway of given logical interface
# 1: destination variable
# 2: interface
# 3: consider inactive gateway if "true" (optional)
network_get_gateway() {
```
So actually replacing the calls
`network_get_gateway gw "$iface"`
with
`network_get_gateway gw "$iface" true`
could do the trick

Will try to test later/tomorrow

stangri, 2021-12-15:
Thank you, I've implemented this into `pbr` as a more elegant way of getting `nexthop`: https://github.com/stangri/source.openwrt.melmac.net/commit/33cca0a9d10ae04846365b4ef4d591fd045c79f3

I'm reluctant to make changes in `vpn-policy-routing` because:
1. The OpenWrt 19.07 is stuck on a very old 0.2.x version, as I don't have the resources to debug the newer versions of VPR on older OpenWrt version. I've had to revert the OpenWrt 19.07 package to an older version due to this.
2. Since I've started working on `pbr` package, I don't use VPR anymore, so I can't efficiently troubleshoot  issues on release (OpenWrt 21.02) and have previously had to revert the 21.02 package to a previous, known good version/commit. Also, for anyone on OpenWrt 21.02 I'd highly recommend using `pbr` instead.
3. For anyone on snapshots, I'd highly recommend using `pbr` instead.

If you want to test `pbr` please check https://docs.openwrt.melmac.net/pbr/.

tew42, 2021-12-16:
Ah, okay, thanks for clarifying! 

With PBR, unfortunately, I'm having some unrelated issues - I can't even get the service set to "enabled" (immediately reverts to disabled, with no error messages logged)... With vpn-policy-routing working fine for now, I'm not currently very inclined to spend a lot of time figuring out what's going on there however

stangri, 2021-12-16:
Thanks, I'll see if I can reproduce the PBR issue.

closed 2021-12-16 by stangri