mossdef-org/source.mossdef.org #149: [https-dns-proxy] Wish: Use confdir instead of uci for DNS servers

dgcampea, 2022-01-27:
The current approach actually causes some unexpected behavior in that it deletes entries that happen to match the check above (which are not necessarily _https-doh-proxy_ entries).
These could be a local/manually-added DNS server or an address (like `/use-application-dns.net/`).
A generated `.conf` file would honor user settings and also make it unnecessary to perform a "uci backup dance".

Instead of overriding `dhcp.@dnsmasq.server` via uci as suggested by:
<https://github.com/stangri/source.openwrt.melmac.net/blob/afc5144bf6c771829e003ab1fd4ce44c05f321d7/https-dns-proxy/files/https-dns-proxy.init#L259>
`dhcp.@dnsmasq[0].confdir` can be queried instead (defaults to `/tmp/dnsmasq.d` if not present) and the https-doh instances added by writing a `.conf` file consisting of `server=` entries to `confdir` and issuing a `/etc/init.d/dnsmasq reload` or by sending a `SIGHUP`.

stangri, 2022-01-27:
> The current approach actually causes some unexpected behavior in that it deletes entries that happen to match the check above (which are not necessarily _https-doh-proxy_ entries).

Could you please clarify what do you consider an expected and unexpected behaviours?

dgcampea, 2022-01-27:
I'd say that https-dns-proxy managing it's own `server=` instances would be expected (as the port or ip they're listening may get changed by the user via uci or luci).
Unexpected would be any entry that wasn't added by https-dns-proxy to get removed from dnsmasq.

stangri, 2022-01-27:
So when you start the `https-dns-proxy` is your intention to still use the legacy resolvers alongside/at the same time as the DoH resolvers?

dgcampea, 2022-01-27:
If by legacy you mean any existing `server=` entry, yes.

stangri, 2022-01-27:
> If by legacy you mean any existing `server=` entry, yes.

Could you please elaborate on your user scenario then? My expectation would have been that if I start the `https-dns-proxy`, I would NOT want the unecrypted DNS traffic to leave my router, hence removal of existing server entries which point to legacy resolvers.

dgcampea, 2022-01-27:
> Could you please elaborate on your user scenario then?

In my case, I have a `server=/use-application-dns.net/` line that I'd like to keep.

> I would NOT want the unecrypted DNS traffic to leave my router, hence removal of existing server entries which point to legacy resolvers.

IIRC the legacy solvers (or ISP upstream ones) are not present in dnsmasq `server=` lines at all.
This is not my current use case but what if someone is using dnsmasq + a private/custom DNS server? (could be a LAN only server that's serving some custom zones that are not meant to be solved upstream)
I don't think there's any harm in keeping the existing dnsmasq `server=` lines in this case.

I think it would be reasonable to expect that any `server=` lines that already exist were explicitly added by the user so they know exactly what their dnsmasq instance is doing.

stangri, 2022-01-28:
> IIRC the legacy solvers (or ISP upstream ones) are not present in dnsmasq `server=` lines at all.

Actually, this is exactly where dnsmasq expects to find upstream resolvers: https://openwrt.org/docs/guide-user/base-system/dhcp#all_options

> In my case, I have a `server=/use-application-dns.net/` line that I'd like to keep.

I see. If you use `server=/use-application-dns.net/` or anything with the `#` the init script will preserve this resolver in the config, so I don't understand what the issue is.

dgcampea, 2022-01-30:
> If you use server=/use-application-dns.net/ or anything with the # the init script will preserve this resolver in the config, so I don't understand what the issue is.

Does it? I remember https-dns-proxy wiping some entries I had manually entered into dnsmasq.
I'm not sure if I added the entries before or after starting the https-dns-proxy though, maybe the order matters?

stangri, 2022-01-31:
> Does it? I remember https-dns-proxy wiping some entries I had manually entered into dnsmasq. I'm not sure if I added the entries before or after starting the https-dns-proxy though, maybe the order [...]

Yes, if you had started `https-dns-proxy` and then added something to dnsmasq, those changes would be wiped. I'll merge a PR which resolves that in a few days.

closed 2022-02-02 by stangri