mossdef-org/source.mossdef.org #151: [pbr] Wish: options to allow 'pbr_wan_dst_net' policy on the top or support using a user-define ipset (hash:net, IP-CIDR) as remote address.

imKnect, 2022-02-16:
**Relate problem**
[image]
I have a policy routing `0-65535` `dst` to my VPN gateway, which I'd like to be the bottom-line. But I also have a lot of IP-CIDRs that need to bypass this VPN gateway, it's a little annoying to manually add a lot of IP-CIDRs to the config.

**Solutions**
Can you provide a option allowing user-script 'pbr_wan_dst_net' policy ('pbr_wan_dst_net', etc) being on the top of the list?
Or can you support using a user-define ipset (hash:net, IP-CIDR) as `dest_addr`?
Both solutions can improve the situation.

BTW, a small bug. Adding IP-CIDR or IP-CIDRs to the `dest_addr` field turns out red alert, fortunately `Save & Apply` works fine.
[image]

stangri, 2022-02-16:
In the user script you can use `ips` command to create an ipset you need, an `ipt` command to insert the iptables policy targeting this ipset at the top of the chain and then import ipset entries from an external file.

imKnect, 2022-02-17:
> In the user script you can use `ips` command to create an ipset you need, an `ipt` command to insert the iptables policy targeting this ipset at the top of the chain and then import ipset entries from an external file.

Thank you but is there any docs about what I can do and how to do in the user script? Or I need to find out by myself?

stangri, 2022-02-18:
It's shell (default OpenWrt shell is `ash`) scripting essentially, so it would be outside of scope of the PBR documentation.

One suggestion on custom scripting for PBR is that you may want to create a policy in PBR settings with one of the known domains from your list to make sure that PBR creates the iptables rule and an ipset for it and then just fill that custom ipset in your script.

stangri, 2022-09-29:
The current version of `pbr` creates multiple sets (4 sets for ipset and 3 sets for nft set), if something is still missing, please reopen this with details.

closed 2022-09-29 by stangri