mossdef-org/source.mossdef.org #152: [pbr] Wish: `firewall4` compatibility

maxromanovsky, 2022-02-16:
I just wanted to track incompatibility of `pbr` with `firewall4`, as per your comment in https://github.com/openwrt/packages/issues/16818#issuecomment-937169645

This would be help enormously those who use openwrt snapshots to decide when to update to the latest build.

stangri, 2022-09-13:
Additional information available at https://docs.openwrt.melmac.net/pbr/#statement-about-openwrt-22030-release-and-this-package.

The very early version of the `nft`-compatible package is also available in my repo, however the transition is not fully complete and, most importantly, until I can build an nft sets-compatible dnsmasq for my system, I can't properly test it.

KeyofBlueS, 2022-09-26:
dnsmasq 2.87 has been released https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=tag;h=refs/tags/v2.87, now we need to wait for the update in openwrt :+1:

stangri, 2022-09-26:
Technically, [at the current state](https://docs.openwrt.melmac.net/pbr/#statement-about-openwrt-22030-release-and-this-package), the `pbr` package is firewall4/nft compatible.

stangri, 2022-09-29:
I'm lacking time to build a dnsmasq-full 2.87 (with nft sets support) for OpenWrt, but current `pbr` should be compatible with the nft sets. As the current version of pbr is fw4/nft-compatible, I'm closing this.

closed 2022-09-29 by stangri