mossdef-org/source.mossdef.org #153: [pbr] Issue: Ipv6 Rules and Mangle table not populated when ipv6_enabled=1

tmoore22, 2022-02-17:
**Describe the bug**
A clear and concise description of what the bug is.

ipv6 rules not applied:

```sh
root@OpenWrt:~# ip -6 rule
0:      from all lookup local
32766:  from all lookup main
4200000000:     from 2001:985:xxxx::x/64 iif br-lan unreachable
4200000001:     from all iif lo failed_policy
4200000017:     from all iif br-lan failed_policy
4200000018:     from all iif wan.4 failed_policy
4200000020:     from all iif vpn failed_policy
4200000023:     from all iif pppoe-wan failed_policy
4200000023:     from all iif pppoe-wan failed_policy
```

ipv4 are fine:
```sh
root@OpenWrt:~# ip rule
0:      from all lookup local
29999:  from all fwmark 0x20000/0xff0000 lookup vpn
30000:  from all fwmark 0x10000/0xff0000 lookup wan
32766:  from all lookup main
32767:  from all lookup default
```

no ip6tables PBR_PREROUTING entries:
```sh
root@OpenWrt:~# ip6tables -L -t mangle
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
PBR_PREROUTING  all      anywhere             anywhere             mark match 0x0/0xff0000

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination
TCPMSS     tcp      anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp      anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
PBR_FORWARD  all      anywhere             anywhere             mark match 0x0/0xff0000

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
PBR_OUTPUT  all      anywhere             anywhere             mark match 0x0/0xff0000

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination

Chain PBR_FORWARD (1 references)
target     prot opt source               destination

Chain PBR_MARK0x010000 (1 references)
target     prot opt source               destination
MARK       all      anywhere             anywhere             MARK xset 0x10000/0xff0000
RETURN     all      anywhere             anywhere

Chain PBR_MARK0x020000 (0 references)
target     prot opt source               destination
MARK       all      anywhere             anywhere             MARK xset 0x20000/0xff0000
RETURN     all      anywhere             anywhere

Chain PBR_OUTPUT (1 references)
target     prot opt source               destination
PBR_MARK0x010000  udp      anywhere             anywhere            [goto]  multiport sports 51820 /* WireGuard_Server */

Chain PBR_PREROUTING (1 references)
target     prot opt source               destination
[... 423 lines cut ...]
create pbr_vpn_dst_ip_cfg026ff5 hash:ip family inet hashsize 1024 maxelem 65536 comment
create pbr_vpn6_dst_ip_cfg026ff5 hash:ip family inet6 hashsize 1024 maxelem 65536 comment
create pbr_vpn_dst_ip_cfg036ff5 hash:ip family inet hashsize 1024 maxelem 65536 comment
create pbr_vpn6_dst_ip_cfg036ff5 hash:ip family inet6 hashsize 1024 maxelem 65536 comment
create pbr_wan_dst_ip_cfg016ff5 hash:ip family inet hashsize 1024 maxelem 65536 comment
create pbr_wan6_dst_ip_cfg016ff5 hash:ip family inet6 hashsize 1024 maxelem 65536 comment
create pbr_wan_dst_ip_cfg026ff5 hash:ip family inet hashsize 1024 maxelem 65536 comment
create pbr_wan6_dst_ip_cfg026ff5 hash:ip family inet6 hashsize 1024 maxelem 65536 comment
create pbr_wan_dst_ip_cfg036ff5 hash:ip family inet hashsize 1024 maxelem 65536 comment
create pbr_wan6_dst_ip_cfg036ff5 hash:ip family inet6 hashsize 1024 maxelem 65536 comment
============================================================
DNSMASQ ipsets
ipset=/bbc.co.uk/pbr_vpn_dst_ip_cfg016ff5,pbr_vpn6_dst_ip_cfg016ff5 # BBC: bbc.co.uk
ipset=/bbc.com/pbr_vpn_dst_ip_cfg016ff5,pbr_vpn6_dst_ip_cfg016ff5 # BBC: bbc.com
ipset=/bbci.co.uk/pbr_vpn_dst_ip_cfg016ff5,pbr_vpn6_dst_ip_cfg016ff5 # BBC: bbci.co.uk
ipset=/bidi.net.uk/pbr_vpn_dst_ip_cfg016ff5,pbr_vpn6_dst_ip_cfg016ff5 # BBC: bidi.net.uk
ipset=/vod-dash-uk-live.akamaized.net/pbr_vpn_dst_ip_cfg016ff5,pbr_vpn6_dst_ip_cfg016ff5 # BBC: vod-dash-uk-live.akamaized.net
ipset=/itv.co.uk/pbr_vpn_dst_ip_cfg026ff5,pbr_vpn6_dst_ip_cfg026ff5 # ITV: itv.co.uk
ipset=/itv.com/pbr_vpn_dst_ip_cfg026ff5,pbr_vpn6_dst_ip_cfg026ff5 # ITV: itv.com
ipset=/scan.co.uk/pbr_vpn_dst_ip_cfg036ff5,pbr_vpn6_dst_ip_cfg036ff5 # Scan: scan.co.uk
ipset=/ingest.sentry.io/pbr_vpn_dst_ip_cfg036ff5,pbr_vpn6_dst_ip_cfg036ff5 # Scan: ingest.sentry.io
============================================================
Your support details have been logged to '/var/pbr-support'. [✓]

   ```

tmoore22, 2022-05-13:
this issue remains, can anyone provide an answer?

stangri, 2022-06-06:
> this issue remains, can anyone provide an answer?

I've only just now had time to revisit the code again, yes, some policies are broken for IPv6 in `pbr` and it's caused by the massive code refactoring between `vpn-policy-routing` and `pbr`. Can you try using the former until I have the time to fix this issue?

I'll need to think on how it's best to address it.

stangri, 2022-06-15:
I've tried to address it in `pbr` version `0.9.5-1`.

tmoore22, 2022-09-05:
> I've tried to address it in `pbr` version `0.9.5-1`.

Hi, just re-installed (pbr) with same configuration version 0.9.5-1 and this issue still remains unfortunately.

stangri, 2022-09-05:
Can I see the config?

tmoore22, 2022-09-05:
> Can I see the config?

its all as-described above, the config is included from before; nothing changed.

I have to add the `ip -6 rules` and the `ip6tables -t mangle` entries manually before it will work

tmoore22, 2022-09-05:
aha, I just found this in `/etc/init.d/pbr status -d`
`NAT IPv6 Table
ip6tables v1.8.7 (legacy): can't initialize ip6tables table `nat': Table does not exist (do you need to insmod?)`

I am using OpenWrt 21.02.3 and `ip6tables` is expecting `mangle` not `nat`

`root@OpenWrt:/tmp# ip6tables -L -t nat
ip6tables v1.8.7 (legacy): can't initialize ip6tables table `nat': Table does not exist (do you need to insmod?)
Perhaps ip6tables or your kernel needs to be upgraded.
root@OpenWrt:/tmp# iptables -L -t mangle
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
......
`

stangri, 2022-09-06:
> aha, I just found this in `/etc/init.d/pbr status -d`
> 
> `NAT IPv6 Table ip6tables v1.8.7 (legacy): can't initialize ip6tables table 'nat': Table does not exist (do you need to insmod?)`

Hey, although this is an unrelated issue, thanks for bringing it up. Given that nft is the future and the code is being rewritten for nft, I don't want to spend time fixing a status output bug in an iptables version.

For the IPv6 woes, after you get `pbr` to the state described in your original post, could you please try:

```
ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg016ff5 dst -m comment --comment BBC -g PBR_MARK0x020000
ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg026ff5 dst -m comment --comment ITV -g PBR_MARK0x020000
ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg036ff5 dst -m comment --comment Scan -g PBR_MARK0x020000
```

and let me know if there are any errors?

tmoore22, 2022-09-06:
root@OpenWrt:~# ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_c
fg016ff5 dst -m comment --comment BBC -g PBR_MARK0x020000
ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg026ff5 dst -m                                                                   comment --comment ITV -g PBR_MARK0x020000
ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg036ff5 dst -m                                                                   comment --comment Scan -g PBR_MARK0x020000
ip6tables v1.8.7 (legacy): goto 'PBR_MARK0x020000' is not a chain

Try `ip6tables -h' or 'ip6tables --help' for more information.
root@OpenWrt:~# ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_c
fg026ff5 dst -m comment --comment ITV -g PBR_MARK0x020000
ip6tables v1.8.7 (legacy): goto 'PBR_MARK0x020000' is not a chain

Try `ip6tables -h' or 'ip6tables --help' for more information.
root@OpenWrt:~# ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_c
fg036ff5 dst -m comment --comment Scan -g PBR_MARK0x020000
ip6tables v1.8.7 (legacy): goto 'PBR_MARK0x020000' is not a chain

Try `ip6tables -h' or 'ip6tables --help' for more information.
root@OpenWrt:~#

tmoore22, 2022-09-06:
```
root@OpenWrt:~# ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg016ff5 dst -m comment --comment BBC -g PBR_MARK0x020000
ip6tables v1.8.7 (legacy): goto 'PBR_MARK0x020000' is not a chain

Try `ip6tables -h' or 'ip6tables --help' for more information.
root@OpenWrt:~# ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg026ff5 dst -m comment --comment ITV -g PBR_MARK0x020000
ip6tables v1.8.7 (legacy): goto 'PBR_MARK0x020000' is not a chain

Try `ip6tables -h' or 'ip6tables --help' for more information.
root@OpenWrt:~# ip6tables -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg036ff5 dst -m comment --comment Scan -g PBR_MARK0x020000
ip6tables v1.8.7 (legacy): goto 'PBR_MARK0x020000' is not a chain

Try `ip6tables -h' or 'ip6tables --help' for more information.

```

tmoore22, 2022-09-06:
although with the addition of ' -t mangle '

```
root@OpenWrt:~# ip6tables -t mangle -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg016ff5 dst -m comment --comment BBC -g PBR_MARK0x020000
root@OpenWrt:~# ip6tables -t mangle -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg026ff5 dst -m comment --comment ITV -g PBR_MARK0x020000
root@OpenWrt:~# ip6tables -t mangle -A PBR_PREROUTING -m set --match-set pbr_vpn6_dst_ip_cfg036ff5 dst -m comment --comment Scan -g PBR_MARK0x020000

```

stangri, 2022-09-06:
Ah, right, forgot to indicate the table in the instructions. I'm puzzled as to why it's not added within the script. If you can pursue this further, could you please add:

`echo "param: $param"; echo "param6: $param6";` 

just above [this line](https://github.com/stangri/source.openwrt.melmac.net/blob/master/pbr/files/etc/init.d/pbr.iptables.init#L644) in the init script?

tmoore22, 2022-09-07:
param6 looks good, but for some reason the `ip6tables -t mangle -L` still shows empty!

If I copy/paste the param6 and use it manually with `ip6tables` then the rule is applied OK... seems something wrong with calling `ip6tables` when actually applying the param6?

```
root@OpenWrt:~# /etc/init.d/pbr reload
Routing 'BBC' via vpn param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg016ff5 dst -m comment --comment BBC
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg016ff5 dst
param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg016ff5 dst -m comment --comment BBC
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg016ff5 dst
param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg016ff5 dst -m comment --comment BBC
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg016ff5 dst
param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg016ff5 dst -m comment --comment BBC
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg016ff5 dst
param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg016ff5 dst -m comment --comment BBC
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg016ff5 dst
param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg016ff5 dst -m comment --comment BBC
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg016ff5 dst
[✓]
Routing 'ITV' via vpn param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg026ff5 dst -m comment --comment ITV
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg026ff5 dst
param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg026ff5 dst -m comment --comment ITV
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg026ff5 dst
[✓]
Routing 'Scan' via vpn param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg036ff5 dst -m comment --comment Scan
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg036ff5 dst
param: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn_dst_ip_cfg036ff5 dst -m comment --comment Scan
param6: -t mangle -A PBR_PREROUTING -g PBR_MARK0x020000 -m set  --match-set pbr_vpn6_dst_ip_cfg036ff5 dst
[✓]
Routing 'WireGuard Server' via wan param: -t mangle -A PBR_OUTPUT -g PBR_MARK0x010000 -p udp -m multiport  --sport 51820 -m comment --comment WireGuard_Server
param6: -t mangle -A PBR_OUTPUT -g PBR_MARK0x010000 -p udp -m multiport  --sport 51820
[✓]
Routing 'Ignore Local Traffic' via ignore param: -t mangle -A PBR_PREROUTING -j RETURN  -d 172.16.0.0/24 -m comment --comment Ignore_Local_Traffic
param6: -t mangle -A PBR_PREROUTING -j RETURN  -d 172.16.0.0/24
[✓]
pbr 0.9.5-1 started with gateways:
wan/94.210.194.1/2001:1c02:2100:0:cd77:c97a:9a06:9079/128
fe80::ee41:18ff:fe0b:810f/64 [✓]
vpn/172.16.0.2/fdb6:4b4f:f8f5:f1f0::2/64
root@OpenWrt:~# ip6tables -t mangle -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
PBR_PREROUTING  all      anywhere             anywhere             mark match 0x0/0xff0000

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination
TCPMSS     tcp      anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp      anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
PBR_OUTPUT  all      anywhere             anywhere             mark match 0x0/0xff0000

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination

Chain PBR_MARK0x010000 (1 references)
target     prot opt source               destination
MARK       all      anywhere             anywhere             MARK xset 0x10000/0xff0000
RETURN     all      anywhere             anywhere

Chain PBR_MARK0x020000 (0 references)
target     prot opt source               destination
MARK       all      anywhere             anywhere             MARK xset 0x20000/0xff0000
RETURN     all      anywhere             anywhere

Chain PBR_OUTPUT (1 references)
target     prot opt source               destination
PBR_MARK0x010000  udp      anywhere             anywhere            [goto]  multiport sports 51820 /* WireGuard_Server */

Chain PBR_PREROUTING (1 references)
target     prot opt source               destination
root@OpenWrt:~# 

```

tmoore22, 2022-09-07:
1. added `ip -6 rules` (totally missing from current script)
2. split line 647 and now `ip6tables` rules are updated correctly
[pbr_ip6.diff.txt](https://github.com/stangri/source.openwrt.melmac.net/files/9505297/pbr_ip6.diff.txt)

tmoore22, 2022-09-07:
also;

```
@@ -640,7 +644,6 @@
                fi

                [ -n "$name" ] && param="$param -m comment --comment $(str_extras_to_underscore "$name")"
+               [ -n "$name" ] && param6="$param6 -m comment --comment $(str_extras_to_underscore "$name")"

                if [ "$param" = "$param6" ]; then

```

stangri, 2022-09-07:
Thank you so much, I've [incorporated your changes](https://github.com/stangri/source.openwrt.melmac.net/commit/654ebd551add8ee09e31c095f02a6476f5d58d11) and builds 0.9.6-2 of `pbr-iptables` and `luci-app-pbr` are available from my repo now. I'd appreciate if you could update to the new IPK and test.

stangri, 2022-09-29:
I have rewritten the IPv4/IPv6 rules creation pretty much from scratch for both iptables and nft, if you encounter specific problems, please open a new issue.

closed 2022-09-29 by stangri