mossdef-org/source.mossdef.org #155: [https-dns-proxy] Canary

dgcampea, 2022-03-06:
https://support.mozilla.org/en-US/kb/canary-domain-use-application-dnsnet

stangri, 2022-03-06:
@dgcampea thank you for bringing it up. I'll create a similar issue in the upstream package, if it's implemented there, there would be no need for OpenWrt-specific change.

stangri, 2022-04-22:
Sorry it took a while, I was waiting for the decision on the upstream package.

I've implemented at in this commit: https://github.com/stangri/source.openwrt.melmac.net/commit/ba8717b5e5a37cfa145a065be3170bde4732a9be and if you want to test it out without compiling the package for yourself, you can just copy the init script from https://raw.githubusercontent.com/stangri/source.openwrt.melmac.net/ba8717b5e5a37cfa145a065be3170bde4732a9be/https-dns-proxy/files/https-dns-proxy.init into your live router.

Please let me know if you can/want to test it before I submit a PR for the OpenWrt packages repo.

dgcampea, 2022-04-23:
I'd use the 'address' directive here instead of 'server'.
https://thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html

stangri, 2022-04-24:
> I'd use the 'address' directive here instead of 'server'. https://thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html

Why? That same doc you've linked explicitly states that ' --address=/example.com/ is equivalent to --server=/example.com/'. There's already code to backup and restore the `server` entries, why would I want to use `address` instead?

dgcampea, 2022-04-24:
I was under the impression that if you somehow had an entry for `use-application-dns.net` in /etc/hosts, `--address` would still return NXDOMAIN whereas `--server` would resolve it. Did a quick test, it turns out both have the same behavior (resolve from /etc/hosts).

I usually see `address=` used for domain blacklists and the directive seems more adequate given the context but since it ends up being equivalent I guess there's no harm in using `server=` either.

stangri, 2022-04-25:
> I usually see `address=` used for domain blacklists and the directive seems more adequate given the context but since it ends up being equivalent I guess there's no harm in using `server=` either.

As a maintainer of `simple-adblock` which uses `server` to get NXDOMAIN for blocked domains, I'm curious to find out which blocklists apps/services use `address` instead.

dgcampea, 2022-05-02:
> I'm curious to find out which blocklists apps/services use address instead.

I remember seeing some lists that used `address=` but I'm afraid I lost the bookmark for it :(