mossdef-org/source.mossdef.org #156: [policy-based-routing] Issue: error with remote domain

KeyofBlueS, 2022-09-10:
Hi,

if I add a remote domain in a policy, i got an error, e.g.:

```
config policy
	option name 'example'
	option interface 'example_iface'
	option dest_addr 'google.com
```

Service Errors:
`iptables -t mangle -A PBR_PREROUTING -g PBR_MARK0x010000 -d 142.250.184.78,2a00:1450:4002:405::200e -m comment --comment example`

if i try to run that command i got:
```
# iptables -t mangle -A PBR_PREROUTING -g PBR_MARK0x010000 -d 142.250.184.78,2a00:1450:4002:405::200e -m comment --comment example
iptables v1.8.7 (legacy): host/network `2a00:1450:4002:405::200e' not found
Try `iptables -h' or 'iptables --help' for more information.
```

I guess pbr is trying to apply ipv4 and ipv6 addresses with iptables command, while, i think, it should use iptables for ipv4 addresses and ip6tables for ipv6 addresses e.g.:
```
# iptables -t mangle -A PBR_PREROUTING -g PBR_MARK0x010000 -d 142.250.184.78 -m comment --comment example
# ip6tables -t mangle -A PBR_PREROUTING -g PBR_MARK0x010000 -d 2a00:1450:4002:405::200e -m comment --comment example
```

Tried with IPv6 Support both enabled and disabled, nothing changes.

Sorry if i've not provided any requested output, but i'm not comfortable with that. If you really need, please ask me.

Thanks and best regards.

stangri, 2022-09-12:
I'd highly recommend you switch to using resolver sets support for the policies with domains, but this particular issue has been fixed in `pbr-iptables` version 0.9.6-5. Please reopen if it's not fixed after updating the package.

KeyofBlueS, 2022-09-12:
With version 0.9.6-5 that error is gone. The fact is, i was using dnsmasq ip set for domains yet.
In any case, the domain added in the policy are not routed via the selected interface, i've flushed dns, rebooted the router and the local devices.

At this point i don't know if this issue should be reopened.
Here there are my configurations, i've omitted some values. I hope it can be usefull for you.

Thanks.

1. ```/etc/config/dhcp```

```
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option localservice '1'
	option ednspacket_max '1232'
	option expandhosts '1'

config dhcp 'lan'
	option interface 'lan'
	option leasetime '12h'
	option dhcpv4 'server'
	list dhcp_option 'omissis'
	option force '1'
	option start '4'
	option limit '200'
	list ra_flags 'none'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'
	option start '100'
	option limit '150'
	option leasetime '12h'
	list dhcp_option 'omissis'
	list ra_flags 'none'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'
```

2. ```/etc/config/network```

 ```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'
	option ipv6 '0'

config interface 'lan'
[... 385 lines cut ...]

Chain prerouting_wan_rule (1 references)
target     prot opt source               destination         

Chain zone_lan_postrouting (1 references)
target     prot opt source               destination         
postrouting_lan_rule  all  --  anywhere             anywhere             /* !fw3: Custom lan postrouting rule chain */

Chain zone_lan_prerouting (1 references)
target     prot opt source               destination         
prerouting_lan_rule  all  --  anywhere             anywhere             /* !fw3: Custom lan prerouting rule chain */

Chain zone_wan_postrouting (4 references)
target     prot opt source               destination         
postrouting_wan_rule  all  --  anywhere             anywhere             /* !fw3: Custom wan postrouting rule chain */
MASQUERADE  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_wan_prerouting (4 references)
target     prot opt source               destination         
prerouting_wan_rule  all  --  anywhere             anywhere             /* !fw3: Custom wan prerouting rule chain */
============================================================
Current ipsets
============================================================
Your support details have been logged to '/var/pbr-support'. [✓]
```

stangri, 2022-09-13:
The `pbr reload` output seems to be truncated.

What does `opkg list-installed | grep pbr` say?

KeyofBlueS, 2022-09-13:
It is the complete output, i've tried again right now, but it wasn't like this in version 0.9.6-3, i remember to see something like "creating table..."

KeyofBlueS, 2022-09-13:
In order to trying to diagnose this issue, i've compared my configs with others in the issues section, and i've noticed that my pbr config had these options:

```
	option resolver_ipset 'dnsmasq.ipset'
	option iptables_rule_option 'append'
	option rule_create_option 'add'
	option resolver_set 'dnsmasq.ipset'
```

it looked weird to me and guessing there may be some misconfiguration, i've resetted and reconfigured pbr from scratch, and now is working as expected. From where `resolver_ipset` and `iptables_rule_option` came from? migrating from vpn-policy-routing maybe?

After that i've tried `/etc/init.d/pbr reload` end the output was complete, however subsequent attempts showed the truncated output.

Nevertheless the routing seems to work just fine, and the domain added in the policy are now routed via the selected interface.

You advised me to use resolver sets support for the policies with domains; how can i check if, even if correctly configured in pbr, i'm really using it?

Thanks!

stangri, 2022-09-13:
> it looked weird to me and guessing there may be some misconfiguration

The first two are legacy settings, in preparation to nft-compatible version I've had to remove any ipset/iptables from the option names.

> You advised me to use resolver sets support for the policies with domains; how can i check if, even if correctly configured in pbr, i'm really using it?

When you run `service pbr status` you should see the section for DNSMASQ ipsets, it shouldn't be empty.

KeyofBlueS, 2022-09-13:
Ok so in the `pbr status` log there is no DNSMASQ ipsets, what should i do to trackdown this issue? Do you need any log?

Thanks.

stangri, 2022-09-15:
> Ok so in the `pbr status` log there is no DNSMASQ ipsets, what should i do to trackdown this issue? Do you need any log?
> 
> Thanks.

Yes, all the files requested in README again would help.

KeyofBlueS, 2022-09-16:
I think i've found what the problem is.
After some experiments, i've found that the DNSMASQ ipsets were correctly created if, for a domain, i set the interface to wan, but not when set to vpn.
The pbr script will fail in the `ips` funcion, command `create_resolver`, precisely at `ipset -q -! create "$ipset" $param 2>/dev/null || failFlag=1`, the exit code is `1`.
My vpn interface name is not "vpn", but a string of 14 characters (the maximum allowed through luci when creating a new interface).
So when i tried the command below, omitting `-q` option:
`# ipset -! create "pbr_vpn_loong_name_dst_ip_cfg016ff5" hash:ip comment`
i get:
`ipset v7.6: Syntax error: setname 'pbr_vpn_loong_name_dst_ip_cfg016ff5' is longer than 31 characters`

It's very late now, tomorrow i'll try to recreate the vpn interface giving it a shorter name to confirm, but i'm confident that the long name is the problem. If a way to manage this problem internally isn't possible, i advice to warn users to give an interface a name not longer than 10 characters, in the readme and in the error log if possible.

Thank you for your work and best regards :+1: