mossdef-org/source.mossdef.org #158: [policy-based-routing] Compatibility issues between pbr-nftables and OpenWRT 22.03

orekerete, 2022-09-12:
Firstly thank you for all the work you have put into this very essential solution. Permit me to explain my observations via the series of tests I have performed as a result of me upgrading to the recently release OpenWRT 22.03 final build, in order to assist you save time during troubleshooting.  As you may be aware 22.03 has moved from firewall3 (iptables) to firewall4 (nftables). So I decided this weekend to upgrade my live setup and got a "bricked" result (its a VM but you get the imagery) which was partly due to me not following best practices I am sure. I preceded to revert to an earlier VM snapshot for testing, and then after that I built a new vanilla image in order to confirm my observations were not peculiar to my original build.

- The first issue / observation came after I added the stangri GitHub repository and installed PBR via the command line with _opkg install pbr luci-app-pbr_ and noticed that this installed **pbr-iptables** instead of **pbr-nftables**.  I would suggest that the installer logic be amended if possible to detect 22.03 and install **pbr-nftables** or the install instructions be amended to read _opkg install pbr-nftables luci-app-pbr_ for 22.03+ builds.
- The second issue I noticed was after I manually removed **pbr-iptables** and replaced with **pbr-nftables**, and enabled the service.  My reasoning for this was because Status -> Firewall kept complaining that it was not advisable to mix nftable rules with iptable rules which occurred when i was using the pbr-iptables module [https://prnt.sc/T5uYSCwrxJJ7](url).  The service status showed as Stopped [https://prnt.sc/2sygXijt1d0H](url) even though the System Log shows the service successfully started [https://prnt.sc/Pd8uqQIIgPY3](url).
- On my vanilla install I attempted to enable one of the default rules but I got the following error [https://prnt.sc/JPkOjFWM3Cl8](url) and when I attempted to execute that command via the terminal I got [https://prnt.sc/JH_w4H9ZOxf2](url).  I adjusted the text label assuming that was the error but still got this [https://prnt.sc/cqcFU2MU3LXS](url)
- I then decided to create a new rule for my MAC to route to WAN and got this error no matter what I did [https://prnt.sc/9wtHJAGc0XaF](url).

I am going to revert my live VM to pre 22.03 and continue using the pbr-iptables module, but I will keep the vanilla system running if you need me to report back any further information.  Thanks in advance for any assistance you can provide.

orekerete, 2022-09-12:
I have just seen [Issue 152](https://github.com/stangri/source.openwrt.melmac.net/issues/152) appears to be referencing this matter.  Should I close this and merge my observations with that?

stangri, 2022-09-13:
Not sure if you've read [this](https://docs.openwrt.melmac.net/pbr/#statement-about-openwrt-22030-release-and-this-package) or noticed changes to the README at large.

I've also updated the issue 152 with additional information on transition plans.

orekerete, 2022-09-13:
Thanks for your prompt response and my apologies for missing the documentation update.  Based on that I have retained 22.03 and installed the PBR-IPTables module. It is working as you confirmed so thanks however I am seeing a bug which I am assuming crept in as of the latest version update (as it was working previously). Specifically that _Strictly enforce policies when their gateway is down_ is not being applied. An example I created a rule to route my traffic via OpenVPN which PBR does, but when i disconnect OpenVPN my traffic is then routed via my WAN instead of being blocked.

Should I close this and open that as a separate issue?

stangri, 2022-09-29:
Please create separate issues if you still encounter nft-related problems with the latest version of `pbr`.

closed 2022-09-29 by stangri