mossdef-org/source.mossdef.org #160: [simple-adblock] Issue: Allowed Domain allows all domains which terminate in that substring, even partial words.

Cheddoleum, 2022-09-30:
**Describe the bug**
Adding a single domain to allowed_domain allows all domains which end with the text of that domain; not just on a dot boundary but even in the middle of words.  E.G., allowing domain "t.co" removes "fat.co" "ketchup.tatertot.co", etc from (e.g.) dnsmasq.servers.

**To Reproduce**
Steps to reproduce the behavior:
Using default blocklists.
Added "t.co" to Additional Domains (one can't follow any links from twitter if this is blocked.)

After restart, t.co has been correctly been removed from /tmp/run/simple-adblock/dnsmasq.servers, but so has every other domain that terminates in that string. A simple grep for t.co returns nothing, and all of the following domains are now absent from dnsmasq.servers:

server=/365redirect.co/
server=/acces.streaming-direct.co/
server=/adbit.co/
server=/apicosdk.onemt.co/
server=/apicssdk.onemt.co/
server=/appleleaveit.co/
server=/coinpot.co/
server=/fpa-cdn.decrypt.co/
server=/fpa-events.decrypt.co/
server=/go.oit.co/
server=/go.recrutement.co/
server=/hmizat.co/
server=/info.ict.co/
server=/p.vibrant.co/
server=/pageredirect.co/
server=/pixel-geo.prfct.co/
server=/plausible.prufit.co/
server=/pleasewait.co/
server=/pony.bylt.co/
server=/prod.sift.co/
server=/s.inspectelement.co/
server=/sift-campaign-creatives.sift.co/
server=/stats.dashbit.co/
server=/stream-direct.co/
server=/t.co/
server=/t.omkt.co/

**Expected behavior**
Either:
Only "t.co" to be removed and the others present,
Or:
Documentation on what pattern to use to specifically exclude only the fully described domain, e.g. "^t.co$"

Cheddoleum, 2022-09-30:
(Interestingly, on reading the code I see that the pattern applied to sed adds a trailing $, which explains why ".com" domains are not removed by the above.  As an experiment I tried using "^t.co" in the luci UI, and while it complained of an error processing it, curiously enough, it worked.  The above list is back, minus "server=/t.co/".)

stangri, 2022-09-30:
Thanks for the detailed report, I'll need to update the sed parameter to allow `^domain.tld$` and `*.domain.tld$`. If you want to fix it and submit a PR, feel free to send it either here to straight to the OpenWrt packages repo.

stangri, 2022-09-30:
I've tried to address this in https://github.com/stangri/source.openwrt.melmac.net/commit/819117ac9b2e8141e15cc8769858d4fa6f325576. I'd like to experiment with combining ^ and \. into a single sed command in the future, but this should do for now.
I'd appreciate if you install the package from my repo and provide feedback after testing.

Cheddoleum, 2022-10-01:
If I understand the code change, this will not require the user to change the pattern (e.g. put "^" in front).

It seems possible that some users may have been relying on the existing behavior to handle subdomains.  Do you think that might be a concern?

Cheddoleum, 2022-10-01:
Installed from your repo and tested. It works well for the described use case. Thanks and kudos for this!

closed 2022-10-01 by Cheddoleum

posita, 2023-01-14:
> I've tried to address this in [819117a](https://github.com/stangri/source.openwrt.melmac.net/commit/819117ac9b2e8141e15cc8769858d4fa6f325576). I'd like to experiment with combining ^ and . into a single sed command in the future, but this should do for now. I'd appreciate if you install the package from my repo and provide feedback after testing.

FWIW, if PCREs are supported, I'm pretty sure you can do `(^|\.)domain\.dom$`:

``` sh
#!/usr/bin/env bash
( cat <<EOF
domain.dom
sub.domain.dom
sub.sub.domain.dom
subdomain.dom
domain.dom.dom
sub.domain.dom.dom
EOF
) | grep -E '(^|\.)domain\.dom$'
```

Prints:

```
domain.dom
sub.domain.dom
sub.sub.domain.dom
```

stangri, 2023-01-14:
> FWIW, if PCREs are supported, I'm pretty sure you can do `(^|\.)domain\.dom$`:

Thanks, the single expression I've had in the more recent version produced the same result, but the one you suggested is more elegant.