mossdef-org/source.mossdef.org #161: [pbr-iptables] no longer creates PBR_OUTPUT chain when necessary

mangajews, 2022-10-14:
I recently upgraded to pbr-iptables 0.9.8-14 and found that it no longer creates the PBR_OUTPUT chain at startup, despite having dependent policies defined.

The simplest repro:

1. start with a clean install of pbr-iptables (and enable the service)
2. enable sample policy 3 (WireGuard Server) which uses the OUTPUT chain
3. restart pbr service and observe the error from iptables

```
Activating Traffic Killswitch [✓]
Setting up routing for 'wan/pppoe-wan/100.65.0.1' [✓]
Setting up routing for 'wg0/10.1.1.2' [✓]
Routing 'WireGuard Server' via wan [✗]
Deactivating Traffic Killswitch [✓]
pbr 0.9.8-14 monitoring interfaces: wan wg0
pbr 0.9.8-14 (iptables) started with gateways:
wan/pppoe-wan/100.65.0.1
wg0/10.1.1.2 [✓]
ERROR: iptables -t mangle -A PBR_OUTPUT -g PBR_MARK_0x010000 -p udp -m multiport  --sport 51820 -m comment --comment WireGuard_Server
```

I believe this regression was introduced between 0.9.7-1 and 0.9.7-2, where pbr.nft and pbr.iptables merged and `load_package_config()` stopped calling the `append_chains_targets()` iterator which was able to expand `$usedChainsList` as needed.

```
append_chains_targets() {
	[...]
	if ! str_contains_word "$usedChainsList" "$chain"; then
		usedChainsList="$usedChainsList $chain"
```

Is this functionality somehow incompatible with nft?

stangri, 2022-10-14:
Thank you, please test `pbr 0.9.8-16`.

closed 2022-10-14 by stangri

mangajews, 2022-10-15:
Confirmed as fixed.

stangri, 2022-10-15:
Thanks again for elaborate report (making it a super easy fix) and quick test!