mossdef-org/source.mossdef.org #163: Unresponsive luci-app-pbr Page

d3adc0d3, 2022-10-28:
The fields and UI elements within the "Policy Based Routing" tab are unresponsive, i.e. they can't have text entered and items can't be added/removed from lists. 

I can start pbr by issuing `service pbr start` and this is reflected on the page but the enable/disable and start/stop buttons don't appear to work either. I've attached a screenshot although I don't expect it will be of much help. 

[image]
[image]

Please let me know what other information I can provide to help resolve this, thanks.

stangri, 2022-10-28:
Are all of the WebUI elements affected? You can't use drop-downs, etc? Which theme is that?

d3adc0d3, 2022-10-28:
The firmware image was from the official openwrt page for my Netgear router (OpenWrt 22.03.2 r19803-9a599fee93).\

I haven't really installed any additional packages besides the dependencies for pbr and pbr/luci-app-pbr

d3adc0d3, 2022-10-28:
To elaborate on my remarks regarding the UI. I am unable to interact with anything on the page.
i.e.
Can't click delete, clicking the x next to "wgserver" does nothing. Can't click the drop down menu "Output Verbosity"...

stangri, 2022-10-28:
Have you tried rebooting the router? Seems like RPCD is stuck.

d3adc0d3, 2022-10-28:
Yes, I have tried rebooting. One other oddity is that if I click the dull save&apply button I see:

`An error occurred while saving the form:

RPC call to uci/set failed with ubus code 6: Permission denied
  at handleCallReply (http://192.168.2.1/luci-static/resources/rpc.js?v=git-22.288.45147-96ec0cd:15:3)
`

Note all other openwrt pages seem to work as expected.

stangri, 2022-10-28:
I'm out of ideas, my exposure to the web ui written in javascript has been minimal, obviously something is not working for you, I don't understand what.

If you need WebUI working immediately, grab [older version here](https://github.com/stangri/repo.openwrt.melmac.net/blob/ff9743482ce81d347572328802b8dd819b8b923c/luci-app-pbr_0.9.9-1_all.ipk), I hope it will work.

Otherwise, if you use Chrome and turn on Developer Tools, you may get more detailed error information which would be helpful for me to fix this problem.

I cannot reproduce it on 22.03.2.

d3adc0d3, 2022-10-28:
I'm new to openwrt, what tab of devtools should I be monitoring to find this more detailed information.

Should I uninstall luci-app-pbr and then install the ipk file you suggested with opkg manually?

stangri, 2022-10-28:
yes, uninstall the current version and install old one from the link.

Console tab of Dev Tools may show the error in red and if you click on it, it may point you to a specific line in the pbr/overview.js file.

d3adc0d3, 2022-10-29:
Thanks for the tip.
Went ahead and checked console while loading different luci pages. Am seeing this info in the console when loading the PBR page:
`SyntaxError: Unhandled token "hex"
  at compile (http://192.168.2.1/luci-static/resources/validation.js?v=git-22.288.45147-96ec0cd:18:8)
  at __init__ (http://192.168.2.1/luci-static/resources/validation.js?v=git-22.288.45147-96ec0cd:2:203)
  at ClassConstructor (http://192.168.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:11:78)
  at create (http://192.168.2.1/luci-static/resources/validation.js?v=git-22.288.45147-96ec0cd:13:306)
  at addValidator (http://192.168.2.1/luci-static/resources/ui.js?v=git-22.288.45147-96ec0cd:347:61)
  at setUpdateEvents (http://192.168.2.1/luci-static/resources/ui.js?v=git-22.288.45147-96ec0cd:8:45)
  at bind (http://192.168.2.1/luci-static/resources/ui.js?v=git-22.288.45147-96ec0cd:11:118)
  at render (http://192.168.2.1/luci-static/resources/ui.js?v=git-22.288.45147-96ec0cd:11:13)
  at renderWidget (http://192.168.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:250:15)
  at promise callback*render (http://192.168.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:240:626)
  at renderChildren (http://192.168.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:34:36)
  at renderOptions (http://192.168.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:90:126)
  at renderUCISection (http://192.168.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:88:23)
  at render (http://192.168.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:240:215)
  at renderChildren (http://192.168.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:34:36)
  at renderContents (http://192.168.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:48:414)
  at promise callback*render (http://192.168.2.1/luci-static/resources/form.js?v=git-22.288.45147-96ec0cd:48:153)
  at render/< (http://192.168.2.1/luci-static/resources/view/pbr/overview.js?v=git-22.288.45147-96ec0cd:8:1328)
  at promise callback*render (http://192.168.2.1/luci-static/resources/view/pbr/overview.js?v=git-22.288.45147-96ec0cd:1:686)
  at promise callback*__init__ (http://192.168.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:147:232)
  at super (http://192.168.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:22:42)
  at ClassConstructor (http://192.168.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:12:11)
  at require/compileClass/< (http://192.168.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:182:111)
  at promise callback*compileClass (http://192.168.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:178:29)
  at promise callback*require (http://192.168.2.1/luci-static/resources/luci.js?v=git-22.288.45147-96ec0cd:184:112)
  at instantiateView (http://192.168.2.1/luci-static/resources/ui.js?v=git-22.288.45147-96ec0cd:352:252)
  at  (http://192.168.2.1/cgi-bin/luci/admin/services/pbr:54:7)
  at promise callback* (http://192.168.2.1/cgi-bin/luci/admin/services/pbr:53:19)`

d3adc0d3, 2022-10-29:
Installing the old version fixed it.

I went ahead and uninstalled the old version and reinstalled the new one from repo and it also works now. Thanks again for the help

closed 2022-10-29 by d3adc0d3