mossdef-org/source.mossdef.org #164: All interface options undefined luci-app-pbr

d3adc0d3, 2022-10-29:
[image]
As can be seen the interface options are all undefined when adding a new policy luci-app-pbr

stangri, 2022-10-29:
Would be nice if you included the app/luci-app version in the future, I'm assuming 0.9.9-7. Please run the following from CLI:
```
ubus -v list luci.pbr
ubus -S call luci.pbr getInitList '{"name": "pbr" }'
ubus -S call luci.pbr getInitStatus '{"name": "pbr" }'
ubus -S call luci.pbr getPlatformSupport '{"name": "pbr" }'
ubus -S call luci.pbr getGateways '{"name": "pbr" }'
ubus -S call luci.pbr getInterfaces '{"name": "pbr" }'
```

d3adc0d3, 2022-10-29:
Everything looks OK according to the getter functions, objects seem to have expected values.

Here is the output of those cmds:
`
root@OpenWrt:~# ubus -S call luci.pbr getInitList '{"name": "pbr" }'
{"pbr":{"enabled":true,"running":true}}
root@OpenWrt:~# ubus -S call luci.pbr getInitStatus '{"name": "pbr" }'
{"pbr":{"enabled":true,"running":true,"running_iptables":false,"running_nft":true,"version":"0.9.9-5","gateways":"wan/192.168.x.x ✓<br />vpn/0.0.0.0<br />","warnings":"","errors":""}}
root@OpenWrt:~# ubus -S call luci.pbr getPlatformSupport '{"name": "pbr" }'
{"pbr":{"ipset_installed":false,"nft_installed":true,"dnsmasq_installed":true,"unbound_installed":false,"dnsmasq_ipset_support":false,"dnsmasq_nftset_support":false}}
root@OpenWrt:~# ubus -S call luci.pbr getGateways '{"name": "pbr" }'
{"pbr":{"interfaces":[{"name":"wan","device_ipv4":"wan","gateway_ipv4":"192.168.x.x","device_ipv6":"wan","gateway_ipv6":"x/64\nx/64\nx/64","default":false,"action":"create","table_id":"201","mark":"0x010000","priority":"30000"},{"name":"vpn","device_ipv4":"","gateway_ipv4":"","device_ipv6":"","gateway_ipv6":"","default":false,"action":"create","table_id":"202","mark":"0x020000","priority":"29999"}]}}
root@OpenWrt:~# ubus -S call luci.pbr getInterfaces '{"name": "pbr" }'
{"pbr":{"interfaces":["wan","wan6","vpn"]}}
`

d3adc0d3, 2022-10-29:
Seems as though running those cmds has updated the UI. I suppose this would be considered resolved for me, am curious if it was just a fluke or if there's some reason the getX methods had to be called.

d3adc0d3, 2022-10-29:
Now am seeing a new bug. I've provided nft versions and error context.

When trying to add a rule for routing a specific port to VPN interface I see:
[image]

When running that cmd manually from terminal I see:
`
root@OpenWrt:~#  nft 'add rule inet fw4 pbr_prerouting tcp sport {1234} dport {0-65535} goto
pbr_mark_0x020000 comment "test"'
Error: syntax error, unexpected dport, expecting end of file or newline or semicolon
add rule inet fw4 pbr_prerouting tcp sport {1234} dport {0-65535} goto pbr_mark_0x020000 comment "test"
`

Here is info on nft packages:
`
root@OpenWrt:~# opkg list-installed | grep nft
kmod-nft-core - 5.10.146-1
kmod-nft-fib - 5.10.146-1
kmod-nft-nat - 5.10.146-1
`

stangri, 2022-10-29:
> expecting end of file or newline or semicolon

Based on above, have you tried `0:65535`?

d3adc0d3, 2022-10-29:
Here is the output of that. Seems like some issue of the version of nft I have not expecting dport arg to be supplied in this fashion?

If I run which nft I see:
`
root@OpenWrt:~# which nft
/usr/sbin/nft
`

`
root@OpenWrt:~# nft 'add rule inet fw4 pbr_prerouting tcp sport {1234} dport {0:
65535} goto pbr_mark_0x020000 comment "test"'
Error: syntax error, unexpected dport, expecting end of file or newline or semicolon
add rule inet fw4 pbr_prerouting tcp sport {1234} dport {0:65535} goto pbr_mark_0x020000 comment "test"
                                                  ^^^^^
`

stangri, 2022-10-29:
Got to the computer to test, it's a bug in the code, I'll push an update later, thank you for catching this!

d3adc0d3, 2022-10-29:
Your welcome, happy to help

stangri, 2022-10-29:
For a quicker fix, remove line 1045 `		unset proto` from `/etc/init.d/pbr`.

closed 2022-10-29 by stangri