mossdef-org/source.mossdef.org #165: [pbr]

d3adc0d3, 2022-11-02:
Seems like there is some issue with nft add rule cmd gen from pbr cfg file. Perhaps missing "ip" prior to daddr?

luci-app-pbr error
[image]

service restart output
[image]

Issuing cmd that results in error upon service restart
[image]

Tried adding ip prior to daddr
[image]

stangri, 2022-11-02:
Please test with `pbr 0.9.9-12` I just pushed to my repo.

This is likely same issue as what I fixed with the protocols, I don't understand why it's happening, I believe I've tested before and `nft` was complaining about duplicates, now it requires them.

stangri, 2022-11-02:
Thanks again for testing/reporting and sorry I forgot to reference the issue in the commit!

closed 2022-11-02 by stangri