mossdef-org/source.mossdef.org #166: [pbr] wish: populate ipset automatically

bonuzzz, 2022-11-02:
I have adguardhome as primary dns server, so dnsmasq doesn't populate ipset list, because it doesn't receive dns requests front clients. To populate ipset I have script, but probably pbr would have internal method to do it.

```
#!/bin/bash
sed -e "s/.*\://" /tmp/dnsmasq.d/pbr |

while IFS=' ' read -r line; do
	nslookup $line 127.0.0.1:5553 >/dev/null
done
```
5553 is port of dnsmasq

stangri, 2022-11-02:
@bonuzzz Can you please help me understand how is it different from setting `resolver_set` to `none`?

bonuzzz, 2022-11-02:
It's not the same, because rules don't affect on subdomains there. But just noticed my script doesn't affect too.

The problem now is I want to use dnsmasq.ipset resolver, I have to use dnsmasq as primary dns server, otherwise dnsmasq doesn't received any requests from clients and ipset list doesn't populated.
I tried to solve it with nslookup or dig to send request through dnsmasq speсially. But I forgot about subdomains.

stangri, 2022-11-02:
Right, so to confirm you want `pbr` to populate ipsets just line `dnsmasq` does?

bonuzzz, 2022-11-02:
Yes.
When I use adguardhome as upstream dns server, it has router's ip as source of requests. This is main problem, which I want to solve.

stangri, 2022-11-02:
> Right, so to confirm you want `pbr` to populate ipsets just line `dnsmasq` does?

Yeah, that's not something I would consider for `pbr`.

If you can convince AdGuardHome people to populate ipsets or nft sets, given a specific file similar to dnsmasq, I can consider implementing support for AGH in the `resolver_set` option.

I'm going to close this as `won't fix`.

closed 2022-11-02 by stangri

bonuzzz, 2022-11-02:
Like this one?
https://github.com/AdguardTeam/AdGuardHome/commit/88812f05f56438151e70856250b1db6de62eaef6

stangri, 2022-11-02:
Yes, that's a great discovery, exactly like that.

I can look into if you can provide a sample of working configuration -- the main yaml config file and the ipset_file.