mossdef-org/source.mossdef.org #167: [pbr][wish] support adguardhome.ipset as resolver_set option

bonuzzz, 2022-11-03:
Steps in adguardhome side:
1. in config file add to parameter ipset_file path to ipset file. Example: ipset_file: "/tmp/ipset"

Steps in pbr side:
1. create ipset lists (ipset create hash:ip) (/tmp/ipset)
2. create new file with ipset list. format   domain/ipset_name
Example:
```
themoviedb.org/pbr_vpn_4_dst_ip_cfg0f6ff5 
tmdb.org/pbr_vpn_4_dst_ip_cfg0f6ff5 
```
3. reload adguardhome service.   /etc/init.d/adguardhome reload or /usr/bin/AdGuardHome -s reload

I remember current agh version in openwrt doesn't support ipset file yet, but I write this manual until I remember how to do it.
Subdomains work with agh like in dnsmasq.
you will get error at adguadhome start  if you write ipset in agh ipset file, which doesn't exist in ipset list
The logic here is the same as in dnsmasq except different file format

stangri, 2022-11-16:
Just linking the previous conversation for posterity: https://github.com/stangri/source.openwrt.melmac.net/issues/166

stangri, 2022-11-16:
> Steps in pbr side:
> 
> 1. create ipset lists (ipset create hash:ip) (/tmp/ipset)
> 2. create new file with ipset list. format   domain/ipset_name
>    Example:
> 
> ```
> themoviedb.org/pbr_vpn_4_dst_ip_cfg0f6ff5 
> tmdb.org/pbr_vpn_4_dst_ip_cfg0f6ff5 
> ```

From what I gathered from AGH docs, all their config/ipset files are yaml, not plain text, so it's going to be more complicated than that and that's why I need someone with working AGH 0.107.13 and later to run some tests first.

I've already refactored a lot of `pbr` code to support resolver sets other that dnsmasq.ipset and dnsmasq.nftset, but I need an example of the working config before I can implement support for AGH.ipset.

bonuzzz, 2022-11-16:
config file is yaml, but ipset file is plain text. I'm using this method since our last conversation. I made dirty workaround for now: replaced in init script to readonly dnsmasqFile="/var/ipset_agh" and echo "${param}/${ipset4}${ipset6:+,$ipset6}" >> "$dnsmasqFile" && ipv4_error=0 and restart adguardhome with dnsmasq

bonuzzz, 2022-11-16:
I have where to test if needed

stangri, 2022-11-16:
Can you post your adguardhome config file?

bonuzzz, 2022-11-16:
[AdGuardHome.zip](https://github.com/stangri/source.openwrt.melmac.net/files/10026254/AdGuardHome.zip)

stangri, 2022-11-17:
Is there an `adguardhome` binary on the system? Does it report version and/or features somehow?

stangri, 2022-11-17:
Install `pbr 0.9.9-28` and `luci-app-pbr 0.9.9-28` and clear your browser cache (otherwise it won't pick up changes in the luci app), you should be able to select AdGuardHome ipset from the resolver set options.

closed 2022-11-17 by stangri

stangri, 2022-11-18:
So are you trying to get it working with adguardhome.ipset option or any other option for resolver_set? I don't understand why you have pbr using nft?

bonuzzz, 2022-11-18:
[image]
I have clean install only with adguardhome and pbr, luci-app-pbr

command -v AdGuardHome
/usr/bin/AdGuardHome/AdGuardHome

ls -la /etc/AdGuardHome.yaml
-rw-r--r--    1 root     root          4921 Nov 18 19:31 /etc/AdGuardHome.yaml

bonuzzz, 2022-11-18:
Adguardhome doesn't allow comments in ipset file.
Example: iplocation.net/pbr_france_4_dst_ip_cfg096ff5

What about compatibility check, I thought it was version compare condition, but it have to cut all check from both files.
As I see, both init and luci files have this check. Is it possible to show in /etc/init.d/pbr status what resolver sets currently supported without using luci?

bonuzzz, 2022-11-18:
Init script has issue only with ipset file format. I wrote about it before.
Luci (luci.pbr) script has compatibility check issue.

I reinstalled everything and tested only with pbr package without luci

bonuzzz, 2022-11-18:
I attached my modifications. Maybe you will have any ideas what's wrong there
[pbr.zip](https://github.com/stangri/source.openwrt.melmac.net/files/10044638/pbr.zip)

stangri, 2022-11-18:
Have you installged AGH as an OpenWrt package or manually? What's the output of `AdGuardHome --version`?

bonuzzz, 2022-11-18:
Manually
AdGuard Home, version v0.107.18

stangri, 2022-11-18:
What's the output in CLI?

To confirm the support for `adguardhome.ipset` option, it does the following:
```sh
readonly agh="$(command -v AdGuardHome)"
readonly aghConfigFile='/etc/AdGuardHome.yaml'

					if [ -n "$agh" ] && [ -s "$aghConfigFile" ]; then
						agh_version="$($agh --version | sed 's|AdGuard Home, version v0\.\(.*\)|\1|')"
						if is_greater_or_equal "$agh_version" '107.13'; then
							resolver_set_supported='true'
echo 'success'!
							return 0
						else
							warningSummary="${warningSummary}${_WARNING_}: Installed AdGuardHome version ($agh_version) doesn't support 'ipset_file' option!\\n"
						fi
					else
						warningSummary="${warningSummary}${_WARNING_}: Resolver set support (${resolver_set}) is enabled in $packageName, but this resolver set is not supported on this system!\\n"
						return 1
					fi
```

Can you run that piece of code on your system?

stangri, 2022-11-18:
> If needed, I could install adguardhome ipk from master

That's what I'm testing with, but the necessary version hasn't been backported to 22.03 yet.

bonuzzz, 2022-11-18:
```
#!/bin/sh 

readonly agh="$(command -v AdGuardHome)"
readonly aghConfigFile='/etc/AdGuardHome.yaml'
is_greater_or_equal() { test "$(printf '%s\n' "$@" | sort -V | head -n 1)" = "$2"; }

					if [ -n "$agh" ] && [ -s "$aghConfigFile" ]; then
						agh_version="$($agh --version | sed 's|AdGuard Home, version v0\.\(.*\)|\1|')"
						if is_greater_or_equal "$agh_version" '107.13'; then
							resolver_set_supported='true'
echo 'success'!
						else
							echo "Installed AdGuardHome version ($agh_version) doesn't support 'ipset_file' option!\\n"
						fi
					else
						echo "Resolver set support (${resolver_set}) is enabled in $packageName, but this resolver set is not supported on this system!\\n"
					
					fi
```

root@dacha:~# sh 4.sh
: not found2:
4.sh: line 19: syntax error: unexpected "fi" (expecting "then")

bonuzzz, 2022-11-18:
I deleted previous message. It was my fault there.
Script said success

stangri, 2022-11-18:
Well, that's exact code is working on pbr start. I don't understand why it's working on its own, but not when part of the script.

stangri, 2022-11-18:
Also, the screenshot you've posted 6 hours ago seems to confirm that WebUI has properly detected AGH ipset suppport.

bonuzzz, 2022-11-18:
I've installed latest ipk from snapshot and two news. One bad news is new config filename is /etc/adguardhome.yaml without capital letters. Good news is after edit filename in script pbr finds adguardhome ipset

stangri, 2022-11-18:
> upd: seems the filename is never changed in openwrt repository.

Yeah, looks like it's always been lower case: https://github.com/openwrt/packages/blob/openwrt-22.03/net/adguardhome/Makefile#L48. Weird that the new package installation didn't bring the yaml file for me.

I've copied your config from the zip-file you've posted above and it was camelized, so I never checked the package sources what the proper name is.

bonuzzz, 2022-11-20:
It's me again:)
1. You forgot to change filename in luci.pbr https://github.com/stangri/source.openwrt.melmac.net/blob/master/luci-app-pbr/root/usr/libexec/rpcd/luci.pbr#L23
2.  Additional dependencies for adh resolver ipset: ipset and iptables-nft
3. Probably it's better to use restart adh than reload, because seems agh doesn't read ipset file for changes during reload (not sure here)
4. Could you add sed to remove ipset filepath in agh config when stop using adh as resolver set?
5. When using agh resolver set, but not any rules are activated, ipset file doesn't created and agh stopped with error. Ipset file for agh should be always created , even empty

stangri, 2022-11-20:
@bonuzzz thanks for the continuing feedback and testing!

> 1. You forgot to change filename in luci.pbr https://github.com/stangri/source.openwrt.melmac.net/blob/master/luci-app-pbr/root/usr/libexec/rpcd/luci.pbr#L23

Fixed in 0.9.9-36.

> 2. Additional dependencies for adh resolver ipset: ipset and iptables-nft

The ipset and iptables should be dependencies for the `pbr-iptables`, as the `pbr` has been designed for fw4/nft, these are not really required. AGH should bring those in actually.

> 3. Probably it's better to use restart adh than reload, because seems agh doesn't read ipset file for changes during reload (not sure here)

While there is the `reload` parameter to the `resolver()` function, it's not currently used, the `resolver 'restart'` is always used than the hashes of the old and new ipset/nftset files do not match.

> 4. Could you add sed to remove ipset filepath in agh config when stop using adh as resolver set?

Fixed in 0.9.9-36.

> 5. When using agh resolver set, but not any rules are activated, ipset file doesn't created and agh stopped with error. Ipset file for agh should be always created , even empty

Fixed in 0.9.9-36.

bonuzzz, 2022-11-20:
About packages I wrote just for documentation if someone wants to use pbr with adguardhome. I use pbr package in 22.03.
What about the issue when adguardhome doesn't populate ipset. I had this at starting system. So, solution is simple: adguardhome should NOT be enabled in autostart in openwrt. Pbr starts it anyway, but in this case ipset population works immediately.

As luci-app-adguardhome ( https://github.com/kongfl888/luci-app-adguardhome ) is mentioned on adguaghome site and this is simple enough method to use actual agh version on openwrt, I will write the difference in config and filenames between it and package from repository. Maybe you could add support both variants of agh usage.

stangri, 2022-11-20:
> What about the issue when adguardhome doesn't populate ipset. I had this at starting system. So, solution is simple: adguardhome should NOT be enabled in autostart in openwrt. Pbr starts it anyway, but in this case ipset population works immediately.

You mean that AGH can't start if the ipset_file is empty? It should be addressed in the AGH OpenWrt package init script, it should touch that file before launching the binary.

bonuzzz, 2022-11-20:
No. If agh is enabled in autostart, so everything starts as usual and ipset file for agh is exist with rules, but agh doesn't populate ipset itself with ips. To get agh start populating ipset, I have to restart agh.

Some more word about iptables-nft. If I don't install this package I have this error on screenshot
I use pbr package and luci-app-pbr on 22.03
[image]