mossdef-org/source.mossdef.org #168: [PBR] Issue: Service starts with "ip: bad line 11: 1 tokens found, 2 needed"

Soberia, 2022-11-17:
OpenWrt 22.03.0 r19685-512e76967f
BusyBox v1.35.0
PBR 0.9.9-30 (nft)

None of the configured policies actually work for me anymore.
I had no problem before updating form 0.9.9-16 to 0.9.9-28.

<details><summary><strong>Config</strong></summary>

```
config pbr 'config'
        option enabled '1'
        option verbosity '2'
        option strict_enforcement '1'
        option resolver_set 'none'
        option ipv6_enabled '0'
        option boot_timeout '30'
        option rule_create_option 'add'
        option procd_reload_delay '1'
        option webui_show_ignore_target '0'
        list webui_supported_protocol 'all'
        list webui_supported_protocol 'tcp'
        list webui_supported_protocol 'udp'
        list webui_supported_protocol 'tcp udp'
        list webui_supported_protocol 'icmp'
        list supported_interface 'openvpn'

config include
        option path '/usr/share/pbr/pbr.user.aws'

config include
        option path '/usr/share/pbr/pbr.user.netflix'

config policy
        option dest_addr '10.0.0.0/8'
        option interface 'ignore'
        option name 'ignore-local'
        option enabled '0'

config policy
        option name 'vpn_lan'
        option src_addr '10.0.2.0/24'
        option interface 'openvpn'

config policy
        option name 'wg_server'
        option src_addr '10.0.4.0/24'
        option enabled '0'
        option interface 'openvpn'

config policy
        option name 'vpn_redirect'
        option src_port '59096 7468 51820'
        option proto 'udp'
        option chain 'output'
        option interface 'openconnect'
        option enabled '0'
```
</details>

<details><summary><strong>Log:</strong></summary>

```
root@XMR3G:/home/soberia# service pbr start
Activating traffic killswitch [✓]
Setting up routing for 'wan/pppoe-wan/172.20.0.32' ip: bad line 10: 1 tokens found, 2 needed
ip: bad line 11: 1 tokens found, 2 needed
ip: bad line 12: 1 tokens found, 2 needed
ip: bad line 13: 1 tokens found, 2 needed
ip: bad line 14: 1 tokens found, 2 needed
[... 1283 lines cut ...]
Routing 'vpn_lan' via openvpn [✓]
Deactivating traffic killswitch [✓]
pbr 0.9.9-30 monitoring interfaces: wan wg_client wg_server wg_client_2 wg_client_3 openvpn openconnect pptp
pbr 0.9.9-30 (nft) started with gateways:
wan/pppoe-wan/172.20.0.32 [✓]
wg_client/0.0.0.0
wg_server/10.0.4.1
wg_client_2/0.0.0.0
wg_client_3/0.0.0.0
openvpn/0.0.0.0
openconnect/0.0.0.0
pptp/0.0.0.0
```
</details>

</br>
</br>

**Irrelevant UI problems:**

- There is no longer an `ignore` option in `Interface` dropdown when adding policies with LuCI.
[image]

- Infinite loading warning when restarting the service with LuCI.
[image]

stangri, 2022-11-17:
> **Irrelevant UI problems:**
> 
> * There is no longer an `ignore` option in `Interface` dropdown when adding policies with LuCI.

From your config: `        option webui_show_ignore_target '0'`.

> ip: bad line 11: 1 tokens found, 2 needed

What's the output of `opkg list-installed | grep ip`?

stangri, 2022-11-17:
Also, output of `service pbr status` and your network config.

Soberia, 2022-11-17:
I just update from `22.03.0` to `22.03.2` right now and everything was fine until updating `BusyBox`.
The problem should be from the shipped `ip` package.

Sorry, last time I updated all of my packages including `BusyBox` and `PBR` and didn't notice that. I should've mentioned it.

<details><summary><strong>ip route</strong></summary>

```
root@XMR3G:/home/soberia# ip route
default via 172.20.0.32 dev pppoe-wan
10.0.0.0/24 dev wan scope link  src 10.0.0.10
10.0.1.0/24 dev br-lan scope link  src 10.0.1.1
10.0.2.0/24 dev br-lan-vpn scope link  src 10.0.2.1
10.0.3.0/24 dev wlan0-2 scope link  src 10.0.3.1
10.0.4.0/24 dev wg_server scope link  src 10.0.4.1
10.8.0.0/24 dev tun0 scope link  src 10.8.0.2
172.20.0.32 dev pppoe-wan scope link  src ***
```
</details>

<details><summary><strong>opkg list-installed | grep ip</strong></summary>

```
root@XMR3G:/home/soberia# opkg list-installed | grep ip
ip-full - 5.15.0-3
iptables-mod-ipopt - 1.8.7-7
iptables-zz-legacy - 1.8.7-7
kmod-ipt-core - 5.10.146-1
kmod-ipt-ipopt - 5.10.146-1
kmod-iptunnel - 5.10.146-1
kmod-nf-ipt - 5.10.146-1
libip4tc2 - 1.8.7-7
libip6tc2 - 1.8.7-7
libiptext0 - 1.8.7-7
libiptext6-0 - 1.8.7-7
libjson-script20220515 - 2022-05-15-d2223ef9-1
luci-lib-ip - git-20.250.76529-62505bd
luci-proto-ipv6 - git-21.148.48881-79947af
odhcpd-ipv6only - 2022-03-22-860ca900-1
resolveip - 2
sqm-scripts - 1.5.2-1
vpnc-scripts - 20151220-2
```
</details>

<details><summary><strong>service pbr status</strong></summary>

```
root@XMR3G:/home/soberia# service pbr status
============================================================
pbr - environment
pbr 0.9.9-30 running on OpenWrt 22.03.2. WAN (IPv4): wan/pppoe-wan/172.20.0.32.
============================================================
Dnsmasq version 2.86  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
============================================================
pbr chains - policies
        chain pbr_forward {
        }
        chain pbr_input {
        }
        chain pbr_output {
        }
        chain pbr_prerouting {
                ip daddr @pbr_wg_client_4_dst_ip_user goto pbr_mark_0x020000
                ip saddr @pbr_wg_client_4_src_ip_user goto pbr_mark_0x020000
                ether saddr @pbr_wg_client_4_src_mac_user goto pbr_mark_0x020000
                ip daddr @pbr_wg_client_2_4_dst_ip_user goto pbr_mark_0x040000
                ip saddr @pbr_wg_client_2_4_src_ip_user goto pbr_mark_0x040000
                ether saddr @pbr_wg_client_2_4_src_mac_user goto pbr_mark_0x040000
[... 235 lines cut ...]
ip: bad line 14: 1 tokens found, 2 needed
ip: bad line 16: 1 tokens found, 2 needed
ip: bad line 17: 1 tokens found, 2 needed
IPv4 table 28 route: unreachable default
IPv4 table 28 rule:
ip: bad line 1: 1 tokens found, 2 needed
ip: bad line 2: 1 tokens found, 2 needed
ip: bad line 3: 1 tokens found, 2 needed
ip: bad line 4: 1 tokens found, 2 needed
ip: bad line 5: 1 tokens found, 2 needed
ip: bad line 6: 1 tokens found, 2 needed
ip: bad line 7: 1 tokens found, 2 needed
ip: bad line 8: 1 tokens found, 2 needed
ip: bad line 9: 1 tokens found, 2 needed
ip: bad line 10: 1 tokens found, 2 needed
ip: bad line 11: 1 tokens found, 2 needed
ip: bad line 12: 1 tokens found, 2 needed
ip: bad line 13: 1 tokens found, 2 needed
ip: bad line 14: 1 tokens found, 2 needed
ip: bad line 16: 1 tokens found, 2 needed
ip: bad line 17: 1 tokens found, 2 needed
IPv4 table 29 route: unreachable default
IPv4 table 29 rule:
```
</details>

stangri, 2022-11-18:
If you have upgraded busybox it probably reset the soft link to up full binary which is required by pbr. 

Completely remove and reinstall ip-full.

Soberia, 2022-11-18:
Reinstalling the `ip-full` solve the problem. Thanks

closed 2022-11-18 by Soberia

stangri, 2022-11-18:
Versions 0.9.9-31 and later will force-use `ip-full` even if the softlink has ben overwritten by a busybox.