mossdef-org/source.mossdef.org #169: [PBR] Issue: Database /etc/iproute2/rt_tables is corrupted

Soberia, 2022-11-18:
OpenWrt 22.03.2
iproute2 5.15.0
PBR 0.9.9-32 (nft)

I have a nested VPN setup. WireGuard client connects through another OpenConnect tunnel to reach the internet.
After running `pbr` service, I have to restart the WireGuard interface manually because it initially tries to connect through the WAN interface (`pbr` service hasn't started just yet) and after `pbr` sets the policies, WireGuard still can't complete the handshake (this protocol is blocked in my region). Therefor I have to restart it manually for the traffic goes from the second tunnel. But after each restart, `pbr` service also restarts and messed up the ip rules. I have `Database /etc/iproute2/rt_tables is corrupted` in my logs. Deleting this file and restarting the service makes no difference.

<details><summary><strong>config</strong></summary>

```
soberia@XMR3G:~$ sudo cat /etc/config/pbr

config pbr 'config'
        option enabled '1'
        option verbosity '2'
        option strict_enforcement '1'
        option resolver_set 'none'
        option ipv6_enabled '0'
        option boot_timeout '30'
        option rule_create_option 'add'
        option procd_reload_delay '1'
        option webui_show_ignore_target '1'
        list webui_supported_protocol 'all'
        list webui_supported_protocol 'tcp'
        list webui_supported_protocol 'udp'
        list webui_supported_protocol 'tcp udp'
        list webui_supported_protocol 'icmp'
        list supported_interface 'openvpn'

config include
        option path '/usr/share/pbr/pbr.user.aws'

config include
        option path '/usr/share/pbr/pbr.user.netflix'

config policy
        option dest_addr '10.0.0.0/8'
        option interface 'ignore'
        option name 'ignore-local'

config policy
        option name 'vpn_lan'
        option src_addr '10.0.2.0/24'
        option interface 'wg_client'

config policy
        option name 'wg_server'
        option src_addr '10.0.4.0/24'
        option interface 'wg_client'

config policy
        option name 'vpn_redirect'
        option proto 'udp'
        option chain 'output'
        option interface 'openconnect'
        option dest_port '59096 7468 51820'
```
</details>

<details><summary><strong>service pbr status</strong></summary>

```
soberia@XMR3G:~$ sudo service pbr status
============================================================
pbr - environment
pbr 0.9.9-32 running on OpenWrt 22.03.2. WAN (IPv4): wan/pppoe-wan/172.20.0.32.
============================================================
Dnsmasq version 2.86  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
============================================================
[... 338 lines cut ...]
33
34
35
36
37
38
39
40
41
42
43
44
45
1 pbr_wan
46
47
48 pbr_wg_client
3 pbr_wg_server
49 pbr_wg_client_2
50 pbr_wg_client_3
6 pbr_openvpn
7 pbr_openconnect
8 pbr_pptp
```
</details>

stangri, 2022-11-18:
It looks like `/etc/iproute2/rt_tables` was molested before `pbr` has started modifying it. `pbr` can't fix what was broken before.

Soberia, 2022-11-18:
I had three different interfaces with `wg_client` in their names. ( `wg_client`,  `wg_client_1`,  `wg_client_2`)
After temporarily deleting the last two, table ids correctly mapped to the interface names in `/etc/iproute2/rt_tables`.

I'm not know much about the source, but I think regex-based tools like `sed` or `grep` should be leveraged more precisely when dealing with names. Like in [`get_rt_tables_id()`](https://github.com/stangri/source.openwrt.melmac.net/blob/master/pbr/files/etc/init.d/pbr.init#L231) and [`interface_routing()`](https://github.com/stangri/source.openwrt.melmac.net/blob/master/pbr/files/etc/init.d/pbr.init#L1524).

stangri, 2022-12-19:
@Soberia you're welcome to submit PR to improve the code!

stangri, 2023-01-14:
@Soberia I believe I've addressed it in 1.0.1-12. The PR into official repo is still pending, you can install the package or build it yourself from my packages or source repos.

closed 2023-01-14 by stangri