mossdef-org/source.mossdef.org #176: [pbr] Issue:

st7105, 2022-12-19:
**Describe the bug**
A clear and concise description of what the bug is.

**Your configs**

1. ```/etc/config/dhcp```

   ```sh

config dnsmasq
        option domainneeded '1'
        option localise_queries '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option localservice '1'
        option ednspacket_max '1232'
        option rebind_protection '0'
        option noresolv '1'
        option localuse '1'
        list server '127.0.0.1#5453'
        list server '0::1#5453'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv4 'server'
        option ndp 'relay'
        list ra_flags 'none'
        option ra 'server'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

config dhcp 'wan6'
        option interface 'wan6'
        option master '1'
        option ra 'relay'
        option ndp 'relay'

   ```

2. ```/etc/config/network```

   ```sh
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd57:ac82:b817::/48'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'
        list ports 'eth2'
        list ports 'eth3'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option netmask '255.255.255.0'
        option ipaddr '192.168.1.1'
        option ip6assign '64'
        list ip6class 'wan6'

config interface 'wan'
        option device 'eth0'
        option proto 'dhcp'
        option ip4table 'pbr_wan'
        option ip6table 'pbr_wan'

config interface 'wan6'
        option device 'eth0'
        option proto 'dhcpv6'
        option peerdns '0'
        option reqaddress 'try'
        option reqprefix 'auto'
        option ip4table 'pbr_wan'
        option ip6table 'pbr_wan'

config interface 'wg0'
        option proto 'wireguard'
        option peerdns '0'
        option defaultroute '0'
        ...
        option ip4table 'pbr_wg0'
        option ip6table 'pbr_wg0'

config wireguard_wg0
        ...
        list allowed_ips '0.0.0.0/0'

config rule 'default'
        option lookup 'pbr_wan'
        option priority '80000'

config rule6 'default6'
        option lookup 'pbr_wan'
        option priority '80000'

   ```

3. ```/etc/config/firewall```

   ```sh
config defaults
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option synflood_protect '1'
        option flow_offloading '1'
        option flow_offloading_hw '1'

config zone
        option name 'lan'
        list network 'lan'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'ACCEPT'

config zone
        option name 'wan'
        option input 'REJECT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option masq '1'
        option mtu_fix '1'
        list network 'wan'
        list network 'wan6'
        list network 'wg0'

config forwarding
        option src 'lan'
        option dest 'wan'

config include 'pbr'
        option fw4_compatible '1'
        option type 'script'
        option path '/usr/share/pbr/pbr.firewall.include'

   ```

4. ```/etc/config/vpn-policy-routing```

   ```sh
config pbr 'config'
        option verbosity '2'
        option src_ipset '0'
        option dest_ipset '0'
        list ignored_interface 'vpnserver wgserver'
        option procd_reload_delay '1'
        option webui_enable_column '0'
        option webui_protocol_column '0'
        option webui_chain_column '0'
        option webui_show_ignore_target '0'
        list webui_supported_protocol 'tcp'
        list webui_supported_protocol 'udp'
        list webui_supported_protocol 'tcp udp'
        list webui_supported_protocol 'icmp'
        list webui_supported_protocol 'all'
        option enabled '1'
        option webui_sorting '0'
        option boot_timeout '60'
        option strict_enforcement '0'
        option ipv6_enabled '0'
        option rule_create_option 'add'
        option resolver_set 'dnsmasq.nftset'

config include
        option path '/etc/pbr.st7105.user'
        option enabled '1'

   ```

**To Reproduce**
Steps to reproduce the behavior:
1. /etc/init.d/pbr reload
4. See error

**Expected behavior**
The problem occurred after updating to 1.0.0-4+

1. Output of ```/etc/init.d/vpn-policy-routing reload``` with ```verbosity``` set to 2:

   ```sh
/etc/init.d/pbr reload -d
Activating traffic killswitch [✓]
Setting up routing for 'wan/eth0/xxx' [✗]
Setting up routing for 'wg0/10.9.0.129' [✗]
Deactivating traffic killswitch [✓]
Running /etc/pbr.st7105.user [✓]
pbr 1.0.1-1 monitoring interfaces: wan wg0 
ERROR: Failed to set up 'wan/eth0/xxx'!
ERROR: Failed to set up 'wg0/10.9.0.129'!
ERROR: Failed to set up any gateway!
   ```

stangri, 2022-12-19:
Fixed in 1.0.1-1.

st7105, 2022-12-19:
@stangri No, the problem is still relevant.

`root@OpenWrt:~# /etc/init.d/pbr version
1.0.1-1
root@OpenWrt:~# /etc/init.d/pbr reload
Activating traffic killswitch [✓]
Setting up routing for 'wan/eth0/188.243.212.1' [✗]
Setting up routing for 'wg0/10.9.0.129' [✗]
Deactivating traffic killswitch [✓]
Running /etc/pbr.st7105.user [✓]
pbr 1.0.1-1 monitoring interfaces: wan wg0 
ERROR: Failed to set up 'wan/eth0/188.243.212.1'!
ERROR: Failed to set up 'wg0/10.9.0.129'!
ERROR: Failed to set up any gateway!`

st7105, 2022-12-19:
```sh
nftset create_user_set wan src mac user  0x010000
+ local 'command=create_user_set' 'iface=wan' 'target=src' 'type=mac' 'uid=user' 'comment=' 'param=0x010000' 'mark=0x010000'
+ local nftset4 nftset6 i param4 param6
+ local 'ipv4_error=1' 'ipv6_error=1'
+ nftset4=pbr_wan_4_src_mac_user
+ nftset6=pbr_wan_6_src_mac_user
+ '[' -x /usr/sbin/nft ]
+ '[' 22 -gt 255 ]
+ nft4 add set inet fw4 pbr_wan_4_src_mac_user '{ type ether_addr; flags interval; auto-merge; policy memory; comment ""; }'
+ '[' -x /usr/sbin/nft ]
+ /usr/sbin/nft add set inet fw4 pbr_wan_4_src_mac_user '{ type ether_addr; flags interval; auto-merge; policy memory; comment ""; }'
+ ipv4_error=0
+ nft6 add set inet fw4 pbr_wan_6_src_mac_user '{ type ether_addr; flags interval; auto-merge; policy memory; comment ""; }'
+ '[' -n  ]
+ return 0
+ ipv6_error=0
+ nft add rule inet fw4 pbr_prerouting ether saddr @pbr_wan_4_src_mac_user goto pbr_mark_0x010000
+ '[' -x /usr/sbin/nft ]
+ /usr/sbin/nft add rule inet fw4 pbr_prerouting ether saddr @pbr_wan_4_src_mac_user goto pbr_mark_0x010000
+ ipv4_error=0
+ nft add rule inet fw4 pbr_prerouting ether saddr @pbr_wan_6_src_mac_user goto pbr_mark_0x010000
+ '[' -x /usr/sbin/nft ]
+ /usr/sbin/nft add rule inet fw4 pbr_prerouting ether saddr @pbr_wan_6_src_mac_user goto pbr_mark_0x010000
+ '[' -z  ]
+ ipv6_error=1
+ return 1
+ s=1
+ return 1
+ state add errorSummary errorFailedSetup wan/eth0/xxx
+ local 'action=add' 'param=errorSummary' 'value=errorFailedSetup'
+ shift 3
+ local 'extras=wan/eth0/188.243.212.1'
+ local line error_id error_extra label
+ eval echo '$errorSummary'
+ echo
+ line=
+ eval 'errorSummary=${line:+$line#}${value}${extras:+ $extras}'
+ errorSummary='errorFailedSetup wan/eth0/xxx'
+ output_fail
+ s=1
+ output 1 '\033[0;31m\xe2\x9c\x97\033[0m'
+ local msg memmsg logmsg
```

stangri, 2022-12-19:
Can you post lines 528-532 from the `/etc/init.d/pbr` on your router?

st7105, 2022-12-19:
@stangri 
```sh
	if [ "$ipv4_error" -eq '0' ] || [ "$ipv6_error" -eq '0' ]; then
		return 0
	else
		return 1
	fi
```

stangri, 2022-12-19:
I've tried enabling one user file on my router and still can't reproduce. To confirm or reject my suspicion, can you please disable the user file on your router and run reload?

stangri, 2022-12-19:
Can you please try `1.0.1-2` from my repo?

st7105, 2022-12-19:
@stangri Okay, just a moment.

closed 2022-12-19 by st7105