mossdef-org/source.mossdef.org #177: [wireshark-helper] Issue: cannot find dependency luci-lua-runtime for luci-app-wireshark-helper

lolyinseo, 2023-01-08:
error during installation luci-app-wireshark-helper 

 * pkg_hash_check_unresolved: cannot find dependency luci-lua-runtime for luci-app-wireshark-helper
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-wireshark-helper found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-wireshark-helper.

openwrt-22.03.2
ramips-mt7620

stangri, 2023-01-08:
I don't know why it's missing from OpenWrt repo, I've made a forum post about it: https://forum.openwrt.org/t/luci-lua-runtime-missing-from-openwrt-22-03/147779. 

There are still many packages in OpenWrt luci repo depending on either `luci-compat` or `luci-lua-runtime`, so I'm guessing it's a glitch with the OpenWrt repo.

stangri, 2023-01-09:
@lolyinseo elaborate on the OpenWrt 22.03 build please and the configured repos for opkg on your router. From the forum discussion it looks like the `luci-lua-runtime` should not be a dependency of `luci-compate` on 22.03.

lolyinseo, 2023-01-09:
It`s official build for Xiaomi Mi Minin https://openwrt.org/toh/xiaomi/miwifi_mini
There are no special changes of repos conf, I just installed the firmware and immediately after that I tried to install wireshark-helper

stangri, 2023-01-09:
Please run `opkg update` and post the output of:
```
cat /etc/os-release
cat /etc/opkg/distfeeds.conf
cat /etc/opkg/customfeeds.conf
opkg info luci-compat
opkg info luci-app-wireshark-helper
opkg whatdepends luci-compat
```

lolyinseo, 2023-01-09:
root@OpenWrt:~# cat /etc/os-release
    NAME="OpenWrt"
    VERSION="22.03.2"
    ID="openwrt"
    ID_LIKE="lede openwrt"
    PRETTY_NAME="OpenWrt 22.03.2"
    VERSION_ID="22.03.2"
    HOME_URL="https://openwrt.org/"
    BUG_URL="https://bugs.openwrt.org/"
    SUPPORT_URL="https://forum.openwrt.org/"
    BUILD_ID="r19803-9a599fee93"
    OPENWRT_BOARD="ramips/mt7620"
    OPENWRT_ARCH="mipsel_24kc"
    OPENWRT_TAINTS=""
    OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
    OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
    OPENWRT_DEVICE_PRODUCT="Generic"
    OPENWRT_DEVICE_REVISION="v0"
    OPENWRT_RELEASE="OpenWrt 22.03.2 r19803-9a599fee93"

    root@OpenWrt:~# cat /etc/opkg/distfeeds.conf
    src/gz openwrt_core https://downloads.openwrt.org/releases/22.03.2/targets/ramips/mt7620/packages
    src/gz openwrt_base https://downloads.openwrt.org/releases/22.03.2/packages/mipsel_24kc/base
    src/gz openwrt_luci https://downloads.openwrt.org/releases/22.03.2/packages/mipsel_24kc/luci
    src/gz openwrt_packages https://downloads.openwrt.org/releases/22.03.2/packages/mipsel_24kc/packages
    src/gz openwrt_routing https://downloads.openwrt.org/releases/22.03.2/packages/mipsel_24kc/routing
    src/gz openwrt_telephony https://downloads.openwrt.org/releases/22.03.2/packages/mipsel_24kc/telephony

    root@OpenWrt:~# cat /etc/opkg/customfeeds.conf
    # add your custom package feeds here
    #
    # src/gz example_feed_name http://www.example.com/path/to/files
    src/gz stangri_repo https://repo.openwrt.melmac.net

    root@OpenWrt:~# opkg info luci-compat
    Package: luci-compat
    Version: git-22.069.45071-03bb0e2
    Depends: libc, luci-base
    Status: unknown ok not-installed
    Section: luci
    Architecture: all
    Size: 46221
    Filename: luci-compat_git-22.069.45071-03bb0e2_all.ipk
    Description: LuCI compatibility libraries

    root@OpenWrt:~# opkg info luci-compat
    Package: luci-compat
    Version: git-22.069.45071-03bb0e2
    Depends: libc, luci-base
    Status: install user installed
    Section: luci
    Architecture: all
    Size: 46221
    Filename: luci-compat_git-22.069.45071-03bb0e2_all.ipk
    Description: LuCI compatibility libraries
    Installed-Time: 1673295691

    root@OpenWrt:~# opkg info luci-app-wireshark-helper
    Package: luci-app-wireshark-helper
    Version: 0.0.1-5
    Depends: libc, luci-compat, luci-base, wireshark-helper, luci-lua-runtime
    Status: unknown ok not-installed
    Section: luci
    Architecture: all
    Size: 4499
    Filename: luci-app-wireshark-helper_0.0.1-5_all.ipk
    Description: Provides Web UI for Wireshark-helper.

    root@OpenWrt:~# opkg whatdepends luci-compat
    Root set:
      luci-compat
    What depends on root set

lolyinseo, 2023-01-09:
root@OpenWrt:~# opkg install luci-app-wireshark-helper
    Unknown package 'luci-app-wireshark-helper'.
    Collected errors:
     * pkg_hash_check_unresolved: cannot find dependency luci-lua-runtime for luci-app-wireshark-helper
     * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-wireshark-helper found, but incompatible with the architectures configured
     * opkg_install_cmd: Cannot install package luci-app-wireshark-helper.

stangri, 2023-01-09:
Sadly, I can reproduce. I get the weird warning even trying to reinstall packages requiring luci-compat which I've already included in my custom build with an image builder. It may have something to do with the 22.03.3 SDK, I'll try rebuilding the IPK using an older SDK and will let you know.

stangri, 2023-01-09:
Try running `opkg update; opkg install --force-depends luci-app-wireshark-helper` again. The IPK rebuilt with the 22.03.2 SDK seems to install just fine.

lolyinseo, 2023-01-10:
> I'll try rebuilding the IPK using an older SDK and will let you know.

Yep, that solve problem. All fine now. Tnx!

closed 2023-01-10 by lolyinseo